Koenkk/zigbee2mqtt #16275: Ikea E2002 goes offline as soon as paired

Freeswiss, 2023-01-20:
### What happened?

I installed Haos on my Pi3 with a Conbee2. Got zigbee2mqtt running and startet adding my tradfri devices.

1. One GU10 Bulb
2. One E1810 Remote
3. One E2002 Remote

I worked with Controllerx to get the Remotes to control the Lamp. That all works with the E1810 but I have a Problem with the E2002.

When I pair the E2002 in zigbee2mqtt it shows it as online, but switches to offline immideatly. When I press any buttons I see nothing in the Log. I only get log entrys when Joining or removing the batteries and restarting the remote. It also shows me 255% Battery level all the time.

### What did you expect to happen?

Logs of buttonpresses.
Device online.
Accurate Battery Level.

### Zigbee2MQTT version

1.29.2

### Adapter firmware version

26720700

### Adapter

Conbee2

### Debug log

Zigbee2MQTT:info  2023-01-20 14:05:20: Zigbee: allowing new devices to join.
Zigbee2MQTT:info  2023-01-20 14:05:20: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"z0ioc-1"}'
Zigbee2MQTT:info  2023-01-20 14:05:28: Device '0x842e14fffe70da64' joined
Zigbee2MQTT:info  2023-01-20 14:05:28: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x842e14fffe70da64","ieee_address":"0x842e14fffe70da64"},"type":"device_joined"}'
Zigbee2MQTT:info  2023-01-20 14:05:28: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x842e14fffe70da64"},"type":"device_connected"}'
Zigbee2MQTT:info  2023-01-20 14:05:28: MQTT publish: topic 'homeassistant/sensor/0x842e14fffe70da64/battery/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0x842e14fffe70da64/availability"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0x842e14fffe70da64"],"manufacturer":"IKEA","model":"STYRBAR remote control (E2001/E2002)","name":"0x842e14fffe70da64","sw_version":"2.4.5"},"device_class":"battery","enabled_by_default":true,"entity_category":"diagnostic","json_attributes_topic":"zigbee2mqtt/0x842e14fffe70da64","name":"0x842e14fffe70da64 battery","state_class":"measurement","state_topic":"zigbee2mqtt/0x842e14fffe70da64","unique_id":"0x842e14fffe70da64_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
Zigbee2MQTT:info  2023-01-20 14:05:28: MQTT publish: topic 'homeassistant/sensor/0x842e14fffe70da64/action/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0x842e14fffe70da64/availability"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0x842e14fffe70da64"],"manufacturer":"IKEA","model":"STYRBAR remote control (E2001/E2002)","name":"0x842e14fffe70da64","sw_version":"2.4.5"},"enabled_by_default":true,"icon":"mdi:gesture-double-tap","json_attributes_topic":"zigbee2mqtt/0x842e14fffe70da64","name":"0x842e14fffe70da64 action","state_topic":"zigbee2mqtt/0x842e14fffe70da64","unique_id":"0x842e14fffe70da64_action_zigbee2mqtt","value_template":"{{ value_json.action }}"}'
Zigbee2MQTT:info  2023-01-20 14:05:28: MQTT publish: topic 'homeassistant/sensor/0x842e14fffe70da64/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0x842e14fffe70da64/availability"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0x842e14fffe70da64"],"manufacturer":"IKEA","model":"STYRBAR remote control (E2001/E2002)","name":"0x842e14fffe70da64","sw_version":"2.4.5"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/0x842e14fffe70da64","name":"0x842e14fffe70da64 linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/0x842e14fffe70da64","unique_id":"0x842e14fffe70da64_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
Zigbee2MQTT:info  2023-01-20 14:05:28: MQTT publish: topic 'homeassistant/sensor/0x842e14fffe70da64/last_seen/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0x842e14fffe70da64/availability"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0x842e14fffe70da64"],"manufacturer":"IKEA","model":"STYRBAR remote control (E2001/E2002)","name":"0x842e14fffe70da64","sw_version":"2.4.5"},"device_class":"timestamp","enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:clock","json_attributes_topic":"zigbee2mqtt/0x842e14fffe70da64","name":"0x842e14fffe70da64 last seen","state_topic":"zigbee2mqtt/0x842e14fffe70da64","unique_id":"0x842e14fffe70da64_last_seen_zigbee2mqtt","value_template":"{{ value_json.last_seen }}"}'
Zigbee2MQTT:info  2023-01-20 14:05:28: MQTT publish: topic 'homeassistant/sensor/0x842e14fffe70da64/update_state/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0x842e14fffe70da64/availability"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0x842e14fffe70da64"],"manufacturer":"IKEA","model":"STYRBAR remote control (E2001/E2002)","name":"0x842e14fffe70da64","sw_version":"2.4.5"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:update","json_attributes_topic":"zigbee2mqtt/0x842e14fffe70da64","name":"0x842e14fffe70da64 update state","state_topic":"zigbee2mqtt/0x842e14fffe70da64","unique_id":"0x842e14fffe70da64_update_state_zigbee2mqtt","value_template":"{{ value_json['update']['state'] }}"}'
Zigbee2MQTT:info  2023-01-20 14:05:28: MQTT publish: topic 'homeassistant/binary_sensor/0x842e14fffe70da64/update_available/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0x842e14fffe70da64/availability"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0x842e14fffe70da64"],"manufacturer":"IKEA","model":"STYRBAR remote control (E2001/E2002)","name":"0x842e14fffe70da64","sw_version":"2.4.5"},"device_class":"update","enabled_by_default":false,"entity_category":"diagnostic","json_attributes_topic":"zigbee2mqtt/0x842e14fffe70da64","name":"0x842e14fffe70da64 update available","payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/0x842e14fffe70da64","unique_id":"0x842e14fffe70da64_update_available_zigbee2mqtt","value_template":"{{ value_json['update']['state'] == \"available\" }}"}'
Zigbee2MQTT:info  2023-01-20 14:05:28: MQTT publish: topic 'homeassistant/update/0x842e14fffe70da64/update/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0x842e14fffe70da64/availability"}],"availability_mode":"all","command_topic":"zigbee2mqtt/bridge/request/device/ota_update/update","device":{"identifiers":["zigbee2mqtt_0x842e14fffe70da64"],"manufacturer":"IKEA","model":"STYRBAR remote control (E2001/E2002)","name":"0x842e14fffe70da64","sw_version":"2.4.5"},"device_class":"firmware","entity_picture":"https://github.com/Koenkk/zigbee2mqtt/raw/master/images/logo.png","json_attributes_topic":"zigbee2mqtt/0x842e14fffe70da64","latest_version_template":"{{ value_json['update']['latest_version'] }}","latest_version_topic":"zigbee2mqtt/0x842e14fffe70da64","name":"0x842e14fffe70da64","payload_install":"{\"id\": \"0x842e14fffe70da64\"}","state_topic":"zigbee2mqtt/0x842e14fffe70da64","unique_id":"0x842e14fffe70da64_update_zigbee2mqtt","value_template":"{{ value_json['update']['installed_version'] }}"}'
Zigbee2MQTT:info  2023-01-20 14:05:28: MQTT publish: topic 'zigbee2mqtt/0x842e14fffe70da64', payload '{"action":null,"battery":null,"last_seen":"2023-01-20T14:05:28+01:00","linkquality":255,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
Zigbee2MQTT:info  2023-01-20 14:05:28: Starting interview of '0x842e14fffe70da64'
Zigbee2MQTT:info  2023-01-20 14:05:28: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x842e14fffe70da64","ieee_address":"0x842e14fffe70da64","status":"started"},"type":"device_interview"}'
Zigbee2MQTT:info  2023-01-20 14:05:28: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x842e14fffe70da64"},"type":"pairing"}'
Zigbee2MQTT:info  2023-01-20 14:05:30: MQTT publish: topic 'zigbee2mqtt/0x842e14fffe70da64', payload '{"action":null,"battery":null,"last_seen":"2023-01-20T14:05:30+01:00","linkquality":255,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
[... cut ...]
Zigbee2MQTT:info  2023-01-20 14:21:21: MQTT publish: topic 'zigbee2mqtt/0x842e14fffe70da64', payload '{"action":null,"battery":255,"last_seen":"2023-01-20T14:21:21+01:00","linkquality":255,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
Zigbee2MQTT:info  2023-01-20 14:21:22: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x842e14fffe70da64","ieee_address":"0x842e14fffe70da64"},"type":"device_announce"}'
Zigbee2MQTT:info  2023-01-20 14:21:22: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x842e14fffe70da64"},"type":"device_announced"}'

malaakso, 2023-01-24:
Seems a duplicate of #15725 , version 2.4.5 not being supported.

Freeswiss, 2023-01-24:
> Seems a duplicate of #15725 , version 2.4.5 not being supported.

Thanks, I'll close this one.

closed 2023-01-24 by Freeswiss