Koenkk/zigbee2mqtt #16278: Occasional flickering of Muller Licht lights

Styx85, 2023-01-20:
### What happened?

Ok I think the problem is very similar to the already solved [icasa problem](https://github.com/Koenkk/zigbee2mqtt/issues/4561), but somehow the Muller Licht lights are not behaving the same way.

The issue is, that the lights flash occasionally, not on a regular time pattern but independent if on or off.
Everey time the flash the following messages are in the log

`Zigbee2MQTT:debug 2023-01-20 20:13:29: Received Zigbee message from 'Wohnzimmer Stehlampe 2', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":537462801,"imageType":268,"manufacturerCode":4137}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2023-01-20 20:13:29: Device 'Wohnzimmer Stehlampe 2' requested OTA`

and shortly therafter again

`Zigbee2MQTT:debug 2023-01-20 20:13:47: Received Zigbee message from 'Wohnzimmer Stehlampe 2', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":537462801,"imageType":268,"manufacturerCode":4137}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2023-01-20 20:13:47: Device 'Wohnzimmer Stehlampe 2' requested OTA`

I got four of these lights and all have the same behaviour.

### What did you expect to happen?

No flickering at all expected

### Zigbee2MQTT version

1.29.2

### Adapter firmware version

20220928

### Adapter

Slaesh

lux73, 2023-01-21:
try to add

```
ota:
  disable_automatic_update_check: true
```

to your configuration.yaml and see if this helps

ASNNetworks, 2023-01-21:
I have around 12 Müller light filament bulbs myself and have noticed this for a year now (well basically since the day I got them). After a while I figured out what caused it. For some reason, if you enter Z2M pairing mode (where you don't select a specific zigbee router) and then disable pairing mode again, all Müller bulbs randomly flash ever few minutes. This will never stop, untill you restart z2m add-on. So my guess is you have been pairing some devices lately as well. Restart z2m add-on and you will notice it wil stop the flashing. Enable pairing mode again and they will all flash randomly (even when off).

When using pairing mode where you use a specific zigbee router to pair through (NOT selecting a Müller bulb) will not cause this. So I just use a specific device to pair through now.

I never reported it because I have no idea how to even troubleshoot stuff like this, since there isn't logged anything. Gonna try the above method to see if it helps.

RubenKelevra, 2023-01-26:
@lux73 wrote
> try to add
> 
> ```
> ota:
>   disable_automatic_update_check: true
> ```
> 
> to your configuration.yaml and see if this helps

That will turn off Z2M function to look into the https://github.com/Koenkk/zigbee-OTA repo for new firmware files. That's by default done once a day – so completely unrelated.

lux73, 2023-01-27:
@RubenKelevra 

After reading the logfile it was clear to me is has anything to with the OTA that trigger the flicker

unlucky we get no response from the Thread Owner at all

Styx85, 2023-01-27:
Sorry I am very busy this week. I will try the several mentioned possible
solutions this weekend and report back.

Styx85, 2023-01-27:
OK, so far following.
As mentioned before the
```
ota:
  disable_automatic_update_check: true
```
does not have any influence on this behavior.

Interestingly, the Z2M pairing mode seems to have an influence. I totally forgot that I had it set to always on as I rebuild my network last October. Now I have configured:
`permit_join: false`
and the flickering is gone so far. I will watch closely if it is gone for sure or only the frequency of flickering decreased to long interval.

Anyhow, the OTA request has not gone, so maybe it is unrelated or somehow triggered in parallel while pairing by the firmware.

lux73, 2023-01-28:
thx für the Feedback, so @RubenKelevra was right with my non working hint 🙈

Styx85, 2023-01-29:
See my latest post in this issue #16371. @Koenkk fixed the OTA reply in a way that the network spamming from the Müller lights is cut down to one quarter of what it was before with the OTA requests.
I think that the flickering of the Müller lights is intended when the pairing mode of the main Adapter is activated. As I wrote in the other thread, it was suspected before, that these kinds of lamps are set to a developer mode by firmware.

Thanks @ASNNetworks so far for the hint. I will observe a little longer (at least a week), but if it is really fixed by turning off pairing mode or not setting it to always on, it is maybe better to leave it as is (wontfix), because it is kind of a warning for the user that pairing mode is (constantly) activated. 
Anyhow, maybe @Koenkk can give us a hint if he needs some additional logs. There are possibly other devices out there with this kind of behavior, and I am just thinking of a smoke sensor randomly sending some alarm at night. ;) 

In the meantime, I'll check if the company has kind of true support and may send out a newer version of the firmware.