Koenkk/zigbee2mqtt #16279: [New device support]:  TS0202' and manufacturer name '_TYZB01_rwb0hxtf

rednet1, 2023-01-20:
### Link

cloudeven

### Database entry

{"id":2,"type":"EndDevice","ieeeAddr":"0xcc86ecfffe484bf6","nwkAddr":45754,"manufId":4098,"manufName":"_TYZB01_rwb0hxtf","powerSource":"Battery","modelId":"TS0202","epList":[1,2],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,1280],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0202","manufacturerName":"_TYZB01_rwb0hxtf","stackVersion":0,"dateCode":""}},"ssIasZone":{"attributes":{"61441":1,"iasCieAddr":"0xe0798dfffe719556","zoneState":1,"currentZoneSensitivityLevel":1}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":262,"inClusterList":[1024],"outClusterList":[],"clusters":{"msIlluminanceMeasurement":{"attributes":{"measuredValue":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1674245354800,"defaultSendRequestWhen":"immediate"}

### Comments

Zigbee model 'TS0202' and manufacturer name '_TYZB01_rwb0hxtf' still recognized as unsupported

023-01-20 21:08:27 Received message from unsupported device with Zigbee model 'TS0202' and manufacturer name '_TYZB01_rwb0hxtf'

Koenkk, 2023-01-21:
added it!

closed 2023-01-21 by Koenkk

rednet1, 2023-01-21:
Hi, I changed to edge now the device is integrated but not fully :

the battery shows the % in red and after the correct symbol, it repeats continuously.

The model also included illuminance lux

Model :
https://www.zigbee2mqtt.io/devices/TS0202_3.html

rednet1, 2023-01-21:
Hi, I changed to edge now the device is integrated but not fully :

the battery shows the % in red and after the correct symbol, it repeats
continuously.

The model also included illuminance lux

Model :
https://www.zigbee2mqtt.io/devices/TS0202_3.html

regards