Koenkk/zigbee2mqtt #16283: [WL-SD001-9W] light doesn't retain it's startup settings

rechichidaniel, 2023-01-21:
### What happened?

When settings the startup behaviour from Z2M/CC2652P2 the [WL-SD001-9W](https://www.zigbee2mqtt.io/devices/WL-SD001-9W.html) doesn't retain it's on/off state, but when setting the startup behaviour using the Hue Bridge and app, the light does retain it's on/off state.
e.g. power cycling the light when it's digitally set to off should leave it off, but it instead turns back on.

I think this is because the Z2M converter for this light is missing a few commands since I can see differences in the network traffic.

Using the Z2M exposes menu for the light, you can write to the `StartUpOnOff (0x4003)` attribute on the `On/Off (0x0006)` cluster, and the `Startup Color Temperature (0x4010)` attribute on the `Color Control (0x0300)` cluster.
When setting the light to "power loss recovery" mode in the Hue app, it writes to those two attributes too. However, it also writes to the `Startup Level (0x4000)` attribute on the `Level Control (0x0008)` cluster as well as two attributes I can't identify which are the  `0x0003` and `0x0004` attributes on the `Color Control (0x0300)` cluster.

Side note: I tried to make my own customer converter for this light by copying the one from [Nue 3A.js](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/nue_3a.js#L350) but I spend days on it and got nowhere. Sorry I couldn't do more and I will post in the discussion board sometime so that I can properly learn how to do this.
I was trying to add [current_level_startup](https://github.com/Koenkk/zigbee-herdsman-converters/blob/a647274238af690d1806feb20864de97458fba6b/lib/exposes.js#L253) from [withLevelConfig](https://github.com/Koenkk/zigbee-herdsman-converters/blob/a647274238af690d1806feb20864de97458fba6b/lib/exposes.js#L239) and didn't have a chance to try adding the other missing commands.

### What did you expect to happen?

I expected the light to retain its off state when power cycled. I suspect the difference in network traffic between Hue and Z2M may be the cause.

### How to reproduce it (minimal and precise)

Use the Z2M Exposes menu for the light to set the available startup states to "previous". Wait a bit for the settings to stick.
Then:
1. Digitally turn the light off over Zigbee.
2. Physically turn the light off at the wall.
3. Physically turn the light back on at the wall. The light came back on when I wanted it to retain its off state.

On a different Zigbee network using the Hue Bridge and the Hue app, set the light to "power loss recovery". Wait a bit for the settings to stick.
Repeat the power cycling steps listed above. The light retains its off state as expected.

### Zigbee2MQTT version

1.29.2

### Adapter firmware version

20220219

### Adapter

TubeZB's CC2652P2 Based Coordinator

### Debug log

Network traffic found using Wireshark and an nRF52840 sniffer. Shared as online code snippets.
All of these are attributes written with the `Write Attributes (0x02)` command.

### StartUpOnOff (0x4003)
Cluster: `On/Off (0x0006)`
Command: `Write Attributes (0x02)`
Attribute: `StartUpOnOff (0x4003)`

Hue [command](https://goonlinetools.com/snapshot/code/#9pxxp8kff5lxmw593fupf) and [response](https://goonlinetools.com/snapshot/code/#sivedlpur72ltikkprzz6)
Z2M [command](https://goonlinetools.com/snapshot/code/#3mtyh8a6yni1oav3j2j1lz) and [response](https://goonlinetools.com/snapshot/code/#nztu2440brm82f21jvtew8)

### Startup Color Temperature (0x4010)
Cluster: `Color Control (0x0300)`
Command: `Write Attributes (0x02)`
Attribute: `Startup Color Temperature (0x4010)`

Hue [command](https://goonlinetools.com/snapshot/code/#v70g26kkr1kxqfszww21xb) and [response](https://goonlinetools.com/snapshot/code/#0159vo1xqo7remr4hy8lzco)
Z2M [command](https://goonlinetools.com/snapshot/code/#8bi5e0n8kum08sa510hu) and [response](https://goonlinetools.com/snapshot/code/#22zqf8dn53qqcjl32em6fe)

### Startup Level (0x4000)
Cluster: `Color Control (0x0008)`
Command: `Write Attributes (0x02)`
Attribute: `Startup Level (0x4000)`

Hue [command](https://goonlinetools.com/snapshot/code/#f82nwbhmslkwkr506hlas8) and [response](https://goonlinetools.com/snapshot/code/#j43x5h4yi9n4ixffhz46r)
Z2M - exposes missing from the devices converter.

### Unknown (0x0003)
Cluster: `Color Control (0x0300)`
Command: `Write Attributes (0x02)`
Attribute: `0x0003`

Hue [command](https://goonlinetools.com/snapshot/code/#4co2kocguxmhjix8paq95) and [response](https://goonlinetools.com/snapshot/code/#269j8dbslok6nz6nyfqry5)
Z2M - exposes missing from the devices converter.

### Unknown (0x0004)
Cluster: `Color Control (0x0300)`
Command: `Write Attributes (0x02)`
Attribute: `0x0004`

Hue [command](https://goonlinetools.com/snapshot/code/#h8qq77wkhn24dnat49wv3) and [response](https://goonlinetools.com/snapshot/code/#3t4ap0ca5mia3wm5sf5r1r)
Z2M - exposes missing from the devices converter.

Koenkk, 2023-01-21:
The `0x0003` and `0x0004` of `Color Control (0x0300)` are Hue specific power on behaviour attributes. The attributes you mention are also written when setting the `hue_power_on_behavior`, see: https://github.com/Koenkk/zigbee-herdsman-converters/blob/ac4eaa873db2d95e6bd0411625f45c46892b3390/lib/philips.js#L202

rechichidaniel, 2023-01-21:
Thanks Keonkk,

How do I add `hue_power_on_behavior` to the [WL-SD001-9W](https://www.zigbee2mqtt.io/devices/WL-SD001-9W.html) light?
I attempted to add `hue_power_on_behavior` to my custom converter but couldn't make it work.
[lib/philips.js (L105)](https://github.com/Koenkk/zigbee-herdsman-converters/blob/ac4eaa873db2d95e6bd0411625f45c46892b3390/lib/philips.js#L105) adds it with a concat to toZigbee, but I didn't have `result` defined so I used `toZigbee.push`.
I couldn't find anywhere in [lib/philips.js](https://github.com/Koenkk/zigbee-herdsman-converters/blob/ac4eaa873db2d95e6bd0411625f45c46892b3390/lib/philips.js) or [devices/philips.js](https://github.com/Koenkk/zigbee-herdsman-converters/blob/ac4eaa873db2d95e6bd0411625f45c46892b3390/devices/philips.js) where `hue_power_on_behavior` was added to exposes so didn't make much progress there.

I also tried using `light_onoff_brightness_colortemp_color` from [lib/philips.js (L99)](https://github.com/Koenkk/zigbee-herdsman-converters/blob/ac4eaa873db2d95e6bd0411625f45c46892b3390/lib/philips.js#L99) instead of from [extend.js (L96)](https://github.com/Koenkk/zigbee-herdsman-converters/blob/a647274238af690d1806feb20864de97458fba6b/lib/extend.js#L96), but I couldn't make that work. I didn't pursue this too much because I wasn't sure if I'd break anything given that this light is only pretending to be a Philips light. Although, it's probably fine given that it works with the Hue Bridge and App.

### Custom Converter Attempts
1. Custom converter that uses the `light_onoff_brightness_colortemp_color` from [lib/philips.js (L99)](https://github.com/Koenkk/zigbee-herdsman-converters/blob/ac4eaa873db2d95e6bd0411625f45c46892b3390/lib/philips.js#L99)
The "extend: philips.extend...." line near the end is copied directly from /devices/philips.js with all the same const definitions so I don't know why it doesn't work.

    * [Converter](https://goonlinetools.com/snapshot/code/#4uiuqqlqwzx8cd65sdbibv)
    * [Z2M Log Error](https://goonlinetools.com/snapshot/code/#3jzt6s1xug6vj865q12rb)

2. Custom converter that tries to add `level_config` from [exposes.js L239](https://github.com/Koenkk/zigbee-herdsman-converters/blob/a647274238af690d1806feb20864de97458fba6b/lib/exposes.js#L239)

    * [Converter](https://goonlinetools.com/snapshot/code/#1lfvnkrysacp1xojdm960n)
    * Z2M can start, but in the Exposes menu, `state` and `brightness` are duplicated and even though the 4 settings/controls in `level_config` are now shown, they don't do anything. No network traffic, except for when the refresh buttons are clicked which sends the Read Attributes command. `exposes.push(e.withLevelConfig())` doesn't work but `exposes.push(e.light_brightness().withLevelConfig())` does. But in the Light class ([exposes.js L213](https://github.com/Koenkk/zigbee-herdsman-converters/blob/a647274238af690d1806feb20864de97458fba6b/lib/exposes.js#L213) shows both functions on the same level.

3. Custom converter that more closely follows the [Zigbee2MQTT docs guide](https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html) and that uses exposes/toZigbee/fromZigbee instead of extend, and copying everything I can from the `light_onoff_brightness_colortemp_color` definitions in both [lib/philips.js L99](https://github.com/Koenkk/zigbee-herdsman-converters/blob/ac4eaa873db2d95e6bd0411625f45c46892b3390/lib/philips.js#L99) and [lib/extend.js L96](https://github.com/Koenkk/zigbee-herdsman-converters/blob/a647274238af690d1806feb20864de97458fba6b/lib/extend.js#L96)

    * [Converter](https://goonlinetools.com/snapshot/code/#vitr1a64quhzlf62vmdtj)
    * [Z2M Log Error](https://goonlinetools.com/snapshot/code/#xzy7ivlrtlfsp3ommtmlz). I got stuck at defining philipsTz from [devices/philips.js L151](https://github.com/Koenkk/zigbee-herdsman-converters/blob/ac4eaa873db2d95e6bd0411625f45c46892b3390/lib/philips.js#L151)

rechichidaniel, 2023-01-21:
The other thing I'm trying to solve here is that sometimes the settings for `color_temp_startup` and `power_on_behavior` revert back from `previous` to something else. This doesn't happen with the Hue Bridge so I'm hoping that setting all the other attributes resolves this.

Koenkk, 2023-01-22:
To use `philips.extend` you need to be on the dev branch (it will also be in the 1 February release). This was added very recently. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html). Also make sure to put `const philips = require('zigbee-herdsman-converters/lib/philips');` at the top of your converter.

rechichidaniel, 2023-01-29:
Thanks Koenkk,
That's pretty soon so I think I'll wait for the update and try again.

rechichidaniel, 2023-02-08:
HI Koenkk, now that I've updated to the latest Zigbee2MQTT I was able to get Z2M to boot using `philips.extend` in a custom converter.
But I can't see any extra settings in the Exposes menu like the missing `Startup Level (0x4000)` attribute on the `Level Control (0x0008)` cluster or the 2 Phiulips specific commands I labelled as "Unknown".

Is it possible to still use the `phlips.extend` in my converter but also add more exposes? Maybe I can add the other commands that way (not sure how to add these commands but that might be a direction I can explore).

This is what I have so far, and by changing the description I can confirm that it's Z2M is definitely using it.
```javascript

// From ".../devices/nue_3a.js" and ".../devices/philips.js"
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const fz = {...require('zigbee-herdsman-converters/converters/fromZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').fromZigbee};
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
// From ".../devices/philips.js"
const ota = require('zigbee-herdsman-converters/lib/ota');
const philips = require('zigbee-herdsman-converters/lib/philips');

const definition = {
    zigbeeModel: ['LXN60-LS27-Z30', 'FEB56-ZCW2CLX1.0'],
    model: 'WL-SD001-9W',
    vendor: 'Nue / 3A',
    description: '9W RGB LED downlight - Custom Converter - Philips Extend',
    extend: philips.extend.light_onoff_brightness_colortemp_color({colorTempRange: [153, 370]}),
};

module.exports = definition;
```

Koenkk, 2023-02-08:
It is not in the exposes indeed, you can publish a MQTT messgage manually: https://www.zigbee2mqtt.io/devices/046677551780.html#power-on-behavior

rechichidaniel, 2023-02-12:
Thanks Koen,

When using the following payload from the link you sent me and the MQTT Z2M set command, all the commands I needed were sent, and the light did exactly what it was meant to (most of the time).
```json
{
    "hue_power_on_behavior": "recover"
}
```

If I want Zigbee2MQTT to always send that command to my lights when they are digitally turned off, do I need to write an extension?

Koenkk, 2023-02-12:
@rechichidaniel you can do it via an extension or via e.g. an automation in Home Assistant.

closed 2023-03-15 by rechichidaniel