Koenkk/zigbee2mqtt #16290: BHT-006GAZB wrong recognized

elemarek, 2023-01-21:
### What happened?

BHT-006GAZB recognized as 	BHT-002-GCLZB. Not all functions work.  "deadzone_temperature": null, "max_temperature_limit": null

### Zigbee2MQTT version

1.29.2-1

### Adapter firmware version

20210120

### Adapter

slaesh cc2652 stick

### Debug log

debug 2023-01-21 17:21:26: Received Zigbee message from '0x84ba20fffe28c0f2', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":36}],"seq":8960}' from endpoint 1 with groupID 0
info  2023-01-21 17:21:26: MQTT publish: topic 'zigbee2mqtt/0x84ba20fffe28c0f2', payload '{"child_lock":"UNLOCK","current_heating_setpoint":23,"deadzone_temperature":null,"heat":"OFF","linkquality":129,"local_temperature":26,"local_temperature_calibration":0,"max_temperature":45,"max_temperature_limit":null,"min_temperature_limit":1,"preset":"hold","preset_mode":"hold","program":{"saturday_p1_hour":6,"saturday_p1_minute":0,"saturday_p1_temperature":24,"saturday_p2_hour":12,"saturday_p2_minute":0,"saturday_p2_temperature":23,"saturday_p3_hour":14,"saturday_p3_minute":30,"saturday_p3_temperature":22,"saturday_p4_hour":17,"saturday_p4_minute":30,"saturday_p4_temperature":21,"sunday_p1_hour":6,"sunday_p1_minute":0,"sunday_p1_temperature":19,"sunday_p2_hour":12,"sunday_p2_minute":30,"sunday_p2_temperature":20,"sunday_p3_hour":14,"sunday_p3_minute":30,"sunday_p3_temperature":21,"sunday_p4_hour":18,"sunday_p4_minute":30,"sunday_p4_temperature":20,"weekdays_p1_hour":6,"weekdays_p1_minute":0,"weekdays_p1_temperature":20,"weekdays_p2_hour":11,"weekdays_p2_minute":30,"weekdays_p2_temperature":21,"weekdays_p3_hour":13,"weekdays_p3_minute":30,"weekdays_p3_temperature":22,"weekdays_p4_hour":17,"weekdays_p4_minute":30,"weekdays_p4_temperature":23},"running_state":"idle","sensor":"AL","system_mode":"heat"}'

elemarek, 2023-01-21:
Link: https://www.aliexpress.com/item/1005004829125369.html?spm=a2g0o.order_list.order_list_main.11.46221802DBk3rN

M450NC, 2023-02-26:
Same issue. Series 006 showing as BHT-002-GCLZB and cannot use max temp, deadzone or adjust programme schedule in Z2M...

Collinio, 2023-03-05:
Pretty new to this but bought this thermostat as well. After a little experimenting i think that min_temperature_limit maps to deadzone_temperature for model 006. And the problem with the scheduled heating program could be a much too long state. I constantly get this error in my home-assistant.log:
```
2023-03-05 16:37:37.390 ERROR (MainThread) [homeassistant] Error doing job: Exception in callback MQTT._mqtt_handle_message(<paho.mqtt.cl... 0x7f928723b0>)
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/asyncio/events.py", line 80, in _run
    self._context.run(self._callback, *self._args)
  File "/usr/src/homeassistant/homeassistant/components/mqtt/client.py", line 722, in _mqtt_handle_message
    self._mqtt_data.state_write_requests.process_write_state_requests()
  File "/usr/src/homeassistant/homeassistant/components/mqtt/models.py", line 267, in process_write_state_requests
    entity.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 559, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 676, in _async_write_ha_state
    self.hass.states.async_set(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1537, in async_set
    state = State(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1186, in __init__
    raise InvalidStateError(
homeassistant.exceptions.InvalidStateError: Invalid state encountered for entity ID: sensor.thermostat_wohnzimmer_program. State max length is 255 characters.
```
While z2m is constantly publishing:
`info  2023-03-05 16:41:44: MQTT publish: topic 'zigbee2mqtt/Thermostat Wohnzimmer', payload '{"child_lock":"UNLOCK","current_heating_setpoint":23,"deadzone_temperature":null,"heat":"ON","linkquality":228,"local_temperature":22,"local_temperature_calibration":-4,"max_temperature":45,"max_temperature_limit":null,"min_temperature_limit":1,"preset":"hold","preset_mode":"hold","program":{"saturday_p1_hour":6,"saturday_p1_minute":0,"saturday_p1_temperature":24,"saturday_p2_hour":12,"saturday_p2_minute":0,"saturday_p2_temperature":23,"saturday_p3_hour":14,"saturday_p3_minute":30,"saturday_p3_temperature":22,"saturday_p4_hour":17,"saturday_p4_minute":30,"saturday_p4_temperature":21,"sunday_p1_hour":6,"sunday_p1_minute":0,"sunday_p1_temperature":19,"sunday_p2_hour":12,"sunday_p2_minute":30,"sunday_p2_temperature":20,"sunday_p3_hour":14,"sunday_p3_minute":30,"sunday_p3_temperature":21,"sunday_p4_hour":18,"sunday_p4_minute":30,"sunday_p4_temperature":20,"weekdays_p1_hour":6,"weekdays_p1_minute":0,"weekdays_p1_temperature":15,"weekdays_p2_hour":11,"weekdays_p2_minute":30,"weekdays_p2_temperature":21,"weekdays_p3_hour":13,"weekdays_p3_minute":30,"weekdays_p3_temperature":22,"weekdays_p4_hour":17,"weekdays_p4_minute":30,"weekdays_p4_temperature":23},"running_state":"heat","sensor":"AL","system_mode":"heat"}'`

Not sure if this is the problem when it is not also reported for 002

M0DDer, 2023-03-11:
I have the same problems

Vesa62, 2023-03-30:
Same problem. Is there a way to set the deadzone_temperature?
Is somebody working to support BHT-006?

closed 2023-05-08 by github-actions[bot]

OleksandrChekmez, 2023-05-19:
Just purchased Moes BHT-006GAZB and got this error when try to add the device:
45Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE204_aoclfnxz'

PulsarFX, 2023-12-06:
maybe see #19991 to get this sorted out?