Koenkk/zigbee2mqtt #16291: CASAIA CCB432 : power and energy not updated when changed (just updated once at setup or reconfiguration

HakunaMatmata, 2023-01-21:
### What happened?

Hello,

I have a Casaia CCB432 (rail din switch in my electric closet) : it let me control my water heater (on/off) and its electrical consuption (power and energy should be supported).
Product : https://www.zigbee2mqtt.io/devices/CCB432.html

I can control the on/off function.
But I can not see the instantaneous power and energy meter in home assistant : 
- these 2 variable are only set once at setup (at 0W or 2000W depending on my water heater state when I set up the CCB432) and are never updated after
- I can refresh these values by manually reconfiguring the device in z2m
- values stays the same in both z2m and homeassistant (the value at setup or at reconfiguration)
- in my logs I see the right values (example below)

Example of logs in z2m : 
`Debug 2023-01-21 20:12:31Received Zigbee message from 'Chauffe-eau', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,10798],"instantaneousDemand":2118}' from endpoint 1 with groupID null
Debug 2023-01-21 20:12:31Received MQTT message on 'zigbee2mqtt/Chauffe-eau/set' with data 'OFF'
Debug 2023-01-21 20:12:31Publishing 'set' 'state' to 'Chauffe-eau'
Info 2023-01-21 20:12:31MQTT publish: topic 'zigbee2mqtt/Chauffe-eau', payload '{"energy":10.79,"linkquality":255,"power":0,"state":"OFF"}'
Debug 2023-01-21 20:12:32Received Zigbee message from 'Chauffe-eau', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,10798],"instantaneousDemand":38}' from endpoint 1 with groupID null
Debug 2023-01-21 20:12:32Received Zigbee message from 'Chauffe-eau', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID null
Info 2023-01-21 20:12:32MQTT publish: topic 'zigbee2mqtt/Chauffe-eau', payload '{"energy":10.79,"linkquality":255,"power":0,"state":"OFF"}'
Debug 2023-01-21 20:12:33Received Zigbee message from 'Chauffe-eau', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,10798],"instantaneousDemand":0}' from endpoint 1 with groupID null`

(chauffe-eau means water-heater, name of my device)

Pease let me know if any further info can be useful !

Thanks a lot for your great work (I love z2m associated to ha!) and for your help !

Mathieu

### What did you expect to happen?

I expected z2m and homeassistant to update the value for power and energy depending on my instaneous electric consumption (instead of being blocked on setup values)

### How to reproduce it (minimal and precise)

setup CASAIA CCB432 (with device switched off => power=0W)
monitor power and energy value in Z2M (expose tab)
switch device on and consume power throught it (power > 0>)

=> power stays in z2m monitoring at 0W

### Zigbee2MQTT version

1.29.2 commit: bb3e8f6

### Adapter firmware version

0x26660700

### Adapter

ConBee2/RaspBee2

### Debug log

Debug 2023-01-21 20:12:22Received Zigbee message from 'Chauffe-eau', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,10792],"instantaneousDemand":2139}' from endpoint 1 with groupID null
Debug 2023-01-21 20:12:23Received Zigbee message from 'Chauffe-eau', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,10793],"instantaneousDemand":2132}' from endpoint 1 with groupID null
Debug 2023-01-21 20:12:25Received Zigbee message from 'Chauffe-eau', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,10794],"instantaneousDemand":2126}' from endpoint 1 with groupID null
Debug 2023-01-21 20:12:25Received Zigbee message from 'Chauffe-eau', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,10794],"instantaneousDemand":2125}' from endpoint 1 with groupID null
Debug 2023-01-21 20:12:26Received Zigbee message from 'Chauffe-eau', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,10795],"instantaneousDemand":2115}' from endpoint 1 with groupID null
Debug 2023-01-21 20:12:27Received Zigbee message from 'Chauffe-eau', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,10795],"instantaneousDemand":2123}' from endpoint 1 with groupID null
Debug 2023-01-21 20:12:28Received Zigbee message from 'Chauffe-eau', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,10796],"instantaneousDemand":2121}' from endpoint 1 with groupID null
Debug 2023-01-21 20:12:29Received Zigbee message from 'Chauffe-eau', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,10797],"instantaneousDemand":2112}' from endpoint 1 with groupID null
Debug 2023-01-21 20:12:30Received Zigbee message from 'Chauffe-eau', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,10797],"instantaneousDemand":2117}' from endpoint 1 with groupID null
Debug 2023-01-21 20:12:31Received Zigbee message from 'Chauffe-eau', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,10798],"instantaneousDemand":2118}' from endpoint 1 with groupID null
Debug 2023-01-21 20:12:31Received MQTT message on 'zigbee2mqtt/Chauffe-eau/set' with data 'OFF'
Debug 2023-01-21 20:12:31Publishing 'set' 'state' to 'Chauffe-eau'
Info 2023-01-21 20:12:31MQTT publish: topic 'zigbee2mqtt/Chauffe-eau', payload '{"energy":10.79,"linkquality":255,"power":0,"state":"OFF"}'
Debug 2023-01-21 20:12:32Received Zigbee message from 'Chauffe-eau', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,10798],"instantaneousDemand":38}' from endpoint 1 with groupID null
Debug 2023-01-21 20:12:32Received Zigbee message from 'Chauffe-eau', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID null
Info 2023-01-21 20:12:32MQTT publish: topic 'zigbee2mqtt/Chauffe-eau', payload '{"energy":10.79,"linkquality":255,"power":0,"state":"OFF"}'
Debug 2023-01-21 20:12:33Received Zigbee message from 'Chauffe-eau', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,10798],"instantaneousDemand":0}' from endpoint 1 with groupID null
Debug 2023-01-21 20:12:47Received Zigbee message from 'Chauffe-eau', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":1,"imageType":0,"manufacturerCode":4412}' from endpoint 1 with groupID null
Debug 2023-01-21 20:12:47Device 'Chauffe-eau' requested OTA
Debug 2023-01-21 20:17:49Received Zigbee message from 'Chauffe-eau', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":1,"imageType":0,"manufacturerCode":4412}' from endpoint 1 with groupID null
Debug 2023-01-21 20:17:49Device 'Chauffe-eau' requested OTA
Debug 2023-01-21 20:22:50Received Zigbee message from 'Chauffe-eau', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":1,"imageType":0,"manufacturerCode":4412}' from endpoint 1 with groupID null
Debug 2023-01-21 20:22:51Device 'Chauffe-eau' requested OTA

HakunaMatmata, 2023-01-30:
Hello @Koenkk ,

I searched for similar problems and saw that you sometimes advise to delete the device or to reconfigure (https://www.zigbee2mqtt.io/guide/usage/mqtt_topics_and_messages.html#zigbee2mqtt-bridge-request-device-configure) 

I tried both manually and via mqtt message, but without success yet.

Looking at the debug logs, it seems that there are some zibgee messages received from the device with power and energy info, but despite the reporting configuration, these info are not published to mqqt, nor are they registered in the state.json.

What could cause this behavior and how can I mitigate it?

Thank you for your help :)

HakunaMatmata, 2023-02-06:
Anyone can help please?
Am I the only one who can not get energy values through mqtt?
Is it a problem with this specific integration?

Thank you

HakunaMatmata, 2023-02-25:
Hello,
Problem still présent :  
- values are updated through zigbee (seen in logs)
- they do not update device state (I think the pb is there) 
- thus, information is not published to home Assistant through mqtt

Koenkk, 2023-02-26:
Can you try to change the min/max reporting interval by clicking on the device in the z2m frontend -> reporting and see if that helps?

HakunaMatmata, 2023-02-26:
Hello,
I tried to switch the min reporting interval from 5 to 1, then to 0 => no change
I also tried the max from 3600 to 10 => no change

The controls seems ok, as the reportings are more or less frequent (visible in logs like : _Debug 2023-01-21 20:12:32Received Zigbee message from..._)
However the state of the device is not updated and then the mqtt message not published.

Someone on the discord channel suggested me that the problem could come from the converter (herdmans project).
I think he is right since another device (OWON CB432 has exactly the same implementation and suffers the same symptoms : https://github.com/Koenkk/zigbee2mqtt/issues/10954 ).

I see that you also run the converter project, do you prefer me to close this issue and to re-open one in herdsmans converter project?

Thank you

Koenkk, 2023-02-27:
Could you provide the data/database entry of your device and your configuration.yaml? (configuration without sensitive information)

HakunaMatmata, 2023-03-02:
Hello @Koenkk 
here is the entry in database.db :

`{"id":13,"type":"Router","ieeeAddr":"0x3c6a2cfffed0cf12","nwkAddr":51999,"manufId":4412,"manufName":"CASAIA","powerSource":"Mains (single phase)","modelId":"CCB432","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":9,"inClusterList":[0,3,4,5,6,1794],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"appVersion":0,"stackVersion":0,"hwVersion":0}},"seMetering":{"attributes":{"multiplier":1,"divisor":1000,"currentSummDelivered":[0,92687],"instantaneousDemand":2040}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00212effff060e0e","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00212effff060e0e","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00212effff060e0e","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0},{"cluster":1794,"attrId":1024,"minRepIntval":5,"maxRepIntval":3600,"repChange":1},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1]}],"meta":{}}},"appVersion":0,"stackVersion":0,"hwVersion":0,"dateCode":"20210308","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1753472553},"lastSeen":1677794480247,"defaultSendRequestWhen":"immediate"}
`

And here is the config.yaml (the device is the last one) : 

`homeassistant: true
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: '!secret.yaml server'
  user: '!secret.yaml user'
  password: '!secret.yaml password'
serial:
  adapter: deconz
  port: /dev/ttyACM0
frontend:
  port: 4231
  auth_token: '!secret.yaml passwordfe'
devices:
  '0x14b457fffec327d7':
    friendly_name: Lampe Patates
  '0x60a423fffe651749':
    friendly_name: Lampe Plafond
  '0x60a423fffe65141c':
    friendly_name: Lampe Puisine
  '0xec1bbdfffeb44d02':
    friendly_name: Interrupteur Salon 1
    legacy: false
  '0xec1bbdfffeb8b32a':
    friendly_name: Interrupteur Salon 2
    legacy: false
  '0x0015bc001a01e559':
    friendly_name: Taupe
  '0x14b457fffec362f2':
    friendly_name: Lampe Pixar
  '0x14b457fffec35a58':
    friendly_name: Disco Kéké
  '0x00047400008d12c4':
    friendly_name: Volet Chambre
    invert_cover: false
  '0x000474000012bb29':
    friendly_name: Radiateur Chambre
  '0xbc33acfffe770c2b':
    friendly_name: Interrupteur Mur Chambre
  '0xec1bbdfffeb8c4fc':
    friendly_name: Interrupteur Chambre
    legacy: false
  '0x680ae2fffe82bd83':
    friendly_name: Lampe Pieu
  '0x000474000012bb3c':
    friendly_name: Radiateur Salle à manger
  '0x000474000012bbfe':
    friendly_name: Radiateur Salon
  '0x00158d000966cba6':
    friendly_name: Compteur Linky
  '0xdc8e95fffe8af666':
    friendly_name: Répéteur
  '0x3c6a2cfffed0cf12':
    friendly_name: Chauffe-eau
    voltage_precision: 0
groups: {}
advanced:
  log_level: debug
`

I use a workaround to make the device work since this monday : I have a cron running every minute to query the device via mqtt.
However, there may be something to change to the converter to make the device auto updating its status through mqtt.
Please let me know if I need to check anything else.

Thank you

Koenkk, 2023-03-04:
Could you check if the issue is fixed with the following external converter: 

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['CCB432'],
    model: 'CCB432',
    vendor: 'CASAIA',
    description: 'Rail-Din relay and energy meter',
    fromZigbee: [fz.electrical_measurement, fz.metering, fz.on_off],
    toZigbee: [tz.on_off],
    exposes: [e.switch(), e.power(), e.energy()],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'seMetering']);
        await reporting.onOff(endpoint);
        await reporting.readMeteringMultiplierDivisor(endpoint);
        await reporting.instantaneousDemand(endpoint);
        await reporting.currentSummDelivered(endpoint);
    },
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

HakunaMatmata, 2023-03-05:
Hello,
I added the external converters (js file + lines in configuration.yaml), then restarted Z2M & MQTT but unfortunatelly there is no change : 

`Debug 2023-03-05 11:31:24Received Zigbee message from 'Chauffe-eau', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,110344],"instantaneousDemand":2053}' from endpoint 1 with groupID null
Debug 2023-03-05 11:31:24Received MQTT message on 'zigbee2mqtt/Chauffe-eau/set' with data 'ON'
Debug 2023-03-05 11:31:24Publishing 'set' 'state' to 'Chauffe-eau'
Info 2023-03-05 11:31:24MQTT publish: topic 'zigbee2mqtt/Chauffe-eau', payload '{"energy":110.16,"linkquality":255,"power":2067,"state":"ON"}'`

In this log I just sent throught MQTT an "on" state to the device so it updated its values.
And energy is stuck at 110.16, power at 2067 whereas the real values received by the device were 110.34 and 2053.

Koenkk, 2023-03-06:
Does the reconfigure through the z2m succeed? (yellow refresh icon in the z2m frontend) You should see something like `Successfully configured ...`

HakunaMatmata, 2023-03-06:
Yes it does, green message successfully configured.

tpawork, 2023-03-07:
@Koenkk would this also fix #16162  or does it need to be fixed sperately?

Koenkk, 2023-03-07:
@HakunaMatmata then I'm not sure why it doesn't start reporting. Are you able to verify if the device works with a different bridge?

HakunaMatmata, 2023-03-11:
Unfortunately I am not.

But I tried it with zha directly in home Assistant, it works.
I prefer z2m so for now I configured a cron every minute to publish the state of this device.

Koenkk, 2023-03-15:
Should be fixed now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-03-15 by Koenkk

HakunaMatmata, 2023-04-09:
Hello, 
I am sorry for my late reply.
Just updated to version 1.30.3 of Z2M, but this does not seem to work.
I saw on the other (owon) post a converter you proposed but not sure of the values I should try, as the device is not from the same brand.

Koenkk, 2023-04-10:
Can you try the ext converter from https://github.com/Koenkk/zigbee2mqtt/issues/16291#issuecomment-1454764581. To verify that it has been loaded, check if the description in the z2m frontend -> device ends with `CUSTOM`. If that doesn't work, provide me the debug logging when the device is expected to report energy (e.g. when attaching a load).

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

JC38, 2023-04-30:
Hello, just for your information. For the CB432 device by Owon, the js file should be set as this :
`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['CB432'],
    model: 'CB432',
    vendor: 'OWON',
    description: '32A/63A power circuit breaker CUSTOM',
    fromZigbee: [fz.electrical_measurement, fz.metering, fz.on_off],
    toZigbee: [tz.on_off],
    meta: {publishDuplicateTransaction: true},
    exposes: [e.switch(), e.power(), e.energy()],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'seMetering']);
        await reporting.onOff(endpoint);
        await reporting.readMeteringMultiplierDivisor(endpoint);
        await reporting.instantaneousDemand(endpoint);
        await reporting.currentSummDelivered(endpoint);
    },
};

module.exports = definition;
`
I am running zigbee2mqtt 1.30.3, and I am a newbie with z2m, using this extension will be mandatory for all the time or will a change be made into the main repository so this ext file will not be necessary anymore?

JC38, 2023-04-30:
Hello, just for your information. For the CB432 device by Owon, the js file should be set as this :
`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['CB432'],
    model: 'CB432',
    vendor: 'OWON',
    description: '32A/63A power circuit breaker CUSTOM',
    fromZigbee: [fz.electrical_measurement, fz.metering, fz.on_off],
    toZigbee: [tz.on_off],
    meta: {publishDuplicateTransaction: true},
    exposes: [e.switch(), e.power(), e.energy()],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'seMetering']);
        await reporting.onOff(endpoint);
        await reporting.readMeteringMultiplierDivisor(endpoint);
        await reporting.instantaneousDemand(endpoint);
        await reporting.currentSummDelivered(endpoint);
    },
};

module.exports = definition;
`
I am running zigbee2mqtt 1.30.3, and I am a newbie with z2m, using this extension will be mandatory for all the time or will a change be made into the main repository so this ext file will not be necessary anymore ?