Koenkk/zigbee2mqtt #16298: Each Lixee ZLinky states in HA contain all attributes, resulting in filling up the recorder database very fast

mwi001, 2023-01-22:
### What happened?

Hello,

I recently installed a Lixee ZLinky device (device supported by the line.js file). The setup with Zigbee2mqtt went flawlessly. But after a few days I realized that my recorder database size grew up from a few MB to several GB. Actually in the "states" table, each ZLinky entity had millions of records each, even if corresponding values didn't change. With further investigations, I realized that the corresponding record in the "state_attributes" table contained references to *all* attributes for a state change, and not only the attribute corresponding to the entity. You can confirm this in the HA "state" dev tools menu : each ZLinky entity references all exposed attributes and not only the corresponding attribute of the entity.

For instance for the "Linky active register tier delivered" entity, the referenced state attributes are (Linky is the friendly name of my ZLinky device): 
_friendly_name: Linky active register tier delivered
active_power: -32768
active_power_max: -32768
active_register_tier_delivered: HP..
apparent_power: 2020
available_power: 60
current_tarif: HC..
current_tier1_summ_delivered: 24155
current_tier2_summ_delivered: 27961
linkquality: 196
meter_serial_number: '021961258464'
rms_current: 8
rms_current_max: 90
rms_voltage: 65535
schedule_h_p_h_c: 0
start_notice_e_j_p: 0
status_register: ''
update:
  installed_version: 12
  latest_version: 12
  state: idle
update_available: false
warn_d_p_s: 0_

Instead I would expect to find only:
_friendly_name: Linky active register tier delivered
active_register_tier_delivered: HP.._

And it is the same for all the ZLinky entities.

The consequence is that that whenever a ZLinky attribute changes, state records are created in the recorded database for all entities instead of just the entity corresponding to the attribute.

### What did you expect to happen?

In the dev tools / state menu, I would expect that for each ZLinky entity only the corresponding entity attribute be referenced as the state attribute, and not the full set of ZLinky attributes. Hopefully this will avoid the recorder database to be overwhelmed with Zlinky state records, and correctly record states only when the corresponding attribute changes (and not when *any* of the ZLinky attributes changes).

### How to reproduce it (minimal and precise)

Connect a Lixee ZLinky device with Zigbee2MQTT, and look at "dev tools/states" for ZLinky entities. For each of them, the whole attribute set is referenced in the state attributes, instead of only the attribute corresponding to the entity. Look also at the recorder database records in the "states" and "state_attributes" tables and search for ZLinky related records.

### Zigbee2MQTT version

1.29.2

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff Zigbee 3 USB dongle plus (EZSP version)

### Debug log

Info 2023-01-22 12:56:59MQTT publish: topic 'zigbee2mqtt/Linky', payload '{"active_power":-32768,"active_power_max":-32768,"active_register_tier_delivered":"HP..","apparent_power":940,"available_power":60,"current_tarif":"HC..","current_tier1_summ_delivered":24155,"current_tier2_summ_delivered":27963,"linkquality":240,"meter_serial_number":"021961258464","rms_current":4,"rms_current_max":90,"rms_voltage":65535,"schedule_h_p_h_c":0,"start_notice_e_j_p":0,"status_register":"","update":{"installed_version":12,"latest_version":12,"state":"idle"},"update_available":false,"warn_d_p_s":0}'
Info 2023-01-22 12:57:07MQTT publish: topic 'zigbee2mqtt/Linky', payload '{"active_power":-32768,"active_power_max":-32768,"active_register_tier_delivered":"HP..","apparent_power":940,"available_power":60,"current_tarif":"HC..","current_tier1_summ_delivered":24155,"current_tier2_summ_delivered":27963,"linkquality":240,"meter_serial_number":"021961258464","rms_current":4,"rms_current_max":90,"rms_voltage":65535,"schedule_h_p_h_c":0,"start_notice_e_j_p":0,"status_register":"","update":{"installed_version":12,"latest_version":12,"state":"idle"},"update_available":false,"warn_d_p_s":0}'
Info 2023-01-22 12:57:27MQTT publish: topic 'zigbee2mqtt/Linky', payload '{"active_power":-32768,"active_power_max":-32768,"active_register_tier_delivered":"HP..","apparent_power":970,"available_power":60,"current_tarif":"HC..","current_tier1_summ_delivered":24155,"current_tier2_summ_delivered":27963,"linkquality":240,"meter_serial_number":"021961258464","rms_current":4,"rms_current_max":90,"rms_voltage":65535,"schedule_h_p_h_c":0,"start_notice_e_j_p":0,"status_register":"","update":{"installed_version":12,"latest_version":12,"state":"idle"},"update_available":false,"warn_d_p_s":0}'
Info 2023-01-22 12:57:45MQTT publish: topic 'zigbee2mqtt/Linky', payload '{"active_power":-32768,"active_power_max":-32768,"active_register_tier_delivered":"HP..","apparent_power":970,"available_power":60,"current_tarif":"HC..","current_tier1_summ_delivered":24155,"current_tier2_summ_delivered":27963,"linkquality":240,"meter_serial_number":"021961258464","rms_current":8,"rms_current_max":90,"rms_voltage":65535,"schedule_h_p_h_c":0,"start_notice_e_j_p":0,"status_register":"","update":{"installed_version":12,"latest_version":12,"state":"idle"},"update_available":false,"warn_d_p_s":0}'

mwi001, 2023-01-22:
Hello,

I've restarted from an empty recorder database and re-registered my ZLinky device. For the moment it seems to have solved the issue (how ?). So I close the case and will watch the behavior in the next days.

Thanks

DamGranger, 2023-01-24:
Hi,

Even though you closed this, if you don't need it, you can disable the fact that every entity is sent as attributes of every other entity by disabling Home Assistant legacy entity attributes in the Home Assistant Integration parameters page (or in configuration.yaml).

[image]
```
advanced:
  homeassistant_legacy_entity_attributes: false
```