Koenkk/zigbee2mqtt #16304: GIDEALED ZC05M topics different than documentation

Nyarly360, 2023-01-22:
### What happened?

I'm using an GIDERWEL RGBWW strip, which identifies as GIDEALED ZC05M in Z2M. I can control it via the Z2M Web UI and state and brightness via MQTT, but I can't set colors. The [documentation](https://www.zigbee2mqtt.io/devices/ZC05M.html) mentions the topics "color_xy" and "color_hs", but I don't see them in the logs:

> zigbee2mqtt-1.29  | Zigbee2MQTT:info  2023-01-22 17:05:25: MQTT publish: topic 'zigbee2mqtt/Lightband_Stairs/color_mode', payload 'color_temp'
zigbee2mqtt-1.29  | Zigbee2MQTT:info  2023-01-22 17:05:25: MQTT publish: topic 'zigbee2mqtt/Lightband_Stairs/color-x', payload '0.3153'
zigbee2mqtt-1.29  | Zigbee2MQTT:info  2023-01-22 17:05:25: MQTT publish: topic 'zigbee2mqtt/Lightband_Stairs/color-y', payload '0.3253'
zigbee2mqtt-1.29  | Zigbee2MQTT:info  2023-01-22 17:05:25: MQTT publish: topic 'zigbee2mqtt/Lightband_Stairs/color-hue', payload '208'
zigbee2mqtt-1.29  | Zigbee2MQTT:info  2023-01-22 17:05:25: MQTT publish: topic 'zigbee2mqtt/Lightband_Stairs/color_mode', payload 'xy'
zigbee2mqtt-1.29  | Zigbee2MQTT:info  2023-01-22 17:05:25: MQTT publish: topic 'zigbee2mqtt/Lightband_Stairs/color-saturation', payload '7'
zigbee2mqtt-1.29  | Zigbee2MQTT:info  2023-01-22 17:05:25: MQTT publish: topic 'zigbee2mqtt/Lightband_Stairs/color-x', payload '0.6942'
zigbee2mqtt-1.29  | Zigbee2MQTT:info  2023-01-22 17:05:25: MQTT publish: topic 'zigbee2mqtt/Lightband_Stairs/color-y', payload '0.2963'
zigbee2mqtt-1.29  | Zigbee2MQTT:info  2023-01-22 17:05:25: MQTT publish: topic 'zigbee2mqtt/Lightband_Stairs/brightness', payload '160'
zigbee2mqtt-1.29  | Zigbee2MQTT:info  2023-01-22 17:05:25: MQTT publish: topic 'zigbee2mqtt/Lightband_Stairs/color-hue', payload '359'
zigbee2mqtt-1.29  | Zigbee2MQTT:info  2023-01-22 17:05:25: MQTT publish: topic 'zigbee2mqtt/Lightband_Stairs/state', payload 'ON'
zigbee2mqtt-1.29  | Zigbee2MQTT:info  2023-01-22 17:05:25: MQTT publish: topic 'zigbee2mqtt/Lightband_Stairs/color-saturation', payload '100'
zigbee2mqtt-1.29  | Zigbee2MQTT:info  2023-01-22 17:05:25: MQTT publish: topic 'zigbee2mqtt/Lightband_Stairs/color_temp', payload '156.47'
zigbee2mqtt-1.29  | Zigbee2MQTT:info  2023-01-22 17:05:25: MQTT publish: topic 'zigbee2mqtt/Lightband_Stairs/last_seen', payload '2023-01-22T17:05:25+01:00'

If I write something to the topic "color_hs", the log shows messages like the following:

> No converter available for 'color_hs' ("0,0,0")
No converter available for 'color_hs' ("{color:{hsb:239,98,1}}")

It appears that the device doesn't match the documentation. Should Z2M not use the (existing) GIDERWEL strip config? What can I do to help fix this?

### What did you expect to happen?

I expected to be able to control colors via MQTT, not only via the Z2M Web UI.

### How to reproduce it (minimal and precise)

Happens every time.

### Zigbee2MQTT version

1.29.2

### Adapter firmware version

20221102

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P

closed 2023-01-23 by Nyarly360

Nyarly360, 2023-01-23:
I misunderstood the strip's configuration, so I have to try some things first.

sbley, 2023-11-26:
I have exactly the same problem with a Philips Hue Light Strip. Could you please share your solution?

Nyarly360, 2023-12-04:
Hi @sbley, as I stated above I didn't get how mqtt works. If you look at the Z2M device state (or "Status" in German), you can see the json structure. There's no topic called "color_hs" or "color_xy", you have to create a correct json object and put that in the set topic of the respective device. I developed my solution in [this thread](https://community.openhab.org/t/zigbee2mqtt-revisited-no-more-ugly-transformations/86362/210) in the openHAB community forums.

sbley, 2023-12-04:
Thank your your answer. I wish the MQTT Binding in openHAB would integrate better with zigbee2mqtt.
Eventually, I created a solution similar to yours. What needs to be considered, is the value range of the brightness. Which is 0-100 in the Color control, but 0-254 in zigbee2mqtt. This needs to be transformed.