Koenkk/zigbee2mqtt #16309: MQTT error: connect ECONNREFUSED <<IP >ddress>>:1833  - Z2M in a Docker Container

yonz2, 2023-01-22:
### What happened?

Hi,
I'm trying to setup a new instance of Homeassistant + MQTT Broker + Zigbee2MQTT on a fresh Linux box (Ubuntu 22.10 on an X86 box). 

I installed all three images using this docker-compose.yaml file:
```
version: '3.8'
services:
  mosquitto:
    container_name: mosquitto
    image: eclipse-mosquitto:latest
    restart: unless-stopped
    user: 1000:1000
    volumes:
      - /var/local/homeassistant/mosquitto/config:/mosquitto/config
      - /var/local/homeassistant/mosquitto/log/:/mosquitto/log/
      - /var/local/homeassistant/mosquitto/data:/mosquitto/data/
    ports:
      - 1883:1883
      - 9001:9001
    
  zigbee2mqtt:
    container_name: zigbee2mqtt
    image: koenkk/zigbee2mqtt
    user: 1000:1000
    restart: unless-stopped
    volumes:
      - /var/local/homeassistant/zigbee2mqtt/data:/app/data
      - /run/udev:/run/udev:ro
    ports:
      # Frontend port
      - 8080:8080
    environment:
      - TZ=Europe/Berlin
    devices:
      # Make sure this matched your adapter location
      - /dev/ttyACM0:/dev/ttyACM0
    privileged: true

  homeassistant:
    container_name: homeassistant
    image: "ghcr.io/home-assistant/home-assistant:stable"
    user: 1000:1000

    volumes:
      - /var/local/homeassistant/homeassistant/config:/config
      - /etc/localtime:/etc/localtime:ro
    restart: unless-stopped
    privileged: true
    ports:
      - "8123:8123"
```
The containers start, I am able to log on to Home Assistant. I installed the MQTT Integration and I'm able to send and receive messages.

Zigbee2MQTT is configured with exactly the same MQTT Username & Password as used in the test above, but I always get an error message `MQTT error: connect ECONNREFUSED 172.24.0.2:1833` (The Server name from the docker-compose file is properly resolved)

Why can't zigbee2mqtt  connect to the MQTT broker? 
homeassistant, running on the same docker network, can connect, using exactly the same credentials.

Here's my mosquitto configuration file:
```
protocol mqtt
listener 1883

log_dest file /mosquitto/log/mosquitto.log
log_dest stdout
log_type error
log_type warning
log_type notice
log_type information
log_timestamp_format %Y-%m-%d %H:%M:%S

persistence true
persistence_location /data/

allow_anonymous false
password_file /mosquitto/config/password.txt

```
And the Zigbee2MQTT configuration.yaml:
```
homeassistant: true
permit_join: true
mqtt:
  server: mqtt://mosquitto:1833
  user: homeassistant
  password: 86zv8LnmdzvDoIzrN3scVOnEF8tZMbsa5PV5f
serial:
  port: tcp://192.168.178.45:8888
  adapter: ezsp
advanced:
  pan_id: 58560
  ext_pan_id: [221, 221, 221, 221, 221, 221, 221, 221]
  channel: 11
  network_key: [1, 3, 5, 7, 9, 11, 13, 15, 0, 2, 4, 6, 8, 10, 12, 13]
devices:
  '0x001788010b77f5b7':
    friendly_name: '0x001788010b77f5b7'

```

### What did you expect to happen?

Zigbee2mqtt connecting to the mosquitto broker, and being able to connect to the zigbee2mqtt frontend

### How to reproduce it (minimal and precise)

Deply Docker-Compose.yaml file (see above) run `docker-compose up` and observe that zigbee2mqtt does not start

### Zigbee2MQTT version

1.29.2.1

### Adapter firmware version

Frontend does not start

### Adapter

Sonoff ZIGBEEBridge

### Debug log

Here's the log section that is repeated in an endless loop:

```
Using '/app/data' as data directory
Zigbee2MQTT:info  2023-01-22 22:18:26: Logging to console and directory: '/app/data/log/2023-01-22.22-18-26' filename: log.txt
Zigbee2MQTT:info  2023-01-22 22:18:26: Starting Zigbee2MQTT version 1.29.2 (commit #bb3e8f6)
Zigbee2MQTT:info  2023-01-22 22:18:26: Starting zigbee-herdsman (0.14.83-hotfix.0)
Zigbee2MQTT:info  2023-01-22 22:18:31: zigbee-herdsman started (resumed)
Zigbee2MQTT:info  2023-01-22 22:18:31: Coordinator firmware version: '{"meta":{"maintrel":"9 ","majorrel":"6","minorrel":"7","product":8,"revision":"6.7.9.0 build 405"},"type":"EZSP v8"}'
Zigbee2MQTT:info  2023-01-22 22:18:31: Currently 1 devices are joined:
Zigbee2MQTT:info  2023-01-22 22:18:31: 0x001788010b77f5b7 (0x001788010b77f5b7): Not supported (Unknown)
Zigbee2MQTT:warn  2023-01-22 22:18:31: `permit_join` set to  `true` in configuration.yaml.
Zigbee2MQTT:warn  2023-01-22 22:18:31: Allowing new devices to join.
Zigbee2MQTT:warn  2023-01-22 22:18:31: Set `permit_join` to `false` once you joined all devices.
Zigbee2MQTT:info  2023-01-22 22:18:31: Zigbee: allowing new devices to join.
Zigbee2MQTT:info  2023-01-22 22:18:31: Connecting to MQTT server at mqtt://mosquitto:1833
Zigbee2MQTT:error 2023-01-22 22:18:31: MQTT error: connect ECONNREFUSED 172.24.0.2:1833
Zigbee2MQTT:error 2023-01-22 22:18:31: MQTT failed to connect, exiting...
Zigbee2MQTT:info  2023-01-22 22:18:31: Stopping zigbee-herdsman...
Zigbee2MQTT:info  2023-01-22 22:18:32: Stopped zigbee-herdsman
```

HakunaMatmata, 2023-01-22:
Hello,
mosquitto is only exposed locally, so you need to call him on localhost or 127.0.0.1.
However with your configuration you set it via containers, so they are not virtually on the same host.

As a workaround you can set your containers to with nework_mode:host

Here is my docker-compose (working) : 
`version: "3.8"
services:
  mosquitto:
    container_name: mosquitto
    image: eclipse-mosquitto
    restart: always
    network_mode: host
    volumes:
      - /share/Containers/mosquitto/conf:/mosquitto/conf
      - /share/Containers/mosquitto/data:/mosquitto/data
      - /share/Containers/mosquitto/log:/mosquitto/log
      
  zigbee2mqtt:
    container_name: zigbee2mqtt
    image: koenkk/zigbee2mqtt
    restart: always
    volumes:
      - /share/Containers/zigbee2mqtt/data:/app/data
      - /run/udev:/run/udev:ro
    network_mode: host
    environment:
      - TZ=Europe/Paris
    devices:
      - /dev/ttyACM0:/dev/ttyACM0:rwm
    depends_on:
      - mosquitto`

yonz2, 2023-01-22:
Strange ... I did check that mosquitto was listening on all interfaces, when running on the docker bridge network.

But thanks for your hint. now it is connecting. And I got the frontend up as well.

closed 2023-01-22 by yonz2