Koenkk/zigbee2mqtt #16310: [Feature request]: Device should be unavailable of not seen for a specific time

HabelStefan, 2023-01-22:
### Is your feature request related to a problem? Please describe

I would like to have my Hue bulbs report that they are unavailable. The reason is that the are behind a switch, so that i can power them off completely. But this does not make them report that they are unavailable  (Z2M should report it of course).

### Describe the solution you'd like

Set the status of the bulb to unavailable!

### Describe alternatives you've considered

Add an additional sensor that shows how long the device is offline.

### Additional context

Nothing more to add :)

sjorge, 2023-01-23:
This mostly already exists
https://www.zigbee2mqtt.io/guide/configuration/device-availability.html#availability-advanced-configuration to configure marking devices as unavailable if they do not respond.

You can also enable last_seen, https://www.zigbee2mqtt.io/guide/usage/mqtt_topics_and_messages.html#zigbee2mqtt-bridge-request

HabelStefan, 2023-01-23:
Oh did not see that option.
How does this state show in Hassio?

sjorge, 2023-01-23:
I don't use hassio but looking at the code i think it marks the device as unavailable too.

closed 2023-03-02 by github-actions[bot]