Koenkk/zigbee2mqtt #16311: Zigbee functions crash after time ( 0x00158d000428ad57/2 )

NoAdO, 2023-01-23:
### What happened?

Well, i do not know why it happens. The problem lasts about 2 months, now more often than earlier. For me as user it looks like i can't control devices in Home Assistant and even in Control Panel of Z2M so i restart Z2M container or Z2M via web UI and this fixes problem.. but also i loose 2 devices from network every time, always same ones, one of them is ZB Switch Living light mentioned in log. I've turned on debug and catched a bug in [log.txt](https://github.com/Koenkk/zigbee2mqtt/files/10477241/log.txt)
 , main error message is:
```
debug 2023-01-23 00:03:18: Error: Command 0x00158d000428ad57/1 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (sendZclFrameToEndpointInternal error)
    at EZSPAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:465:19)
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
debug 2023-01-23 00:04:02: Saving state to file /app/data/state.json
debug 2023-01-23 00:09:02: Saving state to file /app/data/state.json
debug 2023-01-23 00:14:02: Saving state to file /app/data/state.json
debug 2023-01-23 00:19:02: Saving state to file /app/data/state.json
debug 2023-01-23 00:24:02: Saving state to file /app/data/state.json
debug 2023-01-23 00:29:02: Saving state to file /app/data/state.json
debug 2023-01-23 00:34:02: Saving state to file /app/data/state.json
debug 2023-01-23 00:35:38: Received MQTT message on 'zigbee2mqtt/ZB Switch Living light/set' with data 'OFF'
error 2023-01-23 00:35:39: Entity 'ZB Switch Living light' is unknown
```
After first error Z2M stops to control devices. Same error as first message above repeat while i try to control switch. There's nothing here that can help me understand why this happen. My system in Synology DSM 7 with Docker, Z2M is launched as Docker container with this command:

```sudo docker run \
    -itd \
    --name="HA_Zigbee2MQTT" \
    --net=host \
    -e TZ=Europe/Moscow \
    --device=/dev/ttyUSB0 \
    --restart=always \
    -v /volume1/docker/HA_Zigbee2MQTT/data:/app/data \
    koenkk/zigbee2mqtt:latest \
    ;
```

The problem exists at least at version 1.28.4 and 1.29.2. My stick is [EFR32MG21](https://aliexpress.ru/order-list/5072402542291948) with FW version 7.1.1.0 build 273, MQTT and HA are another Docker Containers.
Z2M configuration:

```
homeassistant: true
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://*.*.*.*:1883
  client_id: zigb2mqtt
  include_device_information: true
  user: mqtt
  password: ***
serial:
  adapter: ezsp
  port: /dev/ttyUSB0
  disable_led: true
frontend:
  port: 8200
devices:
...
...
advanced:
  ikea_ota_use_test_url: false
  log_syslog:
    app_name: Zigbee2MQTT
    eol: /n
    host: localhost
    localhost: localhost
    path: /dev/log
    pid: process.pid
    port: 123
    protocol: tcp4
    type: '5424'
  last_seen: ISO_8601
  log_level: debug
  homeassistant_legacy_entity_attributes: false
  legacy_api: false
  legacy_availability_payload: false
device_options:
  legacy: false
```

What am i doing wrong?

### What did you expect to happen?

Expected continuing to work

### How to reproduce it (minimal and precise)

do not know

### Zigbee2MQTT version

1.29.2

### Adapter firmware version

7.1.1.0 build 273

### Adapter

ZigBee 3,0 Silicon Labs Mini EFR32MG21

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/10477241/log.txt)

NoAdO, 2023-01-23:
Worked about 14 hours. got another death:
```info  2023-01-23 13:59:24: MQTT publish: topic 'zigbee2mqtt/ZB Valve Hot Water', payload '{"device":{"applicationVersion":67,"dateCode":"","friendlyName":"ZB Valve Hot Water","hardwareVersion":1,"ieeeAddr":"0xa4c138c8acb18be6","manufacturerID":4417,"manufacturerName":"_TZ3000_h3noz0a5","model":"ZN231392","networkAddress":50663,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"last_seen":"2023-01-23T10:59:23.444Z","linkquality":176,"power_on_behavior":"previous","state":"ON"}'
info  2023-01-23 13:59:24: MQTT publish: topic 'zigbee2mqtt/ZB Valve Hot Water', payload '{"device":{"applicationVersion":67,"dateCode":"","friendlyName":"ZB Valve Hot Water","hardwareVersion":1,"ieeeAddr":"0xa4c138c8acb18be6","manufacturerID":4417,"manufacturerName":"_TZ3000_h3noz0a5","model":"ZN231392","networkAddress":50663,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"last_seen":"2023-01-23T10:59:23.444Z","linkquality":176,"power_on_behavior":"previous","state":"ON"}'
info  2023-01-23 13:59:24: MQTT publish: topic 'zigbee2mqtt/ZB Valve Hot Water', payload '{"device":{"applicationVersion":67,"dateCode":"","friendlyName":"ZB Valve Hot Water","hardwareVersion":1,"ieeeAddr":"0xa4c138c8acb18be6","manufacturerID":4417,"manufacturerName":"_TZ3000_h3noz0a5","model":"ZN231392","networkAddress":50663,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"last_seen":"2023-01-23T10:59:23.444Z","linkquality":176,"power_on_behavior":"previous","state":"ON"}'
info  2023-01-23 13:59:24: MQTT publish: topic 'zigbee2mqtt/ZB Valve Hot Water', payload '{"device":{"applicationVersion":67,"dateCode":"","friendlyName":"ZB Valve Hot Water","hardwareVersion":1,"ieeeAddr":"0xa4c138c8acb18be6","manufacturerID":4417,"manufacturerName":"_TZ3000_h3noz0a5","model":"ZN231392","networkAddress":50663,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"last_seen":"2023-01-23T10:59:23.444Z","linkquality":176,"power_on_behavior":"previous","state":"ON"}'
info  2023-01-23 13:59:24: MQTT publish: topic 'zigbee2mqtt/ZB Valve Hot Water', payload '{"device":{"applicationVersion":67,"dateCode":"","friendlyName":"ZB Valve Hot Water","hardwareVersion":1,"ieeeAddr":"0xa4c138c8acb18be6","manufacturerID":4417,"manufacturerName":"_TZ3000_h3noz0a5","model":"ZN231392","networkAddress":50663,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"last_seen":"2023-01-23T10:59:23.444Z","linkquality":176,"power_on_behavior":"previous","state":"ON"}'
debug 2023-01-23 13:59:31: Saving state to file /app/data/state.json
debug 2023-01-23 14:04:35: Saving state to file /app/data/state.json
debug 2023-01-23 14:05:36: Received MQTT message on 'zigbee2mqtt/ZB Relay Bath/l1/set' with data 'ON'
debug 2023-01-23 14:05:36: Publishing 'set' 'state' to 'ZB Relay Bath'
error 2023-01-23 14:05:36: Publish 'set' 'state' to 'ZB Relay Bath' failed: 'Error: Command 0x00158d000428ad57/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (sendZclFrameToEndpointInternal error)'
```

NoAdO, 2023-01-23:
Cleaned up all except configuration file, repaired all devices, will look does it fix issue or not.

NoAdO, 2023-01-27:
No effect, worked for 4 days and crashed again today. Error is also the same - working correct, then stop mqtt talking, some saving states till receive message and error while publishing state:
```info  2023-01-27 13:38:25: MQTT publish: topic 'zigbee2mqtt/ZB bath presence', payload '{"cli":" ","detection_delay":0,"device":{"applicationVersion":70,"dateCode":"","friendlyName":"ZB bath presence","hardwareVersion":1,"ieeeAddr":"0xf4b3b1fffec9ba65","manufacturerID":4098,"manufacturerName":"_TZE204_ztc6ggyl","model":"TS0601_smart_human_presence_sensor","networkAddress":50446,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"fading_time":3,"illuminance_lux":136,"last_seen":"2023-01-27T10:38:25.084Z","linkquality":156,"maximum_range":2.6,"minimum_range":0,"presence":true,"radar_sensitivity":7,"self_test":"check_success","target_distance":2.35}'
debug 2023-01-27 13:38:25: Received Zigbee message from 'ZB bath presence', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,235],"type":"Buffer"},"datatype":2,"dp":9}],"seq":7424}' from endpoint 1 with groupID 0
info  2023-01-27 13:38:25: MQTT publish: topic 'zigbee2mqtt/ZB bath presence', payload '{"cli":" ","detection_delay":0,"device":{"applicationVersion":70,"dateCode":"","friendlyName":"ZB bath presence","hardwareVersion":1,"ieeeAddr":"0xf4b3b1fffec9ba65","manufacturerID":4098,"manufacturerName":"_TZE204_ztc6ggyl","model":"TS0601_smart_human_presence_sensor","networkAddress":50446,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"fading_time":3,"illuminance_lux":136,"last_seen":"2023-01-27T10:38:25.286Z","linkquality":152,"maximum_range":2.6,"minimum_range":0,"presence":true,"radar_sensitivity":7,"self_test":"check_success","target_distance":2.35}'
debug 2023-01-27 13:38:25: Received Zigbee message from 'ZB bath presence', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,235],"type":"Buffer"},"datatype":2,"dp":9}],"seq":7424}' from endpoint 1 with groupID 0
info  2023-01-27 13:38:25: MQTT publish: topic 'zigbee2mqtt/ZB bath presence', payload '{"cli":" ","detection_delay":0,"device":{"applicationVersion":70,"dateCode":"","friendlyName":"ZB bath presence","hardwareVersion":1,"ieeeAddr":"0xf4b3b1fffec9ba65","manufacturerID":4098,"manufacturerName":"_TZE204_ztc6ggyl","model":"TS0601_smart_human_presence_sensor","networkAddress":50446,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"fading_time":3,"illuminance_lux":136,"last_seen":"2023-01-27T10:38:25.484Z","linkquality":156,"maximum_range":2.6,"minimum_range":0,"presence":true,"radar_sensitivity":7,"self_test":"check_success","target_distance":2.35}'
debug 2023-01-27 13:39:25: Saving state to file /app/data/state.json
debug 2023-01-27 13:44:25: Saving state to file /app/data/state.json
debug 2023-01-27 13:49:25: Saving state to file /app/data/state.json
debug 2023-01-27 13:54:25: Saving state to file /app/data/state.json
debug 2023-01-27 13:56:02: Received MQTT message on 'zigbee2mqtt/ZB Switch Warehouse light/set' with data 'OFF'
debug 2023-01-27 13:56:03: Publishing 'set' 'state' to 'ZB Switch Warehouse light'
error 2023-01-27 13:56:03: Publish 'set' 'state' to 'ZB Switch Warehouse light' failed: 'Error: Command 0x804b50fffe49f1a8/1 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (sendZclFrameToEndpointInternal error)'
debug 2023-01-27 13:56:03: Error: Command 0x804b50fffe49f1a8/1 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (sendZclFrameToEndpointInternal error)
    at EZSPAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:465:19)
    at processTicksAndRejections (node:internal/process/task_queues:95:5)
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
debug 2023-01-27 13:59:25: Saving state to file /app/data/state.json```

gecko17, 2023-02-02:
Same issue for me, seems to be the same issue as https://github.com/Koenkk/zigbee2mqtt/issues/14827.

NoAdO, 2023-02-02:
@gecko17, it was 5 days till i restarted last time, i did not add switch named as _TZ3000_hafsqare in Z2M in this days and all worked fine. Yesterday i've changed the adaper to Sonoff ZB DongeP with cc2652p chip. I's forking fine about 12 hours now and i suggest this is the ezsp adapter problem (Do you remember, it has an experimental support?). I'll keep an eye on this thread and write again later how is testing with cc2652p.

NoAdO, 2023-02-07:
Well, about 6 days passed, no bugs, no leaving device. Watchtower made an update of Z2M and everything still works correctly. I do not know what's the problem within bseed and efr but there's no reaction to this issue for 2 weeks =( I've found a solution for me - change the adapter. But the problem is not solved.

gecko17, 2023-02-07:
@NoAdO the issue is in zigbee-herdsman library: EZSP has only experimental support. There is one GitHub issue tracking the progress on it: https://github.com/Koenkk/zigbee-herdsman/issues/319.
Issue still seems to be there with Z2M 1.30.0. I reported my latest findings in the issue.

I am considering to downgrade to 1.27.2, but the only option is to manually download the ZIP file for the Home Assistant Add-on and add it as a local addon. Probably I would need to make a backup first and also save the current files of Z2M. Then delete Z2M addon and then install the local version of 1.27.2. I don't know whether this will work and as there is no easier way of downgrading an Add-on, I guess I will live with the connection losses for now.

kirovilya, 2023-02-11:
@gecko17 please check on the latest dev/edge version of z2m, I made some changes for ezsp adapter

closed 2023-03-22 by github-actions[bot]