Koenkk/zigbee2mqtt #16313: OTA Tradfri N2001/N2002'(No image available for imageType '4555')

Berlusconi13, 2023-01-23:
### What happened?

I tried to upgrade Ikea E2001/E2002 to the latest version using Zigbee2mqtt — 1.29.2-1 on HomeAssistant.
The message „Failed to check if update available for 'Tradfri Dimmer' (No image available for imageType '4555')“ appears.
When I open „http://fw.ota.homesmart.ikea.net/feed/version_info.json“ I can see that the respective imagetype is/should be availale
„{"fw_binary_url":"http://fw.ota.homesmart.ikea.net/global/GW1.0/01.19.046/bin/10078031-zingo_kt_styrbar_remote-2.4.5.ota.ota.signed","fw_file_version_LSB":581,"fw_file_version_MSB":0,"fw_filesize":218698,"fw_image_type":4555,"fw_manufacturer_id":4476,"fw_type":2}“

I tried the same with the IKEA test url/server, it‘s the same behaviour.

### What did you expect to happen?

Update of IKEA Tradfri E2001/E2002 device

### Zigbee2MQTT version

1.29.2-1

### Adapter firmware version

20220219

### Adapter

CC2652P (Sonoff EBDongle-P)

rdu70, 2023-02-19:
This firmware is strange and cause some issues.  The dirigera hub itself does not update to this version, only the old hub does it.
I suggest you wait before trying to update to 2.4.5 and stay in 1.0.24 as z2m has still some issue with it : https://github.com/Koenkk/zigbee2mqtt/issues/15725

closed 2023-03-29 by github-actions[bot]

Roukie686868, 2023-04-02:
I have the same issue. The mqtt message for the switch shows there is another version as the latest version

{
  "action": "arrow_left_click",
  "battery": 100,
  "linkquality": 244,
  "update": {
    "installed_version": 65572,
    "latest_version": 581,
    "state": "idle"
  }
}

According zigbee2mqtt is have version 1.0.024 install. When I try to update with zigbee2mqtt I get the following messages:

[image]

Is there a way to force newer firmware to this switch?