Koenkk/zigbee2mqtt #16317: Critical Error: Failed to call 'Frontend' 'stop'

stkr22, 2023-01-23:
### What happened?

Started Zigbee2MQTT with a fresh install. But It keeps crashing tried 1.29.2 and latest-dev. Is related to #16207 but opened a new issue since error is different.

### Zigbee2MQTT version

1.29.2

### Adapter firmware version

0x26720700

### Adapter

ConbeeII

### Debug log

Zigbee2MQTT:warn  2023-01-23 13:38:31: Allowing new devices to join.
Zigbee2MQTT:warn  2023-01-23 13:38:31: Set `permit_join` to `false` once you joined all devices.
Zigbee2MQTT:info  2023-01-23 13:38:31: Zigbee: allowing new devices to join.
Zigbee2MQTT:error 2023-01-23 13:39:50: Failed to call 'Frontend' 'stop' (TypeError: Cannot read properties of undefined (reading 'close')
at /app/lib/extension/frontend.ts:88:60
at new Promise (<anonymous>)
at Frontend.stop (/app/lib/extension/frontend.ts:88:16)
at Controller.callExtensions (/app/lib/controller.ts:316:40)
at Controller.stop (/app/lib/controller.ts:193:9)
at stop (/app/index.js:111:5)
at process.handleQuit (/app/index.js:117:9))
Zigbee2MQTT:error 2023-01-23 13:39:50: Not connected to MQTT server!
Zigbee2MQTT:error 2023-01-23 13:39:50: Cannot send message: topic: 'zigbee2mqtt/bridge/state', payload: 'offline
Zigbee2MQTT:info  2023-01-23 13:39:50: Disconnecting from MQTT server
TypeError: Cannot read properties of undefined (reading 'end')
at MQTT.disconnect (/app/lib/mqtt.ts:129:21)
at Controller.stop (/app/lib/controller.ts:198:9)
at stop (/app/index.js:111:5)
at process.handleQuit (/app/index.js:117:9)

Koenkk, 2023-01-23:
Fixed in https://github.com/Koenkk/zigbee2mqtt/issues/16207 (which will be included in the 1 February release).

closed 2023-01-23 by Koenkk

stkr22, 2023-01-23:
@Koenkk you sure? I tried latest-dev just today and got the error message I posted.

Koenkk, 2023-01-23:
@stkr22 whoops, should be fixed for real now!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)