Koenkk/zigbee2mqtt #16319: Sonoff SNZB-02D Configuration Error on battery

PABHudds, 2023-01-23:
### What happened?

After the Interview configuration I get the following error:
Error 2023-01-23 14:52:20Configure failed: Error: ConfigureReporting 0x30fb10fffe6eaec1/1 genPowerCfg([{"attribute":"batteryVoltage","minimumReportInterval":3600,"maximumReportInterval":62000,"reportableChange":0}], {"sendWhen":"active","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')

Seen on two different devices. Repairing etc. does not change this. Battery shows as N/A or null when enabled in HA.

### What did you expect to happen?

No error

### How to reproduce it (minimal and precise)

Force remove device, re-pair.

### Zigbee2MQTT version

1.29.2-1

### Adapter firmware version

20220219

### Adapter

Sonoff Zigbee 3.0 US Dongle Plus (CC2652P)

hankanman, 2023-01-28:
Can confirm similar behaviour using Home Assistant Yellow built in radio

smh51, 2023-01-28:
Can confirm this issue here also with the same adapter running on hass.io

Koenkk, 2023-01-29:
Will be fixed in the z2m 1 February release.

closed 2023-01-29 by Koenkk

smh51, 2023-02-01:
Since 1.30. My SNZB-02D didn't report any measurement. The battery problem was not fixed. I reverted to 1.29.2

trollix, 2023-05-22:
Zigbee2mqtt version 1.30.4-1 
My SNZB-02D doesn't report well battery level. It's only good on "expose".
[image]
[image]
[image]

sulliwane, 2023-07-28:
[image]
[image]
[image]

unsupported here :( any idea?