Koenkk/zigbee2mqtt #16320: [Feature request]:

DoraDeu, 2023-01-23:
### Is your feature request related to a problem? Please describe

The image of a zigbee device is just provided as a jpg with a white background

### Describe the solution you'd like

Changing the image to a transparent background

### Describe alternatives you've considered

New image

### Additional context

For the device Avatto me167 the following image is provided:
https://www.zigbee2mqtt.io/images/devices/TS0601_thermostat_3.jpg

Unfortunately the background of it is in white color instead of a transparent background.
I would appreciate if the image could saved as a png file with a transparent background.

DoraDeu, 2023-02-23:
I hope this will be realized.

junosuarez, 2023-03-28:
I think this would need support from the device manufacturers to get updated artwork - or an especially enthusiastic community contribution. Anybody know any designers looking to contribute? :)

closed 2023-05-06 by github-actions[bot]