Koenkk/zigbee2mqtt #16333: [Feature request]: Wait on startup for tcp connection

jwillmer, 2023-01-24:
### Is your feature request related to a problem? Please describe

When Zigbee2MQTT starts up with a `tcp://` configured serial port device and it is not accessible due to network issues the app crashes. Same happens if the network goes down while Zigbee2MQTT is running. 

### Describe the solution you'd like

Could we wait for the serial port device to (re)connect instead of hard crashing?

### Describe alternatives you've considered

No good alternative.

### Additional context

Running the official `koenkk/zigbee2mqtt` docker image connected to the SLZB-06 Zigbee adapter. All works without any issue except for when the (wired) network experience outages.

klickfisch, 2023-02-19:
Same Problem here with a poe Coordinator running the ZigStar Firmware. While having a lag in the tcp Connection z2m will Crash and restart.

jwillmer, 2023-03-22:
I still feel like this is a valuable feature. So please keep it open.

junosuarez, 2023-03-24:
I use zigbee2mqtt with a tcp adapter. Crashing on start if the adapter is not available is desirable to me. If this behavior changes, please make it behind a config.

In my case, "wait for connection" is implemented at a higher level - the zigbee2mqtt process crashes, and docker (with `restart: always`) is responsible for trying to restart the process until it succeeds. systemd is another alternative. Typically, it is more appropriate for this type of functionality to be left to a supervisor process rather than implemented in an application. This makes monitoring and restart behavior more consistent, which increases understandability, operability, and reliability.

closed 2023-05-01 by github-actions[bot]