Koenkk/zigbee2mqtt #16334: MOES MS-105 is not working correctly. manufacturer: _TZE200_la2c2uo9

ikarisan, 2023-01-24:
### What happened?

My new "MOES MS-105" dimmer is not workling corretly from the beginning.
After pairing it is recognized as "1 gang smart dimmer", manufacturer "_TZE200_la2c2uo9", modell "TS0601" but something seems to be wrong.

First, it is linked to this: https://www.zigbee2mqtt.io/devices/TS0601_dimmer_1.html 
But my dimmer looks like: https://de.aliexpress.com/item/1005002675746940.html

Not every command is working.

1) Sending "{"power_on_behavior": ""}" to "zigbee2mqtt/dimmer2/get" results in: "No converter available for 'get' 'power_on_behavior' ()"

3) Sending "{"state": "TOGGLE"} to "zigbee2mqtt/dimmer2/set" results in "Publish 'set' 'state' to 'Dimmer2' failed: 'Error: Value 'TOGGLE' is not allowed, expected one of ON,OFF'"

2) Changing the "light_type" has no effect. It is always resetted to "led".

What is working:

"{"state": "ON"}" turns the light on and ""{"state": "OFF"}" turns it off. But it takes ~6 seconds from switching the light on with "state": "ON" till it reaches the full level of 254. Don't know if this is dimmer related or a transition time can be set. 
"{countdown": x}" sets a correct counter in seconds

### What did you expect to happen?

The commands are executed without errors.

### How to reproduce it (minimal and precise)

Pair the dimmer from the AE link with the coordinator and send the command.

### Zigbee2MQTT version

1.29.2-dev commit: de4b7fb

### Adapter firmware version

20201010 (zStack30x)

### Adapter

CC2538 + CC2592

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/10491916/log.txt)

ikarisan, 2023-01-24:
I guess this device at least is missing a custom "TZE200_onoff_brightness" function like the "TS110E_onoff_brightness", to handle the TOGGLE state.

Steini1984, 2023-01-25:
I have some issues with this dimmer -> Related https://github.com/Koenkk/zigbee2mqtt/issues/15804

ikarisan, 2023-01-30:
With this custom conmverter I managed it to switch the light OFF with TOGGLE. But to turn it on I still have to send a state:ON

```
const definition = {
    fingerprint: [
        {
            modelID: 'TS0601',
            manufacturerName: '_TZE200_la2c2uo9',
        },
    ],
    model: 'TS0601_dimmer_1',
    vendor: 'TuYa',
    description: '1-Channel dimmer MS-105Z',

    fromZigbee: [fz.moes_105_dimmer, fz.ignore_basic_report],
    toZigbee: [tz.moes_105_dimmer],
    meta: {turnsOffAtBrightness1: true, multiEndpoint: false},
    configure: async (device, coordinatorEndpoint, logger) => {
        await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
        if (device.getEndpoint(2)) await reporting.bind(device.getEndpoint(2), coordinatorEndpoint, ['genOnOff']);
    },
    exposes: [
      e.light_brightness().setAccess('state', ea.STATE_SET).setAccess('brightness', ea.STATE_SET),
    ],
};
```

ikarisan, 2023-02-16:
Next strange behaviour:

At https://www.zigbee2mqtt.io/devices/TS0601_dimmer_1.html it is stated out that I can request the state of the dimmer by sending {"state": ""} to "zigbee2mqtt/FRIENDLY_NAME/get". But this is not working, too.

The log states "error 2023-02-16 19:24:04: No converter available for 'get' 'state' ()" for this. :(

mkcpr, 2023-03-08:
I can also confirm that I've got the same results exactly as described in the OP.

I also read that you can send a "transition" to the payload, but I will get the error "No converter available". So I think there is something missing or the wrong converter is bounded to the dimmer.

Is there something for the devs that I can debug/test/look into to work out the problem?

closed 2023-04-15 by github-actions[bot]