Koenkk/zigbee2mqtt #16337: Gledopto zigbee controller GL-C-009P constantly loses connection to the network

pjcarly, 2023-01-24:
### What happened?

I was able to connect the device to the network, but it does not stays connected.

During the **Permit Join** fase it joins, and every minute it joins again:

Constantly getting messages: Accepting joining not in blocklist device '0xa4a000000000000'
_(not the actual ID)_

After the **Permit Join** fase is over, availability switches to Offline, and is never seen again.

I tried resetting the device (power cycling or holding reset button), but is only active and available during the Joining fase.

### Zigbee2MQTT version

1.29.2-dev commit

### Adapter firmware version

0x26400500

### Adapter

Conbee I

Bastiencc, 2023-01-25:
Same thing here with `TuYa TS0041` (end device), `Philips 324131092621` (end device) and `Moes ZS-EUB_1gang` (powered one) got this several times every day : 

```
info  2023-01-25 17:34:41: Accepting joining not in blocklist device '0x5c0272fffe0a6b0a'
info  2023-01-25 17:34:41: Accepting joining not in blocklist device '0x5c0272fffe0a6b0a'
info  2023-01-25 17:34:42: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"escalier_switch","ieee_address":"0x5c0272fffe0a6b0a"},"type":"device_announce"}'
info  2023-01-25 17:34:42: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"escalier_switch"},"type":"device_announced"}'

info  2023-01-25 16:56:29: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"entree_plafond"},"type":"device_announced"}'
info  2023-01-25 16:56:29: Accepting joining not in blocklist device '0xa4c13825eec1d2ab'

info  2023-01-25 14:01:08: Accepting joining not in blocklist device '0x001788010408b882'
info  2023-01-25 14:01:09: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"chambre_2_switch_mur","ieee_address":"0x001788010408b882"},"type":"device_announce"}'
```

pjcarly, 2023-02-25:
I can tell it is probably related to the conbee stick I was using.

I recently upgraded to Home Assistent Yellow with an integrated ezsp chip, and was able to pair this device and use it without any issues.

closed 2023-04-05 by github-actions[bot]