Koenkk/zigbee2mqtt #16342: Zigbee2Mqtt Infinite retry loop

bdewacht, 2023-01-25:
### What happened?

Setup:
  - This behavior was first observed when upgrading from 1.25.1 to 1.28.4. Can confirm this is still present in 1.29.2.
  - Zigbee2mqtt combined with eclipse-mosquitto:2.0. Mqtt protocol 5 selected.
  - Paired device is an RGB lamp [Livarno Lux E27 bulb RGB/TS0505B]. Retain is set and retention is set to 60s. 
 
Repro steps:
1. Power on lamp
2. Set color (see example payload below)
3. Power off lamp
4. Repeat 100 times: Set color (see example payload below), wait 10s
5. Power on lamp
6. Set color (see example payload below)

Payload example: 
> MQTT publish: topic 'zigbee2mqtt/0x0c4314fffXXXXXXX', payload '{"brightness":254,"color":{"x":0.5580738435778532,"y":0.434207284959034},"color_mode":"xy","color_temp":519,"last_seen":1674638537051,"linkquality":132,"state":"ON"}'

=> Messages in step 4 seem to be "persisted" and are continuously being resent (see repro steps for more details)

### What did you expect to happen?

Expected result:
In step 2: Lamp changes color
In step 4: Timeout reported
In step 6: Lamp changes color according to latest payload

### How to reproduce it (minimal and precise)

Actual result:
In step 2: Ok
In step 4: Ok
In step 6: Part of the messages "queued" in step 4 are continuously and at fast speed (+/- 15 messages per second) being sent to the lamp. The color set in step 6 is immediately being overwritten by the "queued" messages in step 4. The "queued" messages are not cleared from the system. The resending seems to be infinite.

Workaround:
Restart zigbee2mqtt (emphasis: not the broker, just zigbee2mqtt) fixes the issue: the "queued" messages are discarded and everything works normally.

I can not find any reference in the documentation about queueing multiple messages for the same topic or how to manage that queue. Hence I feel this is a bug.

### Zigbee2MQTT version

1.29.2

### Adapter firmware version

zStack3x0 20210708

### Adapter

Sonoff USB Stick

tsspmq, 2023-01-28:
This happens for me as well, but the only devices I have that I noticed it on was covers.   I wonder if this happens if not using version 5.  That is the only setting I changed in MQTT.  

I use Keen Home vents and have a custom appdaemon app I wrote that moves them quite a bit at times based on where the sun is hitting the house some rooms dont need as much heat while others do.  It does check the level and sends a command every time there is a change in either the vent or room temperature so if one stops responding in a cycle it can send dozens or even more level changes.

If a cover stops responding for some reason, every command that was queued goes through once it wakes up (Usually a remove/reinstall battery).   I just restart Z2M if it's been a while or just let it run through the queued actions.   I'm using version 5 with eclipse and have retain on.  I thought it was just things queued up in MQTT or HA, never really dug into it.

I can imagine that with Adaptive Lighting this can be a problem with bulbs because that thing can send a bunch of changes throughout the day.

bdewacht, 2023-01-29:
I can confirm that there is no link with MQTT version 5. Switching to version 4 has the same behavior.

bdewacht, 2023-03-01:
I can confirm the same behavior with 1.30.1

closed 2023-04-09 by github-actions[bot]