Koenkk/zigbee2mqtt #16343: End device left network after restarting z2m if there're other devices in network

willieee802, 2023-01-25:
### What happened?

1. got one inline switch and some other zigbee devices such as inline dimmer, wall switch, curtain motor. 
2. They get paired successfuly for the first time and are working well. But when I restart z2m, the inline switch always left network immediately, and no debug infos printed about this. Then I have to repair it.
3. However, if I only pair this inline switch, it will stay after a restart. It seems like this inline switch always left if there are other devices in the network.
4. Also, an inline dimmer and a curtain motor staying together won't have such issue, just that inline switch won't stay with others
5. And I can confirm that some time eariler, when it was in about July 2022,  the z2m / zh version around that time, do not produce such problem, at that time I got the same inline switch and curtain motor stay together working well.
6. It seems to me that one major difference between this inline switch and others is that the inline switch always joins as an end device, while others are joining as routers, which might be a clue. This inline switch as end device do not stay with other routers. while the other devices joining as routers can stay together fine.
7. The inline switch model: 
  Zigbee Model
  TS0012
  Zigbee Manufacturer
  _TZ3000_ohnmmnne
**8. After tests with previous versions, I can see this issue happens in zigbee-herdsman [0.14.65](https://github.com/Koenkk/zigbee-herdsman/commit/adfc5984d18e121f5a79474c23ea63e42a3f11b8) but not in versions before it**

### What did you expect to happen?

I expceted that the device will stay without leaving network after restart.

### How to reproduce it (minimal and precise)

Start z2m app, pair the inline switch and another one, say a curtain motor, both working fine. But after a restart of z2m, the inline switch always left network

### Zigbee2MQTT version

1.29.2

### Adapter firmware version

 6.7.8.0 build 373

### Adapter

EZSP V8

### Debug log

I've turned on debug logging, but there's no interesting logs about this behavior, just a warn says device left network

donutVillan, 2023-01-26:
Looks like I might have the same issue [https://github.com/Koenkk/zigbee2mqtt/discussions/16358](url). It at least shows a very similar behaviour. 
@willieee802 do you have to reset the device before re-pair it? I tried now once in the morning but trying to join it without resetting it did nothing.

donutVillan, 2023-01-26:
I also have an EZSP controller (Sonoff ZBDongle-E).

ottoketney, 2023-01-28:
I have similar behaviour with ZBDongle-E

donutVillan, 2023-01-28:
I moved over to the Sonoff ZBDongle-P and that seems to have solved the problem.

As a side note I should mention that when I started deleting the other devices from my network the previously disconnected wall switches started coming back. That was nice since I didn't have to pull them out of wall to reset them then.

willieee802, 2023-02-08:
> I moved over to the Sonoff ZBDongle-P and that seems to have "solved" the problem.
>
> As a side note I should mention that when I started deleting the other devices from my network the previously [...]

Same behaviour, but I need all devices in the network. And also I am not sure what you mean by 'reset before repair'. For my particular inline switch, I just hold the switch button for 10 secs, it will leave network and enter pairing mode, seems like reset and repair are the same thing for me.

donutVillan, 2023-02-08:
> Same behaviour, but I need all devices in the network. And also I am not sure what you mean by 'reset before repair'. For my particular inline switch, I just hold the switch button for 10 secs, it will leave network and enter pairing mode, seems like reset and repair are the same thing for me.

What I was trying to figure out was if the device was actually deleted from the network or if you had to reset the device before including it again.
For me they had not actually left the network so just pairing them again without first doing a reset did not work.
This became even more evident when they started appearing again when I deleted the other devices from the list.

willieee802, 2023-02-08:
> What I was trying to figure out was if the device was actually deleted from the network or if you had to reset the device before including it again. For me they had not actually left the network so [...]

Alright it seems there are indeed some differences between our cases. For my inline switch or wall switch, there's no like 'reset' holes on them, I guess yours got one? so resetting and repairing are different. In my case, When I restart z2m, logs in debug mode clearly logs device left network, and hold 10s to re-enter pairing mode, will have it paired again. But after a restart, same behaviour, it left immediately when z2m is started. But, I also see same behaviou like yours, if I enter-pairing mode or remove the wall switch that never left network, the inline switch joins again, even when permit join is disabled, a restart would not kick inline switch out as long as the wall switch is not joined.

All I did at the moment, was just a roll back to a z2m version that uses zigbee-herdsman 0.14.64, which works fine. Have not tested recent updates though.

willieee802, 2023-02-19:
Can confirm this issue is fixed, after bumping dependency to "zigbee-herdsman": "^0.14.93"
Thanks !

closed 2023-02-19 by willieee802