Koenkk/zigbee2mqtt #16345: Aqara SRTS-A01 persist OTA

leroyloren, 2023-01-25:
### What happened?

with the following commit https://github.com/Koenkk/zigbee-herdsman-converters/pull/5363 the firmware version started to be displayed correctly, but OTA is still offered to upgrade to the same version.

[image]

[image]

### What did you expect to happen?

In the case of the latest version, do not offer an update

### Zigbee2MQTT version

1.29.2-dev commit: a930539

### Adapter firmware version

20221226

### Adapter

sonoff zb dongleP

### Debug log

`Info 2023-01-25 15:21:45MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"Topeni kuchyne","updateAvailable":true},"status":"ok","transaction":"jp40b-7"}'
Debug 2023-01-25 15:21:45Received Zigbee message from 'Topeni pokoj', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":25,"imageType":5017,"manufacturerCode":4447}' from endpoint 1 with groupID 0
Info 2023-01-25 15:21:45MQTT publish: topic 'zigbee2mqtt/Topeni pokoj', payload '{"away_preset_temperature":"5.0","battery":89,"calibrated":true,"child_lock":"UNLOCK","device_temperature":30,"internal_heating_setpoint":21,"last_seen":"2023-01-25T15:21:48+01:00","linkquality":142,"local_temperature":21.3,"occupied_heating_setpoint":21,"power_outage_count":0,"preset":"manual","sensor":"external","setup":false,"system_mode":"heat","update":{"installed_version":25,"latest_version":2329,"state":"available"},"update_available":true,"valve_alarm":false,"valve_detection":"OFF","voltage":2900,"window_detection":"OFF","window_open":false}'
Debug 2023-01-25 15:21:45Got OTA request '{"fieldControl":0,"manufacturerCode":4447,"imageType":5017,"fileVersion":25}'
Debug 2023-01-25 15:21:45ZigbeeOTA: downloaded main index
Debug 2023-01-25 15:21:45Is new image available for '0x54ef4410005ab6b2', current '{"fieldControl":0,"manufacturerCode":4447,"imageType":5017,"fileVersion":25}', latest meta '{"fileVersion":2329,"fileSize":261468,"url":"https://github.com/Koenkk/zigbee-OTA/raw/master/images/Xiaomi/20221122180726_OTA_lumi.airrtc.agl001_0.0.0_0925_20221122_A6C469.ota","sha512":"40e4980fd6518dc32b19f6afd0aa3e49dd38ffa0aee1f2f7b05709cdf90273fb836e35468b6362e9756c63f19293907f50d381c379d5b9921074394e0cef4d62"}'
Debug 2023-01-25 15:21:45Update available for '0x54ef4410005ab6b2': YES
Info 2023-01-25 15:21:45Update available for 'Topeni pokoj'
Info 2023-01-25 15:21:45MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Update available for 'Topeni pokoj'","meta":{"device":"Topeni pokoj","status":"available"},"type":"ota_update"}'
Info 2023-01-25 15:21:45MQTT publish: topic 'zigbee2mqtt/Topeni pokoj', payload '{"away_preset_temperature":"5.0","battery":89,"calibrated":true,"child_lock":"UNLOCK","device_temperature":30,"internal_heating_setpoint":21,"last_seen":"2023-01-25T15:21:48+01:00","linkquality":142,"local_temperature":21.3,"occupied_heating_setpoint":21,"power_outage_count":0,"preset":"manual","sensor":"external","setup":false,"system_mode":"heat","update":{"installed_version":25,"latest_version":2329,"state":"available"},"update_available":true,"valve_alarm":false,"valve_detection":"OFF","voltage":2900,"window_detection":"OFF","window_open":false}'`

DmxLarchey, 2023-02-04:
Same problem here. Update file is loaded but the Aqara valve does not update itself in the end.

nTx-Cheerok, 2023-02-06:
*bump* Hi there, got the same issue.

andrej-hrbcek, 2023-02-08:
[image]
no.2 and 4. already updated, OTA still finding new FW.
all have pending update like this:
[image]

Kh3nsu, 2023-02-11:
Can confirm! This error happens for me too!
I can update from 0.0.0_0025 to 0.0.0_0925 but then after updating it still finds 
a newer one. If I try to update it says succesfull but stays at 0.0.0_0925 and so on.

jzielke84, 2023-02-16:
Yup, same here for three of those.

mendes588, 2023-02-17:
Can confirm this.

OlehNazar, 2023-02-18:
I have the same problem

C4-Dimitri, 2023-02-22:
Problem still persists. I have the same.

Koenkk, 2023-02-23:
Can somebody provide the debug log when doing the OTA check?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

jzielke84, 2023-02-23:
> Can somebody provide the debug log when doing the OTA check?

```shell
2023-02-23T16:24:16.279Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100,254,3,68,128,0,1,27,221]
2023-02-23T16:24:16.279Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100,254,3,68,128,0,1,27,221]
2023-02-23T16:24:16.280Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2023-02-23T16:24:16.280Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2023-02-23T16:24:16.280Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,27,221]
2023-02-23T16:24:16.280Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,27] - 221
2023-02-23T16:24:16.281Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":27}
2023-02-23T16:24:16.281Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2023-02-23T16:24:16.281Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":20063,"destendpoint":1,"srcendpoint":1,"clusterid":25,"transid":28,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[25,20,0,0,100]}}
2023-02-23T16:24:16.281Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,15,36,1,95,78,1,1,25,0,28,0,30,5,25,20,0,0,100,76]
Zigbee2MQTT:debug 2023-02-23 17:24:16: ZigbeeOTA: downloaded main index
Zigbee2MQTT:debug 2023-02-23 17:24:16: Is new image available for '0x54ef4410005bb0e9', current '{"fieldControl":0,"manufacturerCode":4447,"imageType":5017,"fileVersion":25}', latest meta '{"fileVersion":2329,"fileSize":261468,"url":"https://github.com/Koenkk/zigbee-OTA/raw/master/images/Xiaomi/20221122180726_OTA_lumi.airrtc.agl001_0.0.0_0925_20221122_A6C469.ota","sha512":"40e4980fd6518dc32b19f6afd0aa3e49dd38ffa0aee1f2f7b05709cdf90273fb836e35468b63
62e9756c63f19293907f50d381c379d5b9921074394e0cef4d62"}'
Zigbee2MQTT:debug 2023-02-23 17:24:16: Update available for '0x54ef4410005bb0e9': YES
Zigbee2MQTT:info  2023-02-23 17:24:16: Update available for 'Heizung Wohnzimmer'
Zigbee2MQTT:info  2023-02-23 17:24:16: MQTT publish: topic 'zigbee2mqtt/Heizung Wohnzimmer', payload '{"away_preset_temperature":"5.0","battery":100,"calibrated":false,"child_lock":"LOCK","device_temperature":28,"internal_heating_setpoint":25,"linkquality":163,"local_temperature":23.1,"occupied_heating_setpoint":25,"power_outage_count":4,"preset":"manual","schedule":"OFF","schedule_settings":"mon,tue,wed,thu,fri|8:00,24.0|18:00,17.0|23:00,22.0|8:00
,22.0","sensor":"internal","setup":false,"system_mode":"heat","update":{"installed_version":25,"latest_version":2329,"state":"available"},"update_available":null,"valve_alarm":false,"valve_detection":"ON","voltage":3300,"window_detection":"OFF","window_open":false}'
Zigbee2MQTT:info  2023-02-23 17:24:16: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"Heizung Wohnzimmer","updateAvailable":true},"status":"ok","transaction":"821nu-2"}'
```

Koenkk, 2023-02-23:
From the log:
- current file version `25`
- available file version: `2329`

So according to the Zigbee spec an update is available. 

- Are you sure the OTA has been applied? 
- @astappiev are you aware of any issues with this OTA? (https://github.com/Koenkk/zigbee-OTA/pull/195)

leroyloren, 2023-02-23:
I would say the firmware is correct, imho it can't be related to https://github.com/Koenkk/zigbee-herdsman-converters/commit/3055f455778d011d53cdcdb1ef1c800c80fa1b40 after this commit the version started to display correctly, but this issue looks as if the current code version is not updated.

Koenkk, 2023-02-23:
The software build ID you see in the z2m ui does not determine whether an OTA update available. That is the `fileVersion` which is send with the OTA update request from the device. So an incorrect software build ID does not influence the OTA update functionallity.

jzielke84, 2023-02-23:
> From the log:
> 
> * current file version `25`
> * available file version: `2329`
> 
> So according to the Zigbee spec an update is available.
> 
> * Are you sure the OTA has been applied?

yes I’ve watched the update process completing for that device multiple times and I even hear the valve restarting. This happens for all three valves of the same model and brand I own.

Cisneiros, 2023-02-23:
Adding some data that may be useful here: Right after doing an update in one of my TRVs, the state shows as 

```json
    "update": {
        "installed_version": 2329,
        "latest_version": 2329,
        "state": "idle"
    },
```

However, as soon as I click "Check for updates" on the OTA page for the same TRV, the state changes to
```json
    "update": {
        "installed_version": 25,
        "latest_version": 2329,
        "state": "available"
    },
```

jzielke84, 2023-02-24:
Looks like they use a versioning without `.` delimiters. Probably `xxyy` where `xx` is the major version and `yy` the minor version and just `xx` equals `00` for `yy`. Hence `23(.)29` will always be hreater than `25`. But that's just a wild guess..

leroyloren, 2023-03-03:
Hi, after updating to version **0.0.0_0930**, I noticed that an error appears when repairing:

```
2023-03-03 12:36:06zigbee-herdsman-converters:aqara_trv: Unknown key 252 = 0
Warning 2023-03-03 12:36:10Device 'Topeni pokoj' left the network
Warning 2023-03-03 12:37:03zigbee-herdsman-converters:aqara_trv: Unknown key 238 = 2334
Warning 2023-03-03 12:37:06zigbee-herdsman-converters:aqara_trv: Unknown key 238 = 2334
```

Where **2334** is the current versionCode, wouldn't that be the key to success?

ifajardoj, 2023-03-03:
> Adding some data that may be useful here: Right after doing an update in one of my TRVs, the state shows as
>
> ```json
>     "update": {
>         "installed_version": 2329,
>         "latest_version": [...]

Same behavior here.

leroyloren, 2023-03-04:
> Hi, after updating to version **0.0.0_0930**, I noticed that an error appears when repairing:
>
> ```
> 2023-03-03 12:36:06zigbee-herdsman-converters:aqara_trv: Unknown key 252 = 0
> Warning 2023-03-03 [...]

@Koenkk let me ping you, because the topic is getting cluttered. I found another observation about OTA and fileVersion. After the next OTA was published, the fileVersion changed from 25 to 30

`Is new image available for '0x54ef4410005accc', current '{"fieldControl":0,"manufacturerCode":4447,"imageType":5017,"fileVersion":30}', latest meta '{"fileVersion":2334,"fileSize":263916,"url":"https://cdn.aqara.com/cdn/opencloud-product/mainland/product-firmware/prd/lumi.airrtc.agl001/20230221164442_OTA_lumi.airrtc.agl001_0.0.0_0930_20230220_0DE632.ota","sha512":"e7d4d0fdbe1d6cea2b1d3c292061fa05902a625c2cc9aaee436c6a7d5996dce4c2725e3fa16939e6d32fba6f87ee509b87c87f2dbd7e01e15a2da13f9f812023"}'`

Would it be possible to use key 238, which contains the current fileVersion to read?

Koenkk, 2023-03-05:
@leroyloren nice!

I've did an attempt to fix the issue, please check if it works. Before checking for OTA you may need to wait a couple of hours.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

leroyloren, 2023-03-06:
After re-pairing now Ok, thx.

[image]

Koenkk, 2023-03-06:
Great!

closed 2023-03-06 by Koenkk

xsstaff, 2023-03-06:
Updated Z2M, updated firmware and everything is fine now! Thanks! :)

xSirrioNx, 2023-03-22:
@Koenkk Hi!
When u plan to release it? )

Koenkk, 2023-03-22:
Next release will be on 1 April (no joke)

kzajac83, 2023-03-22:
Also confirm, working on Edge.

leroyloren, 2023-03-23:
Maybe fix the typo with the leading zero and it will be perfect :-)

0.0.0_**0**1030

[image]

deniskozlovzaz, 2023-03-26:
Dont UPGRADE! It can make a "brick" from your device... I have firmware  Feb 20 2023 | 0.0.0_01030, but button "update" was active, I try to make update and in result my version was "downgraded?!?!" to Feb 20 2023 | 0.0.0_0030 and my device stop to work it just stuck in F1.1 mode(start calibration initiation). I try to re-pair or reset. nothing helps it just stuck in F1.1, for several seconds after 3pushes(calibarion) it shows main screen with temp, but then return F1.1 again......

I was using last stable build when upgrading, I change to latest-dev, but nothing changed...

[image]

DenDeXTeR51, 2023-03-26:
Dont UPGRADE! It can make a "brick" from your device... I have firmware  Feb 20 2023 | 0.0.0_01030, but button "update" was active, I try to make update and in result my version was "downgraded?!?!" to Feb 20 2023 | 0.0.0_0030 and my device stop to work it just stuck in F1.1 mode(start calibration initiation). I try to re-pair or reset. nothing helps it just stuck in F1.1, for several seconds after 3pushes(calibarion) it shows main screen with temp, but then return F1.1 again......

[image]

DenDeXTeR51, 2023-03-27:
On "latest-dev" updated all 8 SRTS-A01 , wait 1 hour after upgrade and version changed from 0030 on 01030. Had only 1 faulty E1 that somehow bricks after upgrade(on current z2m stable firmware)...

ShadowRep, 2023-03-28:
Just a heads up for you @DenDeXTeR51:
When the battery in my Aqara started to run low (it actually showed about 20-30% left, if I remember correctly) it started to behave strangely. It would show the F11 screen, calibrate and then after a few seconds (probably when it tried to regulate the temperature) it would show F11 again.

So what happened was that it was too low on energy to work and when it tried, it simply rebootet, forgetting all its settings.

New batteries solved the problem. Maybe your problem is similar?
It might be worth to try out new batteries, regardless of the indicated battery status.

DaN66O, 2023-04-03:
After upgrading to the latest version 1.30.3 I can't update from  the 0.0.0_0825
There are errors in the logs
`Error 2023-04-03 14:42:14Failed to check if update available for 'thermo2' (Did not receive a current fileVersion from 'lumi.airrtc.agl001' yet, please wait some hours and try again)`

Neokeek, 2023-04-03:
Same Error here

Unloadable, 2024-04-19:
My Aqara SRTS-A01 has also been updated to firmware version 0.0.0_1030 (Feb 20 2023). The update took a while but went well in principle. However, the device now has a problem: the push button can no longer be pressed; it seems to be pressing against the housing from the inside. When I remove the housing, it is not pressed in.

I have tried calibrating it several times, but this did not fix the problem.

Has anyone else experienced this issue and do you have any suggestions on how to resolve it?

mkdev-de, 2024-06-19:
> I wonder if this is a result of the update or if there's something mechanical going on with my unit.

In my case, there was a mechanical problem inside. The screw mount was broken and it was pressing on the buttons.
Check if you can lift the front (after removing the case so you can access the batteries) on one side.

imho failure of design having 3 mini screws in mini pieces of plastic pressing against a valve...

jzielke84, 2024-06-19:
> In my case, there was a mechanical problem inside. The screw mount was broken and it was pressing on the buttons. Check if you can lift the front (after removing the case so you can access the [...]

I can confirm the mechanical issue. Exchanged that thermostat with one from Hama.