Koenkk/zigbee2mqtt #16348: Paulmann 501.34 dimming not working

IngoK1, 2023-01-25:
### What happened?

Dimming feature of the Paulmann switch 501.34 does not work.

Pairing works as designed and no error messages appear. 
The on/off action values: on_1, off_1, on_2, off_2 work as expected
Binding of the on/off functions also work as designed

But dimming does not work.  In the documentation there are the following values mentioned for the dimming actions: brightness_move_up_1, brightness_move_down_1, brightness_move_stop_1, brightness_move_up_2, brightness_move_down_2, brightness_move_stop_2
These do not work or are not beeing seen. On the device a green LED indicated the dimming commands are send but they do not appear in z2m.

### What did you expect to happen?

The dimming action values should appear in zigbee2mqtt as described in the documentation.

### How to reproduce it (minimal and precise)

Pair the device to zigbee2mqtt and short press a button to see if the on/off comand appears. Then long press a button. See the gren LED on the device appear indicating the command has been send. Check if the command appears on Zigbee2MQTT

### Zigbee2MQTT version

1.29.2 commit: bb3e8f6

### Adapter firmware version

6.10.3.0 build 297

### Adapter

QinHeng Electronics SONOFF Zigbee 3.0 USB Dongle Plus V2

IngoK1, 2023-01-25:
Hello,

I could identify the problem, please add the entries

 fz.command_move, fz.command_stop

in the fromZigbee section of the  zigbeeModel: ['501.34'] section of the devices/paulman.js

Then all works as expected.

Thank you

Ingo

Koenkk, 2023-01-26:
Integrated the fix, thanks!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)