Koenkk/zigbee2mqtt #16350: Zigbee host adapter stops responding after a while

watercrossing, 2023-01-25:
### What happened?

After several hours of normal use, Z2M suddenly receives no more messages from any devices. No errors appear, the devices appear to have just stopped talking to Z2M. 

When doing a network scan, the following error message appears:
```
info  2023-01-25 16:33:55: Starting network scan (includeRoutes 'false')
error 2023-01-25 16:34:01: Failed to execute LQI for 'Coordinator'
debug 2023-01-25 16:34:01: Error: SREQ '--> ZDO - mgmtLqiReq - {"dstaddr":0,"startindex":0}' failed with status '(0xc2: NWK_INVALID_REQUEST)' (expected '(0x00: SUCCESS)')
    at Object.func (/srv/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:322:27)
    at Queue.executeNext (/srv/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)

```

Restarting Z2M (not even the pi) fixes the issues for a few hours. After restarting network scans succeed.

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

20220219

### Adapter

CC2652P (https://www.aliexpress.com/item/1005004443820263.html)

watercrossing, 2023-01-26:
Hmm, this error also occurs when setting `permit_join`
```
debug 2023-01-26 14:45:00: Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data 'false'
info  2023-01-26 14:45:00: Zigbee: disabling joining new devices.
error 2023-01-26 14:45:00: Request 'zigbee2mqtt/bridge/request/permit_join' failed with error: 'SREQ '--> AF - dataRequestExt - {"dstaddrmode":2,"dstaddr":"0x000000000000fffd","destendpoint":242,"dstpanid":0,"srcendpoint":242,"clusterid":33,"transid":8,"options":0,"radius":30,"len":6,"data":{"type":"Buffer","data":[25,9,2,10,0,0]}}' failed with status '(0x02: INVALID_PARAM)' (expected '(0x00: SUCCESS)')'
debug 2023-01-26 14:45:00: Error: SREQ '--> AF - dataRequestExt - {"dstaddrmode":2,"dstaddr":"0x000000000000fffd","destendpoint":242,"dstpanid":0,"srcendpoint":242,"clusterid":33,"transid":8,"options":0,"radius":30,"len":6,"data":{"type":"Buffer","data":[25,9,2,10,0,0]}}' failed with status '(0x02: INVALID_PARAM)' (expected '(0x00: SUCCESS)')
    at Object.func (/srv/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:322:27)
    at Queue.executeNext (/srv/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
```

jsvensson, 2023-01-30:
I'm having the same issue, Z2M v1.29.2. I'm running it in Kubernetes on my little Raspberry cluster, been trying to figure out if that might be the cause or something. Eventually Z2M just seems to stop communicating with anything, until I delete the pod and a new one fires up. this can happen within 3-4 hours, or as much as 24h.

I haven't set up proper data persistence yet so unfortunately I can't provide any decent logs at the moment, I'll look into that.

pagaille, 2023-02-12:
More or less same here, running as a docker container. Frontend seems crashed (port closed), nothing lives except the "saving state to file" event every now and then. 
Interestingly in this case it happened just before the log file change.

Log file 1 : 

```
debug 2023-02-12 00:55:50: Received Zigbee message from 'Coordinator', type 'commandNotification', cluster 'greenPower', data '{"data":[25,197,2,11,254,0],"type":"Buffer"}' from endpoint 242 with groupID null, ignoring since it is from coordinator
debug 2023-02-12 00:58:51: Saving state to file /app/data/state.json
debug 2023-02-12 00:59:10: Received Zigbee message from 'Coordinator', type 'commandNotification', cluster 'greenPower', data '{"data":[25,198,2,11,254,0],"type":"Buffer"}' from endpoint 242 with groupID null, ignoring since it is from coordinator
debug 2023-02-12 01:02:30: Received Zigbee message from 'Coordinator', type 'commandNotification', cluster 'greenPower', data '{"data":[25,199,2,11,254,0],"type":"Buffer"}' from endpoint 242 with groupID null, ignoring since it is from coordinator
debug 2023-02-12 01:03:51: Saving state to file /app/data/state.json
debug 2023-02-12 01:05:50: Received Zigbee message from 'Coordinator', type 'commandNotification', cluster 'greenPower', data '{"data":[25,200,2,11,254,0],"type":"Buffer"}' from endpoint 242 with groupID null, ignoring since it is from coordinator
debug 2023-02-12 01:08:51: Saving state to file /app/data/state.json
debug 2023-02-12 01:09:10: Received Zigbee message from 'Coordinator', type 'commandNotification', cluster 'greenPower', data '{"data":[25,201,2,11,254,0],"type":"Buffer"}' from endpoint 242 with groupID null, ignoring since it is from coordinator
debug 2023-02-12 01:12:30: Received Zigbee message from 'Coordinator', type 'commandNotification', cluster 'greenPower', data '{"data":[25,202,2,11,254,0],"type":"Buffer"}' from endpoint 242 with groupID null, ignoring since it is from coordinator
debug 2023-02-12 01:13:51: Saving state to file /app/data/state.json
debug 2023-02-12 01:15:50: Received Zigbee message from 'Coordinator', type 'commandNotification', cluster 'greenPower', data '{"data":[25,203,2,11,254,0],"type":"Buffer"}' from endpoint 242 with groupID null, ignoring since it is from coordinator
debug 2023-02-12 01:18:49: Received Zigbee message from 'Télécommande cuisine', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":55}' from endpoint 1 with groupID 0
info  2023-02-12 01:18:49: MQTT publish: topic 'zigbee2mqtt/Télécommande cuisine', payload '{"battery":55,"linkquality":112,"update":{"installed_version":65572,"latest_version":581,"state":"idle"}}'
debug 2023-02-12 01:18:51: Saving state to file /app/data/state.json
debug 2023-02-12 01:19:10: Received Zigbee message from 'Coordinator', type 'commandNotification', cluster 'greenPower', data '{"data":[25,204,2,11,254,0],"type":"Buffer"}' from endpoint 242 with groupID null, ignoring since it is from coordinator
debug 2023-02-12 01:22:30: Received Zigbee message from 'Coordinator', type 'commandNotification', cluster 'greenPower', data '{"data":[25,205,2,11,254,0],"type":"Buffer"}' from endpoint 242 with groupID null, ignoring since it is from coordinator
debug 2023-02-12 01:23:51: Saving state to file /app/data/state.json
debug 2023-02-12 01:25:15: Saving state to file /app/data/state.json
```

next log file :

```
root@mattlabs:~# cat /mnt/timemachine/zigbee2mqtt/log/2023-02-12.
2023-02-12.01-28-57/ 2023-02-12.10-41-49/ 
root@mattlabs:~# cat /mnt/timemachine/zigbee2mqtt/log/2023-02-12.01-28-57/log.txt 
debug 2023-02-12 01:29:01: Loaded state from file /app/data/state.json
info  2023-02-12 01:29:01: Logging to console and directory: '/app/data/log/2023-02-12.01-28-57' filename: log.txt
debug 2023-02-12 01:29:01: Removing old log directory '/app/data/log/2023-02-05.17-19-22'
info  2023-02-12 01:29:02: Starting Zigbee2MQTT version 1.30.0 (commit #8f781db)
info  2023-02-12 01:29:02: Starting zigbee-herdsman (0.14.89)
debug 2023-02-12 01:29:02: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/app/data/coordinator_backup.json","databaseBackupPath":"/app/data/database.db.backup","databasePath":"/app/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyACM0"}}'
debug 2023-02-12 01:34:03: Saving state to file /app/data/state.json
debug 2023-02-12 01:39:03: Saving state to file /app/data/state.json
debug 2023-02-12 01:44:03: Saving state to file /app/data/state.json
debug 2023-02-12 01:49:03: Saving state to file /app/data/state.json
debug 2023-02-12 01:54:03: Saving state to file /app/data/state.json
debug 2023-02-12 01:59:03: Saving state to file /app/data/state.json
debug 2023-02-12 02:04:03: Saving state to file /app/data/state.json
debug 2023-02-12 02:09:03: Saving state to file /app/data/state.json
debug 2023-02-12 02:14:03: Saving state to file /app/data/state.json
debug 2023-02-12 02:19:03: Saving state to file /app/data/state.json
debug 2023-02-12 02:24:04: Saving state to file /app/data/state.json
debug 2023-02-12 02:29:04: Saving state to file /app/data/state.json
```

and son on until I restart the container. Does it ring a bell to anybody ?

OlegKarasik, 2023-02-13:
I am having exactly the same issue periodically. Tried to find steps to reproduce but issue looks rather random in nature.

pagaille, 2023-03-01:
I installed the dev version of the docker image and it fixed the problem on my side.

OlegKarasik, 2023-03-01:
I also can confirm that latest dev version have fixed the issue for me (tested for three days now).

watercrossing, 2023-03-02:
No, updating to 1.30.2 (which came out yesterday) does not fix it for me. After 2h the adapter freezes just as before:

```
Request 'zigbee2mqtt/bridge/request/permit_join' failed with error: 'SREQ '--> AF - dataRequestExt - {"dstaddrmode":2,"d staddr":"0x000000000000fffd","destendpoint":242,"dstpanid":0,"srcendpoint":242,"clusterid":33,"transid":65,"options":0,"radius":30,"len":6,"data":{ "type":"Buffer","data":[25,46,2,10,0,0]}}' failed with status '(0x02: INVALID_PARAM)' (expected '(0x00: SUCCESS)')'
```

BTW, I am using the homeassistant script below to attempt to set `permit_join` to false every couple of minutes. If it fails, I restart zigbee2mqtt automatically.
```
automation:
  - id: zigbee2mqtt_auto_restart
    alias: Restart Zigbee2MQTT if setting permit_join fails
    trigger:
      platform: mqtt
      topic: 'zigbee2mqtt/bridge/logging'
    condition:
      condition: template
      value_template: '{{trigger.payload_json.message.startswith("Request ''zigbee2mqtt/bridge/request/permit_join'' failed with error: ''SREQ ''--> AF - dataRequestExt")}}'
    action:
      service: script.turn_on
      target:
        entity_id: script.restart_zigbee2mqtt

  - id: zigbee2mqtt_regular_permit_off
    alias: Regularly turn permit join off
    trigger:
      platform: time_pattern
      minutes: "/3"
    condition:
      - condition: state
        entity_id: timer.zigbee_permit_join
        state: "idle"
    action:
      - service: mqtt.publish
        data:
          payload: "false"
          topic: "zigbee2mqtt/bridge/request/permit_join"

script:
  restart_zigbee2mqtt:
    alias: Restart Zigbee2MQTT
    sequence:
      service: shell_command.restart_zigbee2mqtt

shell_command:
  # Requires sudoers file to be set up
  restart_zigbee2mqtt: /usr/bin/sudo /usr/bin/systemctl restart zigbee2mqtt.service
```

```

OlegKarasik, 2023-04-08:
After updating to 1.30.3 the issue resumed again.

closed 2023-05-17 by github-actions[bot]