Koenkk/zigbee2mqtt #16353: Problem using my aqara light with my aqara switch

Mcyavel, 2023-01-25:
### What happened?

I installed an aqara bulb and an aqara switch without neutral in a room. I have a conbee 2 bridge and am using zigbee2mqtt. The switch is in decoupled mode.
I created 2 automatisms: 1 which turns off the bulb when I click on the switch if the bulb is on and 1 other which turns on the bulb if it is off. I use the single event.

Or when I press the button and the bulb is on, it goes out fine.
On the other hand, when I then want to turn on the light bulb again using my second automatic mechanism, this one does not work by looking at the logs, it is as if the switch was disconnected.

When I turn the bulb back on via the interface, I hear the relay click and the switch turns off, turning off the bulb at the same time.

Did I do something wrong I don't understand why I have this behavior ?

### Zigbee2MQTT version

1.29.2-1

### Adapter firmware version

0x26580700

### Adapter

conbee 2

galdaka, 2023-01-29:
I have Xiaomi aqara ws-euk01 and ws-euk02. The two switches not work in decoupled mode over Zigbee2mqtt with CC2652P dongle.

Mcyavel, 2023-01-30:
I tested with another bulb (a philip hue one) and it's work fine. So i supposed that is the aqara bulb that don't let enough power through the wire when the bulb is off.

closed 2023-03-09 by github-actions[bot]