Koenkk/zigbee2mqtt #16356: [New device support]:

halpz, 2023-01-25:
### Link

https://www.zigbee2mqtt.io/devices/SNZB-01.html

### Database entry

0x00124b0028928a7a

### Comments

I have paired the device and I can see button presses in the Z2M logs, but they do not appear in Hombridge logs. 

I see entries like:
`MQTT publish: topic 'zigbee2mqtt/sonoff_button', payload '{"action":"single","battery":100,"linkquality":144,"voltage":3300}'`

I can see the device in the accessories tab of homebridge, but it does not appear in HomeKit.

There is no way for me to set up an automation using the button, because it's inaccessible to me

Koenkk, 2023-01-26:
This an issue in the HomeKit <-> Z2M integration, not Z2M itself. Please create an issue there.

halpz, 2023-01-26:
it actually seems to be zigbee2mqtt.

if i leave it for a few hours, it will reconnect as a new accessory. and say interview failed

[image]

halpz, 2023-01-26:
@Koenkk

halpz, 2023-01-26:
[image]

Koenkk, 2023-01-29:
- What coordinator are you using?
- What z2m version are you using? Make sure to use 1.29.2 as this has a fix for SONOFF devices disconnecting. After pairing it the configure has to succeed (the yellow refresh button from your screenshot, you can force it by clicking there). If it fails, make sure to wake up the device right before pressing it.

halpz, 2023-01-30:
thanks, latest z2m version seems to fix this