Koenkk/zigbee2mqtt #16360: Aqara Cube (MFKZQ01LM) action_angle error with Home Assistant 2023.2b0

RobertD502, 2023-01-26:
### What happened?

As of Home Assistant 2023.2b0, the action_angle missing a device class or state class is resulting in Home Assistant complaining.

Home Assistant Log:

```
Sensor sensor.bedroom_cube_action_angle has device class None, state class None and unit ° thus indicating it has a numeric value; however, it has the non-numeric value: None (<class 'str'>); Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
```

### Zigbee2MQTT version

1.29.2

### Adapter firmware version

20200925

### Adapter

Electrolama

snp88, 2023-03-09:
Remove stale tag

RobertD502, 2023-03-09:
Re-enabled the sensor in 2023.2 - error is no longer being logged. Closing this issue as it has been resolved.

closed 2023-03-09 by RobertD502