Koenkk/zigbee2mqtt #16364: [New device support]: WDE002388 Schneider Electric

mstjerna, 2023-01-26:
### Link

https://www.se.com/se/sv/download/document/WDE002388_ROHS_CHINA_DECLARATION/

### Database entry

{"id":31,"type":"Router","ieeeAddr":"0x50325ffffe4d2c4b","nwkAddr":51431,"manufId":4190,"manufName":"Schneider Electric","powerSource":"Mains (single phase)","modelId":"NHROTARY/DIMMER/1","epList":[21,242,3],"endpoints":{"3":{"profId":260,"epId":3,"devId":257,"inClusterList":[0,3,4,5,6,8,769,2821],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"21":{"profId":260,"epId":21,"devId":260,"inClusterList":[0,3,2821,65303],"outClusterList":[3,4,5,6,8,258],"clusters":{"genBasic":{"attributes":{"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":6,"hwVersion":1,"dateCode":"","swBuildId":"002.004.015 R","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1674721731538,"defaultSendRequestWhen":"immediate"}

### Comments

It's connecting fine and reports back the status and current dim-value but I can't trigger on/off event from Zigbee2Mqtt or change anything on the device.
It's identifying itself as a WDE002334 instead of a WDE002388.
I hope this would be an easy fix to add the triggering event towards the device.

Nothing is as said happening to the device but the logfiles are putting out this:

Info 2023-01-26 10:23:43MQTT publish: topic 'zigbee2mqtt/0x50325ffffe4d2c4b', payload '{"ballast_maximum_level":null,"ballast_minimum_level":null,"brightness":220,"dimmer_mode":null,"last_seen":"2023-01-26T09:23:44.349Z","linkquality":40,"state":"ON"}'

Info 2023-01-26 10:24:19MQTT publish: topic 'zigbee2mqtt/0x50325ffffe4d2c4b', payload '{"ballast_maximum_level":null,"ballast_minimum_level":null,"brightness":220,"dimmer_mode":null,"last_seen":"2023-01-26T09:24:20.483Z","linkquality":36,"state":"OFF"}'

Error 2023-01-26 10:26:21Publish 'get' 'brightness' to '0x50325ffffe4d2c4b' failed: 'Error: Read 0x50325ffffe4d2c4b/21 genLevelCtrl(["currentLevel"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'

### Supported color modes

Dimmer value 0-254

bozman2021, 2023-02-05:
Got the same problem, would like to have this added.

mstjerna, 2023-02-06:
> Got the same problem, would like to have this added.

I updated to the latest version 1.30.0 and it's now supported. :-)

joggs, 2023-02-22:
Hi, I got the same problem as you, that it identifies as 2344, even though i have z2m 1.30.1.1
Did it automatically idetify correctly for you? Is it possible to manually change to 2388?

joggs, 2023-02-22:
WDE002388 is not in the list over compatible devices. Please add support for it

joggs, 2023-02-24:
I created an external converter skeleton with the  right model and vendor and restarted and I can now control the WDE002388. Not sure if it is my skeleton or just the retart that made the difference. The only thing that does not work is current_level_startup , as the dimmer always starts on max brightness, which is annoying.

I just added this as a js-file:
`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['lumi.sens'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'WDE002388', // Vendor model number, look on the device for a model number
    vendor: 'Schneider Electric', // Vendor of the device (only used for documentation and startup logging)
    description: 'Rotary dimmer', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [], // We will add this later
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
    exposes: [e.battery(), e.temperature(), e.humidity()], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
};

module.exports = definition;`

One example from the log:   (**"current_level_startup":"previous"** does not work ) . Great if you could fix that as I am not sure how to do that

_Info 2023-02-24 20:23:00MQTT publish: topic 'zigbee2mqtt/Schneider2388', payload '{"ballast_lamp_alarm_lamp_burn_hours":false,"ballast_maximum_level":237,"ballast_minimum_level":6,"ballast_status_lamp_failure":false,"ballast_status_non_operational":false,"brightness":24,"dimmer_mode":"auto","last_seen":"2023-02-24T20:23:00+01:00","level_config":{"current_level_startup":"previous","execute_if_off":false,"on_level":254},"linkquality":240,"state":"ON"}'_

bozman2021, 2023-02-25:
It works fine for me with the 1.30.1-1 version, it's stills recognise the unit as [WDE002334] but after doing a couple reconfiguring it works fine.
But the behavior is the same, turn on and then it's max brightness.

joggs, 2023-03-12:
Still relevant.
To summarize: It should be working without reconfiguring and the max brightness is still anissue.

Xaqumuti, 2023-03-24:
Hi,
In case it helps, you can download the Zigbee attributes specification for Schneider Electric / Merten devices from here:
https://shop.exchange.se.com/en-US/apps/103010/zigbee-interface-and-behaviour-documentation-for-schneider-electric-connected-devices
Only e-mail registration needed.

There's an attribute to set the Max and Min brightness levels. See page 19 in the document attached
Cheers
[ZB Spec - Rotary Dimmer - 110422.pdf](https://github.com/Koenkk/zigbee2mqtt/files/11062664/ZB.Spec.-.Rotary.Dimmer.-.110422.pdf)

Caine72, 2023-04-23:
I have this dimmer as well. And I was actually fooled that it was identified correctly as most things seemed to work.

But as I noticed the min/max ballast i started digging around.

Anyone else having that same issue?

joggs, 2023-04-24:
I suppose everyone has this problem as it is not implemented.
Schneider is the number one brand and the brand of choice for professionals
in big parts of Europe so I think it is important that this device works as
expected.

Caine72, 2023-04-24:
Starting to play around with this a bit.
Here is the converter for WDE002334 rebranded as WDE002338
I'll see if I can figure out more. ( 100% beginner when it comes to this)
But between kids and work will not promise anything.

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['NHROTARY/DIMMER/1'],
    model: 'WDE002338',
    vendor: 'Schneider Electric',
    description: 'Rotary dimmer',
    fromZigbee: [fz.on_off, fz.brightness, fz.level_config, fz.wiser_lighting_ballast_configuration],
    toZigbee: [tz.light_onoff_brightness, tz.level_config, tz.ballast_config, tz.wiser_dimmer_mode],
    exposes: [e.light_brightness().withLevelConfig(),
        exposes.numeric('ballast_minimum_level', ea.ALL).withValueMin(1).withValueMax(254)
            .withDescription('Specifies the minimum light output of the ballast'),
        exposes.numeric('ballast_maximum_level', ea.ALL).withValueMin(1).withValueMax(254)
            .withDescription('Specifies the maximum light output of the ballast'),
        exposes.enum('dimmer_mode', ea.ALL, ['auto', 'rc', 'rl', 'rl_led'])
            .withDescription('Sets dimming mode to autodetect or fixed RC/RL/RL_LED mode (max load is reduced in RL_LED)')],
    configure: async (device, coordinatorEndpoint, logger) => {
                const endpoint = device.getEndpoint(3);
                await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl', 'lightingBallastCfg']);
                await reporting.onOff(endpoint);
                await reporting.brightness(endpoint);
            }
};

module.exports = definition;
`

Caine72, 2023-04-24:
Some end of the day observations, via the dev console.

genLevelCtrl, onLevel
 - seems to set the starting level. 
 
genLevelCtrl, startUpCurrentLevel
- set:able but doing so it seems to do nothing.
- Represented in UI as current_level_startup

The transition values all return unsupported_attribute.

lightingBallastCfg, min/max-Level
- can be read, can be set. 
  Enforces nothing.

Caine72, 2023-05-11:
> Hi, In case it helps, you can download the Zigbee attributes specification for Schneider Electric / Merten devices from here: [...]

> Hi, In case it helps, you can download the Zigbee attributes specification for Schneider Electric / Merten devices from here: [...]

Seems to be an issue with the pdf. It cannot be opened.

Caine72, 2023-05-11:
Learning stuff.

> StartUpCurrentLevel
> 
> Specifies the initial level to be applied after **reboot**. When this attribute is set to the invalid value (0xFF), the light will return to the previously active level (before power was cut), when it is turned on again. Otherwise the current level will be set to the value specified here subject to range limitations imposed by the ballast configuration cluster on this endpoint.

Did not know this only applied to reboot of the device. OK!

> OnLevel
> Specifies the level that shall be applied, when an on/toggle command causes the light to turn on. When this attribute is set to the invalid value (0xFF), the light will return to the previously active level (before it was turned off), when it is turned on again.

So we need that one visible.. at least.
Will publish the new extension when i figured out how to do that.... and manage to resync it to z2m again ..

Caine72, 2023-05-12:
```
        exposes.numeric('on_level',ea.ALL).withValueMin(1).withValueMax(254)
            .withDescription('Specifies the Start Up level of the light'),
```
Tried adding this to the expose list. But .. it only says that it's missing the converter when trying to set it.
Though I can see that it is on_level is part of the tz and fz files.
Scratching my head. Probably me doing something wrong. 
I can see that the value is read as well at some points.. so i thought i could expose it ..but ..

closed 2023-06-20 by github-actions[bot]

Cityloue15, 2023-12-14:
Hello, can anyone confirm that the problem has been closed and not resolved? 
My switch, when I press it, starts at max level without any possible configuration.
Best regards Alexandre