Koenkk/zigbee2mqtt #16371: Müller Licht bulbs constantly request OTAs

RubenKelevra, 2023-01-26:
### What happened?

I noticed that my Müller-Licht bulbs request an OTA update very often and Zigbee2MQTT seems to not react to it at all.

### What did you expect to happen?

Maybe Z2M can tell devices that there's no OTA available, if they are not yet supported with firmware updates, to give them a response to that and back off with their requests?

### How to reproduce it (minimal and precise)

Use Müller Licht bulbs in a Zigbee Network with Z2M/zStack as coordinator

### Zigbee2MQTT version

1.29.2

### Adapter firmware version

20221102

### Adapter

Slaesh's CC2652RB stick

### Debug log

[ota_log.txt](https://github.com/Koenkk/zigbee2mqtt/files/10510684/ota_log.txt)

Koenkk, 2023-01-26:
Normally Zigbee2MQTT should respond to these requests: https://github.com/Koenkk/zigbee2mqtt/blob/140213950d73492217e33e5d8f04c7a0cb81868c/lib/extension/otaUpdate.ts#L124 . However as this is not always done, can you provide me the herdsman debug log so I can double check this?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

Styx85, 2023-01-27:
[putty.log](https://github.com/Koenkk/zigbee2mqtt/files/10523271/putty.log)

Well, while I wasn't able to transfer it to the internal log file, I did a grab with putty.
So the Muller Lights are:
- Vorgarten Tür
- Vorgarten Treppe
- Garten Wohnzimmer
- Wohnzimmer Stehlampe 2

I hope this helps.

Koenkk, 2023-01-29:
It is a bit hard to see in this log, I've improved the logging, the herdsman debug logging is not needed anymore, just the normal z2m debug logging is enough. After each `Device 'xx' requested OTA` you will now see either `Responded to OTA request of 'xx' with 'NO_IMAGE_AVAILABLE'` or `Did not respond to OTA request of 'xx' (no OTA endpoint)`

Styx85, 2023-01-29:
Yep now it is easier: 

```
Debug 2023-01-29 18:31:50Received Zigbee message from 'Vorgarten Treppe', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":537462801,"imageType":268,"manufacturerCode":4137}' from endpoint 1 with groupID 0
Debug 2023-01-29 18:31:50Device 'Vorgarten Treppe' requested OTA
Debug 2023-01-29 18:31:50Did not respond to OTA request of 'Vorgarten Treppe' (no OTA endpoint)
Debug 2023-01-29 18:32:08Received Zigbee message from 'Vorgarten Treppe', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":537462801,"imageType":268,"manufacturerCode":4137}' from endpoint 1 with groupID 0
Debug 2023-01-29 18:32:08Device 'Vorgarten Treppe' requested OTA
Debug 2023-01-29 18:32:08Did not respond to OTA request of 'Vorgarten Treppe' (no OTA endpoint)
Debug 2023-01-29 18:32:26Received Zigbee message from 'Vorgarten Treppe', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":537462801,"imageType":268,"manufacturerCode":4137}' from endpoint 1 with groupID 0
Debug 2023-01-29 18:32:26Device 'Vorgarten Treppe' requested OTA
Debug 2023-01-29 18:32:26Did not respond to OTA request of 'Vorgarten Treppe' (no OTA endpoint)
```
```
Debug 2023-01-29 18:31:13Received Zigbee message from 'Vorgarten Tür', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":537462801,"imageType":268,"manufacturerCode":4137}' from endpoint 1 with groupID 0
Debug 2023-01-29 18:31:13Device 'Vorgarten Tür' requested OTA
Debug 2023-01-29 18:31:14Did not respond to OTA request of 'Vorgarten Tür' (no OTA endpoint)
Debug 2023-01-29 18:31:32Received Zigbee message from 'Vorgarten Tür', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":537462801,"imageType":268,"manufacturerCode":4137}' from endpoint 1 with groupID 0
Debug 2023-01-29 18:31:32Device 'Vorgarten Tür' requested OTA
Debug 2023-01-29 18:31:32Did not respond to OTA request of 'Vorgarten Tür' (no OTA endpoint)
Debug 2023-01-29 18:31:50Received Zigbee message from 'Vorgarten Tür', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":537462801,"imageType":268,"manufacturerCode":4137}' from endpoint 1 with groupID 0
Debug 2023-01-29 18:31:50Device 'Vorgarten Tür' requested OTA
Debug 2023-01-29 18:31:50Did not respond to OTA request of 'Vorgarten Tür' (no OTA endpoint)
```
```
Debug 2023-01-29 18:31:43Received Zigbee message from 'Wohnzimmer Stehlampe 2', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":537462801,"imageType":268,"manufacturerCode":4137}' from endpoint 1 with groupID 0
Debug 2023-01-29 18:31:43Device 'Wohnzimmer Stehlampe 2' requested OTA
Debug 2023-01-29 18:31:43Did not respond to OTA request of 'Wohnzimmer Stehlampe 2' (no OTA endpoint)
Debug 2023-01-29 18:32:01Received Zigbee message from 'Wohnzimmer Stehlampe 2', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":537462801,"imageType":268,"manufacturerCode":4137}' from endpoint 1 with groupID 0
Debug 2023-01-29 18:32:01Device 'Wohnzimmer Stehlampe 2' requested OTA
Debug 2023-01-29 18:32:01Did not respond to OTA request of 'Wohnzimmer Stehlampe 2' (no OTA endpoint)
Debug 2023-01-29 18:32:19Received Zigbee message from 'Wohnzimmer Stehlampe 2', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":537462801,"imageType":268,"manufacturerCode":4137}' from endpoint 1 with groupID 0
Debug 2023-01-29 18:32:19Device 'Wohnzimmer Stehlampe 2' requested OTA
Debug 2023-01-29 18:32:19Did not respond to OTA request of 'Wohnzimmer Stehlampe 2' (no OTA endpoint)
```

Koenkk, 2023-01-29:
`Did not respond to OTA request of 'Wohnzimmer Stehlampe 2' (no OTA endpoint)` -> I've changed the code to now always respond, in case non has an OTA endpoint it will respond to the one it was requested with.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Styx85, 2023-01-29:
I can confirm this changes and reduces the spamming with OTA requests. The bulbs now fire the OTA request roughly every minute (one minute + at least one second). Before it was every 15 plus x seconds. That takes some pressure from the network, but is still not nice. It has already been suspected before that the firmware in these kinds of lamps are incorrectly in a developer mode. So I think that is kind of the best we can get (maybe I get in touch with the company to ask for an updated firmware).

I will now dig deeper into #16278, @ASNNetworks there already suggested that the bulbs react with flickering when pairing mode "all" is activated (so far confirmed from my side). So it seems to be kind of developer mode, too. Maybe you can give some hints in the other thread what to look for in the logs.

```
Debug 2023-01-29 21:03:00Received Zigbee message from 'Wohnzimmer Stehlampe 2', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":537462801,"imageType":268,"manufacturerCode":4137}' from endpoint 1 with groupID 0
Debug 2023-01-29 21:03:00Device 'Wohnzimmer Stehlampe 2' requested OTA
Debug 2023-01-29 21:03:00Responded to OTA request of 'Wohnzimmer Stehlampe 2' with 'NO_IMAGE_AVAILABLE'
Debug 2023-01-29 21:04:01Received Zigbee message from 'Wohnzimmer Stehlampe 2', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":537462801,"imageType":268,"manufacturerCode":4137}' from endpoint 1 with groupID 0
Debug 2023-01-29 21:04:01Device 'Wohnzimmer Stehlampe 2' requested OTA
Debug 2023-01-29 21:04:01Responded to OTA request of 'Wohnzimmer Stehlampe 2' with 'NO_IMAGE_AVAILABLE'
Debug 2023-01-29 21:05:02Received Zigbee message from 'Wohnzimmer Stehlampe 2', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":537462801,"imageType":268,"manufacturerCode":4137}' from endpoint 1 with groupID 0
Debug 2023-01-29 21:05:02Device 'Wohnzimmer Stehlampe 2' requested OTA
Debug 2023-01-29 21:05:02Responded to OTA request of 'Wohnzimmer Stehlampe 2' with 'NO_IMAGE_AVAILABLE'
Debug 2023-01-29 21:06:04Received Zigbee message from 'Wohnzimmer Stehlampe 2', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":537462801,"imageType":268,"manufacturerCode":4137}' from endpoint 1 with groupID 0
Debug 2023-01-29 21:06:04Device 'Wohnzimmer Stehlampe 2' requested OTA
Debug 2023-01-29 21:06:04Responded to OTA request of 'Wohnzimmer Stehlampe 2' with 'NO_IMAGE_AVAILABLE'
Debug 2023-01-29 21:07:05Received Zigbee message from 'Wohnzimmer Stehlampe 2', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":537462801,"imageType":268,"manufacturerCode":4137}' from endpoint 1 with groupID 0
Debug 2023-01-29 21:07:05Device 'Wohnzimmer Stehlampe 2' requested OTA
Debug 2023-01-29 21:07:05Responded to OTA request of 'Wohnzimmer Stehlampe 2' with 'NO_IMAGE_AVAILABLE'
Debug 2023-01-29 21:08:06Received Zigbee message from 'Wohnzimmer Stehlampe 2', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":537462801,"imageType":268,"manufacturerCode":4137}' from endpoint 1 with groupID 0
Debug 2023-01-29 21:08:06Device 'Wohnzimmer Stehlampe 2' requested OTA
Debug 2023-01-29 21:08:06Responded to OTA request of 'Wohnzimmer Stehlampe 2' with 'NO_IMAGE_AVAILABLE'
```
```
Debug 2023-01-29 21:08:17Received Zigbee message from 'Vorgarten Tür', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":537462801,"imageType":268,"manufacturerCode":4137}' from endpoint 1 with groupID 0
Debug 2023-01-29 21:08:17Device 'Vorgarten Tür' requested OTA
Debug 2023-01-29 21:08:17Responded to OTA request of 'Vorgarten Tür' with 'NO_IMAGE_AVAILABLE'
Debug 2023-01-29 21:09:18Received Zigbee message from 'Vorgarten Tür', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":537462801,"imageType":268,"manufacturerCode":4137}' from endpoint 1 with groupID 0
Debug 2023-01-29 21:09:18Device 'Vorgarten Tür' requested OTA
Debug 2023-01-29 21:09:18Responded to OTA request of 'Vorgarten Tür' with 'NO_IMAGE_AVAILABLE'
Debug 2023-01-29 21:10:19Received Zigbee message from 'Vorgarten Tür', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":537462801,"imageType":268,"manufacturerCode":4137}' from endpoint 1 with groupID 0
Debug 2023-01-29 21:10:19Device 'Vorgarten Tür' requested OTA
Debug 2023-01-29 21:10:19Responded to OTA request of 'Vorgarten Tür' with 'NO_IMAGE_AVAILABLE'
```

closed 2023-03-08 by github-actions[bot]