Koenkk/zigbee2mqtt #16372: Network unreachable after saving state.json

RubenKelevra, 2023-01-26:
### What happened?

I noticed that nothing in my Z2M seems to work. In Z2M all devices showed beeing "last seen" at least an hour ago.

So I removed the Zigbee Stick for several seconds, plugged it back in and restarted Z2M.

Everything returned to normal.

The debug log shows that all worked well, then the state.json gets written and everything is offline.

### What did you expect to happen?

Stable operation

### How to reproduce it (minimal and precise)

unknown

### Zigbee2MQTT version

1.29.2

### Adapter firmware version

20221102

### Adapter

Slaesh's CC2652RB stick

### Debug log

[crash_log.txt](https://github.com/Koenkk/zigbee2mqtt/files/10510907/crash_log.txt)

Koenkk, 2023-01-26:
This is not related to the `state.json` saving. The root cause is that your adapter either crashes or is disconnected (error: `SRSP - AF - dataRequest after 6000ms`)

- Make sure that no other services are accessing the devices (e.g. Home Assistant ZHA USB discovery).
- This fw might help: https://github.com/Koenkk/Z-Stack-firmware/tree/6.10.01.01/coordinator/Z-Stack_3.x.0/bin , `20221102` is not a stable fw

RubenKelevra, 2023-01-28:
Hey @Koenkk 

> * This fw might help: https://github.com/Koenkk/Z-Stack-firmware/tree/6.10.01.01/coordinator/Z-Stack_3.x.0/bin , `20221102` is not a stable fw

Yeah, I know, but with the stable I have massive issues with latency and messages not beeing received. See https://github.com/Koenkk/zigbee2mqtt/issues/15707#issuecomment-1368089862

Also, just found that someone else has reported the same issue. However, he tried multiple firmware versions and, which didn't solve it for him.

> This is not related to the `state.json` saving. The root cause is that your adapter either crashes or is disconnected (error: `SRSP - AF - dataRequest after 6000ms`)

Well, I think it's a Z2M issue, not a firmware one:

- Restarting Z2M with the “tools”-tab does not fix the issue.
- Restarting Z2M-Addon in HA did solve the issue.

So it's not required to unplug and replug the stick to get the network working again.

RubenKelevra, 2023-01-28:
Here's the next "hungup" where all devices did go offline.

Again, restarting add-on worked, the "tools" -> "Restart Z2M" did not.

[crash_log 2.txt](https://github.com/Koenkk/zigbee2mqtt/files/10524213/crash_log.2.txt)

Koenkk, 2023-01-29:
@RubenKelevra I'm not sure what restarting the z2m addon does, maybe it also reset the serial port. I don't consider a restart/reset a fix of this issue, this issue shouldn't happen in the first place. What hardware are you running z2m/HA on?

RubenKelevra, 2023-02-02:
@Koenkk Hardware is a Raspberry PI 4B (8GB) with a 128 HB sdcard with high endurance.

The Zigbee Stick is connected with a premium quality 3m USB cable and no other USB devices are connected to RasPi.

Network connection is established, to the RasPi, via Ethernet.

I've updated to the 20220219 as it crashed once again on the day I made my last comment.

No crash yet, however I had 20221102 installed multiple weeks without having issues. So I'll report back soon. :)

RubenKelevra, 2023-02-28:
This issue has been fixed by updating the Z-Stack firmware, as you recommended. Thanks!

closed 2023-02-28 by RubenKelevra