Koenkk/zigbee2mqtt #16375: [New device support]: Tuya zigbee Dimmer ts0601

admiralroflknife, 2023-01-26:
### Link

https://www.amazon.com/dp/B0BFKQKJC9?ref=ppx_yo2ov_dt_b_product_details&th=1

### Database entry

{"id":10,"type":"Router","ieeeAddr":"0xa4c13876541006d3","nwkAddr":30474,"manufId":4417,"manufName":"_TZE200_ctq0k47x","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"J*e+iL*e+i","65506":54,"65508":0,"65534":0,"stackVersion":0,"dateCode":"","appVersion":70,"manufacturerName":"_TZE200_ctq0k47x","zclVersion":3,"modelId":"TS0601","powerSource":1}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-708457359},"lastSeen":1674761296263,"defaultSendRequestWhen":"immediate"}

### Comments

tried a few configs that were close but just keep getting bootloops. I'd really appreciate some help

Koenkk, 2023-01-29:
Could you check if the issue is fixed with the following external converter: 

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [
        {modelID: 'TS0601', manufacturerName: '_TZE200_whpb9yts'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_ebwgzdqq'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_9i9dt8is'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_dfxkcots'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_w4cryh2i'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_ojzhk75b'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_swaamsoy'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_ctq0k47x'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_3p5ydos3'},
    ],
    model: 'TS0601_dimmer',
    vendor: 'TuYa',
    description: 'Zigbee smart dimmer',
    fromZigbee: [fz.tuya_dimmer, fz.ignore_basic_report],
    toZigbee: [tz.tuya_dimmer_state, tz.tuya_dimmer_level],
    configure: async (device, coordinatorEndpoint, logger) => {
        await tuya.configureMagicPacket(device, coordinatorEndpoint, logger);
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
    },
    exposes: [e.light_brightness().withMinBrightness().withMaxBrightness().setAccess(
        'state', ea.STATE_SET).setAccess('brightness', ea.STATE_SET).setAccess(
        'min_brightness', ea.STATE_SET).setAccess('max_brightness', ea.STATE_SET)],
    whiteLabel: [
        {vendor: 'Larkkey', model: 'ZSTY-SM-1DMZG-EU'},
        {vendor: 'Earda', model: 'EDM-1ZAA-EU'},
        {vendor: 'Earda', model: 'EDM-1ZAB-EU'},
        {vendor: 'Earda', model: 'EDM-1ZBA-EU'},
        {vendor: 'Mercator Ikuü', model: 'SSWD01'},
        {vendor: 'Moes', model: 'ZS-USD'},
        {vendor: 'Moes', model: 'EDM-1ZBB-EU'},
    ],
};

module.exports = definition;

```
- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

admiralroflknife, 2023-01-29:
Dude your awesome. Works perfectly!

Koenkk, 2023-01-29:
I assume that means "yes it works now", added it! 😄 

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-01-29 by Koenkk

ssrsmart, 2023-12-31:
> Could you check if the issue is fixed with the following external converter:
>
> ```js
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = [...]

Hi,
I have done all the steps but i got this error

[22:23:04] INFO: Preparing to start...
[22:23:04] INFO: Socat not enabled
[22:23:05] INFO: Starting Zigbee2MQTT...

/app/node_modules/zigbee-herdsman-converters/index.js:172
        if (converter.options) {
                      ^
TypeError: Cannot read properties of undefined (reading 'options')
    at Object.addDefinition [as addDeviceDefinition] (/app/node_modules/zigbee-herdsman-converters/index.js:172:23)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:15:17)
    at new Controller (/app/lib/controller.ts:84:58)
    at start (/app/index.js:106:18

Any idea? 
Thanks