Koenkk/zigbee2mqtt #16376: Bosch Radiator Thermostats II (Bosch BTH-RA) become inaccessible after some time

Uarml, 2023-01-26:
### What happened?

A few days after pairing The Bosch Thermostats 2 I can't communicate with them anymore.
The ZigBee stick is only 5 meters away from the thermostats.
Full reboot makes all thermostats functional again, whereas a simple restart does not.
Maybe they go in sleep mode or something like that. 
I have a thermostat in my bathroom, i can still communicate with it. 
Maybe it doesn't go in a sleep mode because it is often switched to window open and closed mode.

### What did you expect to happen?

After pairing thermostats should be available permanently.

### How to reproduce it (minimal and precise)

Pair thermostats and wait.

### Zigbee2MQTT version

1.29.2 running on Home Assistant

### Adapter firmware version

20220219

### Adapter

Sonoff Zigbee 3.0 USB Dongle plus

### Debug log

Publish 'set' 'system_mode' to 't.w.h' failed: 'Error: Write 0x18fc26000003b444/1 hvacThermostat({"16391":{"value":5,"type":48}}, {"sendWhen":"active","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4617,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'

erzhan89, 2023-01-27:
Same issue...

Uarml, 2023-01-27:
> Same issue...  
> 

That's bad.

mipoat, 2023-01-29:
Could it be this one is related to: https://github.com/Koenkk/zigbee2mqtt/issues/13478 where the TRV gets unresponsive when connected to the coordinator but works flawless when connected via a router?

Uarml, 2023-01-29:
> Could it be this one is related to: #13478 where the TRV gets unresponsive when connected to the coordinator but works flawless when connected via a router?

Maybe, but i can't test it now because, i've no router devices at the moment.

BBJake, 2023-01-31:
As commented in https://github.com/Koenkk/zigbee2mqtt/issues/13478 it seems to be definitiv the same problem.
A have this problems only if my thermostat is connected to the Coordinator. Devices that are connected to a route have no problem!
I have this problem with Bosch Thermostat II, but not with Saswell TS0601 (Hama).
Also look here: https://github.com/Koenkk/zigbee2mqtt/issues/13478#issuecomment-1383942462

gekberlin, 2023-02-03:
I also have the same issue with an Sonoff Zigbee Dongle Plus. In my case the issue beginns even after just a few hours.

Uarml, 2023-02-04:
> I also have the same issue with an Sonoff Zigbee Dongle Plus. In my case the issue beginns even after just a few hours.

Sometimes this happens after a few hours after pairing and sometimes after a few days.

BBJake, 2023-02-16:
There seems to be hope for a fix to this problem... 
#https://github.com/Koenkk/zigbee2mqtt/issues/13478#issuecomment-1432643349

xdubx, 2023-02-21:
Yes same over here. Sonoff Zigbee 3.0 USB Dongle Plus. With stock firmware. After the device is not reachable in the log don't a publish message appears.

closed 2023-04-01 by github-actions[bot]

slugzero, 2023-04-10:
There is a new firmware available for testing which should fix this issue: [20230410](https://github.com/Koenkk/Z-Stack-firmware/tree/f5a6270875cc13d32387524b54d460c9b6274841/coordinator/Z-Stack_3.x.0/bin)

mldytech, 2023-04-20:
> There is a new firmware available for testing which should fix this issue: [20230410](https://github.com/Koenkk/Z-Stack-firmware/tree/f5a6270875cc13d32387524b54d460c9b6274841/coordinator/Z-Stack_3.x.0/bin)

Just for information: For me the issue seems to be fixed. I'm running 3 bosch TRVs for over a week now without any issues. Thanks!

xdubx, 2023-05-16:
F.y.i. after one month they disconnects. ....

lieblinger, 2024-02-18:
Still having the same issue on latest Z2M with Sonoff E-Dongle on stock firmware. Is there something I can do (firmware update)?

xdubx, 2024-03-08:
Iam driving curently 20230507. You can find it here. https://www.zigbee2mqtt.io/guide/adapters/#recommended. But my rtvs still disconnects.

xdubx, 2024-03-08:
@Uarml  can you approve the new firmware fix the issue? And reopen the bug?

Uarml, 2024-07-12:
> @Uarml can you approve the new firmware fix the issue? And reopen the bug?

I haven't used my thermostats for quite a while, the last time was around early March. Back then, all thermostats except one worked. Maybe I'll try to connect them again and see what happens, if you would like to.