Koenkk/zigbee2mqtt #16381: Bosch Thermostat II - pairing does not work

stkr22, 2023-01-27:
### What happened?

Fresh install of zigbee2mqtt entering paring mode. Bosch Thermostat immediately joins the network but is "not supported". Which suprised me because in the [docs](https://www.zigbee2mqtt.io/devices/BTH-RA.html) it seems like the device is just supported as is. I also tried adding the security code via the frontend but that also did not change the interview failing.

Did some digging and read #13520, #15576, #16160. I tried adding a new extensions via the frontend. My Device IEEE address is "0x18fc260000089829" and security code "RB01SG0D8310182648008000000000000000000018FC260000089829DLK476DE40AF3935A818BA42720D834A0543FE3". This is what I added:

```
const constants = require('zigbee-herdsman/dist/adapter/z-stack/unpi/constants');
const {Buffer} = require('buffer');
console.log(Buffer);
const Subsystem = constants.Subsystem;

class MyExampleExtension {
    constructor(zigbee, mqtt, state, publishEntityState, eventBus, settings, logger) {
        this.zigbee = zigbee;
        this.mqtt = mqtt;
        this.state = state;
        this.publishEntityState = publishEntityState;
        this.eventBus = eventBus;
        this.settings = settings;
        this.logger = logger;
    }

    /**
     * This method is called by the controller once Zigbee2MQTT has been started.
     */
    async start() {
        console.log('Adding install code!');
        const payload = {
            installCodeFormat: 1,
            ieeeaddr: '0x18fc260000089829',
            installCode: Buffer.from([0xAE, 0x3B, 0x28, 0x72, 0x81, 0xCF, 0x16, 0xF5, 0x50, 0x73, 0x3A, 0x0C, 0xEC, 0x38, 0xAA, 0x31, 0xE8, 0x02]),
        };
        const k = await this.zigbee.herdsman.adapter.znp.request(Subsystem.APP_CNF, 'bdbAddInstallCode', payload);
        console.log(k);
        console.log('Added install code');
    }
}

module.exports = MyExampleExtension;
```

However, that does not seem to work and I guess it because I have to change `0xAE, 0x3B, 0x28, 0x72, 0x81, 0xCF, 0x16, 0xF5, 0x50, 0x73, 0x3A, 0x0C, 0xEC, 0x38, 0xAA, 0x31, 0xE8, 0x02` to my actual install code but I don't get how? 

### Zigbee2MQTT version

1.29.2

### Adapter firmware version

20220219

### Adapter

Zigstar v4

Uarml, 2023-01-28:
You've to add the Install Code. It' printed in in the battery slots.

[image]

stkr22, 2023-01-29:
@Uarml Yeah I tried that but it did not work. 

However, the issue turned out to be my ZigStart v4 stick. I ordered a Sonos Zigbee 3.0 Stick yesterday and it just worked without any issues. 

Closing.

closed 2023-01-29 by stkr22

Uarml, 2023-01-29:
> @Uarml Yeah I tried that but it did not work.
> 
> However, the issue turned out to be my ZigStart v4 stick. I ordered a Sonos Zigbee 3.0 Stick yesterday and it just worked without any issues.
> 
> Closing.

OK, good luck.