Koenkk/zigbee2mqtt #16382: [Feature request]: Auto reconnect the device if z2m is NOT in paring mode

dinhchinh82, 2023-01-27:
### Is your feature request related to a problem? Please describe

At current, if the zigbee device (such as smart switch) is reset accidently, that device is left from zigbee network automatically and need to pair it again manually to make it work again.

That's quite normal case when some smart zigbee touch switch is quite easy to be put into the paring mode by only pushing 1 button for about 5 seconds. 

### Describe the solution you'd like

 if the zigbee device is reset accidently, it will re-connect to the coordinator automatically after 3 minutes (Tuya gateway is currently did like that)

### Describe alternatives you've considered

There is another solution that add "auto reconnect" option under Settings => Advanced from Z2m front-end which allows to turn on/off the auto reconnect zigbee device from reseting accidently.

1. If "auto reconnect" option is set to ON: if the zigbee device is reset accidently, it will re-connect to the coordinator automatically after 3 minutes (Tuya gateway is currently did like that
2. If "auto reconnect" option is set to OFF: everything works as current.

### Additional context

If auto reconnect is supported, the zigbee devices would not be left from the network even if it is reset accidently.

danieledwardgeorgehitchcock, 2023-01-27:
I don't think this would be possible due to the way the Zigbee protocol works.. When a device pairs, a series of unique keys are exchanged between the device and your co-ordinator. When a device leaves the network, this information is forgotten by the device

dinhchinh82, 2023-01-27:
> I don't think this would be possible due to the way the Zigbee protocol works.. When a device pairs, a series of unique keys and information are exchanged between the device and your co-ordinator. When a device leaves the network, this information is forgotten by the device

I already tested this function with Tuya zigbee gateway and some new Zigbee smart switch and it works well. If I tried to reset the switch by long pressing a buton about 10 seconds without touching to the app or gateway, the switch is left from Tuya zigbee gateway then it reconnect to that gateway again after 3 minutes.

dinhchinh82, 2023-01-29:
[image]
Here is the spec of Tuya smart switch which supported the anti-error detetion function.
@Koenkk could you please take a look about this feature? 
Thank you.

closed 2023-03-08 by github-actions[bot]