Koenkk/zigbee2mqtt #16383: Aqara E1 roller shutter (ZNJLBL01LM)

wouterverduin, 2023-01-27:
### What happened?

Hey guys

Suddenly overnight one of my aqara roller shutters stops working.  The left one works, the right one doenst. Restart Home assistant, Z2M without effect. Also resetted the device (gives a red light then proceeds to blinking blue). However, still the time-outs and doesnt seem to communicate with Z2M.

Any help would be greatly appreciated!

### What did you expect to happen?

Both my aqara roller shutters to work.

### Zigbee2MQTT version

1.29.2-1

### Adapter firmware version

ConBee2/RaspBee2  0x26720700

### Adapter

10-09-2022

### Debug log

[09:19:13] INFO: Preparing to start...
[09:19:15] INFO: Socat not enabled
[09:19:22] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:warn  2023-01-27 09:19:40: `permit_join` set to  `true` in configuration.yaml.
Zigbee2MQTT:warn  2023-01-27 09:19:40: Allowing new devices to join.
Zigbee2MQTT:warn  2023-01-27 09:19:40: Set `permit_join` to `false` once you joined all devices.
Zigbee2MQTT:error 2023-01-27 09:21:07: Publish 'set' 'state' to 'BlinderingR' failed: 'Error: Write 0x54ef44100044f43a/1 genAnalogOutput({"85":{"value":100,"type":57}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
Zigbee2MQTT:error 2023-01-27 09:21:11: Publish 'set' 'state' to 'BlinderingR' failed: 'Error: Write 0x54ef44100044f43a/1 genAnalogOutput({"85":{"value":100,"type":57}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
Zigbee2MQTT:error 2023-01-27 09:31:51: Publish 'set' 'state' to 'BlinderingR' failed: 'Error: Write 0x54ef44100044f43a/1 genAnalogOutput({"85":{"value":100,"type":57}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
Zigbee2MQTT:error 2023-01-27 10:29:43: Publish 'set' 'state' to 'BlinderingR' failed: 'Error: Write 0x54ef44100044f43a/1 genAnalogOutput({"85":{"value":100,"type":57}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
Zigbee2MQTT:error 2023-01-27 10:48:22: Publish 'set' 'state' to 'BlinderingR' failed: 'Error: Write 0x54ef44100044f43a/1 genAnalogOutput({"85":{"value":100,"type":57}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
Zigbee2MQTT:error 2023-01-27 10:48:25: Publish 'set' 'state' to 'BlinderingR' failed: 'Error: Write 0x54ef44100044f43a/1 genAnalogOutput({"85":{"value":0,"type":57}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
Zigbee2MQTT:error 2023-01-27 11:24:07: Failed to configure 'BlinderingR', attempt 1 (Error: Read 0x54ef44100044f43a/1 aqaraOpple([1034], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)
    at DeconzAdapter.sendZclFrameToEndpoint (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:658:23)
    at Endpoint.sendRequest (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:299:20)
    at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:472:28)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/devices/xiaomi.js:1931:13)
    at Configure.configure (/app/lib/extension/configure.ts:117:13)
    at Configure.onMQTTMessage (/app/lib/extension/configure.ts:55:21))
Zigbee2MQTT:error 2023-01-27 11:32:17: Publish 'set' 'state' to 'BlinderingR' failed: 'Error: Write 0x54ef44100044f43a/1 genAnalogOutput({"85":{"value":0,"type":57}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
Zigbee2MQTT:error 2023-01-27 11:33:08: Publish 'set' 'state' to 'BlinderingR' failed: 'Error: Write 0x54ef44100044f43a/1 genAnalogOutput({"85":{"value":100,"type":57}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
Zigbee2MQTT:error 2023-01-27 11:33:29: Publish 'set' 'state' to 'BlinderingR' failed: 'Error: Write 0x54ef44100044f43a/1 genAnalogOutput({"85":{"value":100,"type":57}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
Zigbee2MQTT:error 2023-01-27 11:33:33: Publish 'set' 'state' to 'BlinderingR' failed: 'Error: Write 0x54ef44100044f43a/1 genMultistateOutput({"presentValue":2}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
Zigbee2MQTT:error 2023-01-27 11:33:42: Request 'zigbee2mqtt/bridge/request/device/remove' failed with error: 'Failed to remove device 'BlinderingR' (block: false, force: false) (Error: waiting for response TIMEOUT)'
Zigbee2MQTT:error 2023-01-27 11:33:56: Request 'zigbee2mqtt/bridge/request/touchlink/scan' failed with error: 'not supported'
Zigbee2MQTT:error 2023-01-27 11:34:09: Request 'zigbee2mqtt/bridge/request/device/remove' failed with error: 'Failed to remove device 'BlinderingR' (block: false, force: false) (Error: waiting for response TIMEOUT)'
Zigbee2MQTT:error 2023-01-27 11:34:17: Request 'zigbee2mqtt/bridge/request/device/remove' failed with error: 'Failed to remove device 'BlinderingR' (block: false, force: false) (Error: waiting for response TIMEOUT)'
Zigbee2MQTT:error 2023-01-27 11:38:59: Update or check for update already in progress for 'BlinderingR'
Zigbee2MQTT:error 2023-01-27 11:39:45: Failed to check if update available for 'BlinderingR' (Device didn't respond to OTA request)

danieledwardgeorgehitchcock, 2023-01-27:
Looks like your device is struggling to get a route back to your co-ordinator - how far is it away from a zigbee router device or your co-ordinator?

wouterverduin, 2023-01-27:
Yeah i was thinking it was a connectivity issue aswell at this point. The strange thing is it is at the same window as my other controller and is the closest to the hub.

I eventuelly got it sorted this morning. Not sure how but after updating the other one and restarting Z2M and finally factory resetting the driver it reconnected to the Z2M and is working again!

Thanks for your time!

closed 2023-03-06 by github-actions[bot]