Koenkk/zigbee2mqtt #16385: zigbee2mqtt crash / restart

alexwala, 2023-01-27:
### What happened?

noticed that zigbee2mqtt sometimes restarts because of 0x10: MEM_ERROR

### What did you expect to happen?

zigbee2mqtt shouldn't restart that often

### Zigbee2MQTT version

1.29.2-dev commit: 98154282

### Adapter firmware version

20211115

### Adapter

CC2531

### Debug log

```
Jan 27 12:50:39 raspberrypi npm[502]: Error: SREQ '--> AF - dataRequestExt - {"dstaddrmode":2,"dstaddr":"0x000000000000fffd","destendpoint":242,"dstpanid":0,"srcendpoint":242,"clusterid":33,"transid":145,"options":0,"radius":30,"len":6,"data":{"type":"Buffer","data":[25,131,2,11,254,0]}}' failed with status '(0x10: MEM_ERROR)' (expected '(0x00: SUCCESS)')
Jan 27 12:50:39 raspberrypi npm[502]:     at Object.func (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:322:27)
Jan 27 12:50:39 raspberrypi npm[502]:     at Queue.executeNext (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
Jan 27 12:50:39 raspberrypi systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
Jan 27 12:50:39 raspberrypi systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
Jan 27 12:50:39 raspberrypi systemd[1]: zigbee2mqtt.service: Service RestartSec=100ms expired, scheduling restart.
Jan 27 12:50:39 raspberrypi systemd[1]: zigbee2mqtt.service: Scheduled restart job, restart counter is at 1.
Jan 27 12:50:39 raspberrypi systemd[1]: Stopped zigbee2mqtt.
Jan 27 12:50:39 raspberrypi systemd[1]: Started zigbee2mqtt.

Jan 27 13:44:59 raspberrypi npm[22183]: Error: SREQ '--> AF - dataRequestExt - {"dstaddrmode":2,"dstaddr":"0x000000000000fffd","destendpoint":242,"dstpanid":0,"srcendpoint":242,"clusterid":33,"transid":9,"options":0,"radius":30,"len":6,"data":{"type":"Buffer","data":[25,5,2,11,254,0]}}' failed with status '(0x10: MEM_ERROR)' (expected '(0x00: SUCCESS)')
Jan 27 13:44:59 raspberrypi npm[22183]:     at Object.func (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:322:27)
Jan 27 13:44:59 raspberrypi npm[22183]:     at Queue.executeNext (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
Jan 27 13:44:59 raspberrypi systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
Jan 27 13:44:59 raspberrypi systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
Jan 27 13:44:59 raspberrypi systemd[1]: zigbee2mqtt.service: Service RestartSec=100ms expired, scheduling restart.
Jan 27 13:44:59 raspberrypi systemd[1]: zigbee2mqtt.service: Scheduled restart job, restart counter is at 6.
Jan 27 13:44:59 raspberrypi systemd[1]: Stopped zigbee2mqtt.
Jan 27 13:44:59 raspberrypi systemd[1]: Started zigbee2mqtt.
Jan 27 13:45:01 raspberrypi npm[22824]: > zigbee2mqtt@1.29.2-dev start
```

Koenkk, 2023-01-29:
` 0x10: MEM_ERROR` in combination with the CC2531 is common. The CC2531 chip is just not powerful enough, I suggest upgrading to one of the recommended adapters: https://www.zigbee2mqtt.io/guide/adapters/

alexwala, 2023-01-30:
even with only 15 devices ?

Koenkk, 2023-01-30:
Yes it can happen with even 15 devices

alexwala, 2023-01-31:
Ok will try a Sonos adapter with current config.

alexwala, 2023-02-04:
So installed the Sonos adapter (P Version). IEEE copied. works.

BUT: having an Ikea Bulb as router and I am no longer able to join a xiaomi temperature sensor via Ikea Bulb (permit join ikea bulb) - it still worked in this way with the cc2531 ... ?

Does the sonoff zigbee coordinator needs router with zigbee 3.0 ?

alexwala, 2023-02-06:
seems to work better now :)

closed 2023-02-06 by alexwala