Koenkk/zigbee2mqtt #16388: IKEA Bulbs doesn't seem to have correct state to reality

samip5, 2023-01-27:
### What happened?

An automation triggered lights to turn on, and only one of them actually did and after checking in the frontend and HA, it's reported as being on, despite in reality that did not happen.

This is happening with:

[LED1624G9](https://www.zigbee2mqtt.io/devices/LED1624G9.html#ikea-led1624g9)
[LED1732G11](https://www.zigbee2mqtt.io/devices/LED1732G11.html#ikea-led1732g11)

### What did you expect to happen?

I would have expected it to return the correct state from the device.

### Zigbee2MQTT version

1.29.2 commit: bb3e8f6

### Adapter firmware version

20220219

### Adapter

Tube's PoE coordinator

merlinschumacher, 2023-02-08:
I have the same Issue with a L1529 and a LED1623G12. The L1529 claims to be on when it isn't. Even updating the status of it yields it being on while it's off. The LED1623G12 behaves completely erratic for a few days now. It won't react at all, but will claim to be on or off and react to brightness changes. In reality, the lamp is still off. Resetting the LED1623G12 doesn't help at all.

Zigbee2MQTT Version 1.30.1

Coordinator-Type zStack3x0

Coordinator-Version 20221226

Adapter: Electrolama zig-ah-zig-ah!

merlinschumacher, 2023-02-08:
#16484 seems to be a similar issue

samip5, 2023-02-09:
Hey @merlinschumacher out of curiosity, which firmware versions are your problem LED1623G12 running?

Mine [LED1624G9](https://www.zigbee2mqtt.io/devices/LED1624G9.html#ikea-led1624g9) is on firmware 2.3.093 and the mostly working [LED1924G9](https://www.zigbee2mqtt.io/devices/LED1924G9.html#ikea-led1924g9) one is on 1.0.021

Lampyridae1984, 2023-02-10:
I have similar issues with my [LED1924G9](https://www.zigbee2mqtt.io/devices/LED1924G9.html#ikea-led1924g9). After a 6h power outage, I had to re-pair all of my IKEA bulbs. The LED1924G9 kinda work, but everything takes forever. Every change/command takes up to 10 seconds to perform. All other bulbs seem to work fine.

Zigbee2MQTT Version 1.30.1-1

Coordinator-Type zStack3x0

Coordinator-Version 20221226

Adapter: Sonoff Zigbee 3.0 USB Dongle Plus

merlinschumacher, 2023-02-11:
> Hey @merlinschumacher out of curiosity, which firmware versions are your problem LED1623G12 running?
>
> Mine [LED1624G9](https://www.zigbee2mqtt.io/devices/LED1624G9.html#ikea-led1624g9) is on [...]

My lamps are all on the newest firmware.

Nevertheless, i found the fundamental issue: A node I used in my Node-RED setup was outdated, and I also had disabled retain (which was neede for the node to work) for the lights that caused issues. The problems seem to have gone now.

samip5, 2023-02-16:
I have retain enabled, and it's still affecting me on [1.30.1](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.30.1) commit: [eb878d3](https://github.com/Koenkk/zigbee2mqtt/commit/eb878d3)

samip5, 2023-03-19:
It's still an issue that probably would need to be looked at.

LookedPath, 2023-04-20:
This issue has been present for a while now and I honestly wouldn't know how to help in debugging it.
I can replicate this both when turning on the bulbs on Home Assistant on the app and with voice commands coming from a Google Home.

I have found that when using voice commands if instead of asking "Turn on xxx" I ask "Set xxx to 100%" the light turns on normally.

Gizzje, 2023-05-21:
I have this issue with the [L1527](https://www.zigbee2mqtt.io/devices/L1527.html#ikea-l1527) and [LED1545G12](https://www.zigbee2mqtt.io/devices/LED1545G12.html#ikea-led1545g12). Multiple times a week the lamps are still on after a HA-automation (motion based) should have turned them off and even Z2M and HA report them to be off. 

The other IKEA TRADFRI bulb I have, the [LED1934G3_E27](https://www.zigbee2mqtt.io/devices/LED1934G3_E27.html#ikea-led1934g3_e27), seems not affected by this problem.

Mr-Quin, 2023-06-13:
I have this issue with a few LED2003G10 bulbs.
In my case turning off adaptive lighting seems to solve it. If this is the case it may have something to do with sending commands in quick succession causing the bulb to "get stuck"

Gizzje, 2023-06-13:
@Mr-Quin interesting observation. I also use [Adaptive lighting](https://github.com/basnijholt/adaptive-lighting). 

I use it with the default interval of 90 seconds between commands being sent to the lights, which doesn't seem that short of an interval. But I will try increasing the interval, maybe that helps.

LookedPath, 2023-06-13:
I'm having this issue but I don't use Adaptive lighting so I think, at least in my case, it must be something else.

samip5, 2023-06-14:
I also don't use adaptive lightning.

rnorth, 2023-07-22:
Seeing the same with [LED2005R5](https://www.zigbee2mqtt.io/devices/LED2005R5.html#ikea-led2005r5) bulbs too. I'm using adaptive lighting, but not on the affected bulbs. I'm pretty sure this has only started for me in the last 24h, after an extended power outage to a subset of the bulbs in my mesh.

rnorth, 2023-07-23:
> Seeing the same with [LED2005R5](https://www.zigbee2mqtt.io/devices/LED2005R5.html#ikea-led2005r5) bulbs too. I'm using adaptive lighting, but not on the affected bulbs. I'm pretty sure this has only started for me in the last 24h, after an extended power outage to a subset of the bulbs in my mesh.

Followup: actually I gave in and decided to re-pair all the affected lights. This seems to have resolved the problem for now - not sure it's a 'fix', but might be worth a try as a workaround.

LookedPath, 2023-07-26:
> Followup: actually I gave in and decided to re-pair all the affected lights. This seems to have resolved the problem for now - not sure it's a 'fix', but might be worth a try as a workaround.

I've tried this but I still experience issues.
For instance when I turn on a [LED1925G6](https://www.zigbee2mqtt.io/devices/LED1925G6.html#ikea-led1925g6) on Home Assistant it's displayed as having full brightness but in reality it turns on at minimum brightness.
On [LED2003G10](https://www.zigbee2mqtt.io/devices/LED2003G10.html#ikea-led2003g10) instead when I turn them on they are actually off unless I turn them off and back on or if I manually change their brightness. When this happens I also checked straight on Z2M, to exclude any issues with MQTT and HA, but the status of the bulbs is wrong.

merlinschumacher, 2023-08-26:
Still an issue. Commenting for the bot

merlinschumacher, 2023-09-26:
Still an issue, dear bot.

legalgig, 2023-10-01:
I'm experiencing the same issue with LED2004G8

Wikibear, 2023-10-04:
I can repeat here:

zigbee2mqtt/lamp/set/brightness <-- switch will be on or off

and

zigbee2mqtt/lamp/brightness/set <-- switch will not change only the brightness value

On Off works better if I reload the state.

Koenkk, 2023-10-04:
Can someone provide a short as possible debug log of this? 

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

LookedPath, 2023-10-05:
> Can someone provide a short as possible debug log of this?
> 
> See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

This is a short a log as I could. The devices I was testing were Bedroom Nightstand Hannah (0x94deb8fffeb313b0) andBedroom Nightstand Simone (0x94deb8fffe9dca64). I was simply turning the lights on on Home Assistant without touching their brightness.
The lights report max brightness but in reality they turn on at minimum brightness and they stay like that until you force a brightness change.
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/12821151/log.txt)

Koenkk, 2023-10-05:
@LookedPath thanks, I see the brightness is not set yet. I need to better understand when this exactly happens. Can you do the following sequence, after each command, let me know if the z2m state matches the actual state.

- Publish to `zigbee2mqtt/Bedroom Nightstand Simone/set` payload `{"state":"ON", "brightness": 250}`
- Publish to `zigbee2mqtt/Bedroom Nightstand Simone/set` payload `{"state":"OFF"}`
- Publish to `zigbee2mqtt/Bedroom Nightstand Simone/set` payload `{"state":"ON"}`

LookedPath, 2023-10-05:
@Koenkk I ran the 3 commands and the state in z2m started matching reality right after the first one. The rest of the commands worked as intended. 
I noticed that this issue is only present after a while the bulbs have been off.

With the LED2003G10 it's even worse because when you just send an ON command they report to be on but in reality they are not. To have them actually turn on you can:
* Turn them off and then back on and then they'll be at their minimum brightness
* Change their brightness
Even with these ones the issue appears only after a while they've been off.

Koenkk, 2023-10-05:
So what is the sequence to trigger the issue?

- Publish to `zigbee2mqtt/Bedroom Nightstand Simone/set` payload `{"state":"ON", "brightness": 250}`
- Publish to `zigbee2mqtt/Bedroom Nightstand Simone/set` payload `{"state":"OFF"}`
- Wait 1 hour (??)
- Publish to `zigbee2mqtt/Bedroom Nightstand Simone/set` payload `{"state":"ON"}` -> now state is incorrect?

LookedPath, 2023-10-05:
Correct, the sequence is exactly like the one you outlined. I'm not sure on how long you need to wait before the problem starts showing up but I can try finding out.

Koenkk, 2023-10-06:
> but I can try finding out.

Please do, it would be very strange if only the change in time causes this.

legalgig, 2023-10-23:
I tried to investigate this issue a little further and found out one thing. After setting report option to true in advanced settings the bulb changes it's state from off to on after exactly 5mins. Zigbee2mqtt is showing that the state change came from one of the bulbs.
```
Received Zigbee message from 'dummy-bulb-2', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
```
btw. I have two same bulbs in one group

MRobi1, 2023-11-09:
Just chiming in to say I've got the same issue with 3 [LED2003G10](https://www.zigbee2mqtt.io/devices/LED2003G10.html#ikea-led2003g10) bulbs. 

For me I've got a motion automation in HA that turns the lights on/off. They turn on reliably, however the light itself often stays on while reporting a state of off in both HA and Z2M.

davidorlea, 2023-11-09:
Just to provide some context since some of you mentioned the Adaptive Lighting component of Home Assistant: IKEA Tradfri lights might be unresponsive and report erroneous states due to dropped commands during ongoing (color) transitions!

For reference: https://github.com/basnijholt/adaptive-lighting#bulb-bulb-specific-issues

MRobi1, 2023-11-09:
Great data point. However I don't think it's the root cause of the issue. I was a few days before setting up adaptive lighting in this area and had the issue right from adding the bulbs onto the network. There are also a few above who aren't using it at all. However I will hapilly implement these recommendations and see if it makes any difference!

Stooovie, 2023-11-19:
Same issue, IKEA LED2003G10 - state not correctly reflected in dashboard and bulbs turning on by binding even when only LevelCtrl is bound.

MRobi1, 2023-11-19:
> For reference: https://github.com/basnijholt/adaptive-lighting#bulb-bulb-specific-issues
I made these adjustments to my adaptive lighting config. It did not completely solve the issue, but I've gone from working properly about 50% of the time to probably around 98% of the time. I've only noticed the lights staying on while in the wrong state maybe once or twice in the past week. It's been a significant and noticeable improvement.

yschen5812, 2023-12-19:
Same with my ikea L1529 panels. I have two of them and behave the same. Sometimes when the lights are triggered I don't find anything in the logs. I assume it's some zigbee binding that is not accessible via the zigbee2mqtt dashboard. I have a couple aqara motion sensors around so maybe there are some unwanted binding.

legalgig, 2024-01-08:
After reading of some other issues with IKEA bulbs I think I've found the issue!
I have 2x LED2004G8 which were grouped in zigbee2mqtt. I've removed that group and set Transition to 0 for all bulbs (I've grouped them in Home Assistant). 
It seems to be working reliably for the past 4 days

psi-4ward, 2024-04-03:
Similar problem here with L1528 v2.3.087

Switched the panel off and update the state: receives `on` but is actually off

```
Debug 2024-04-03 19:03:32Received MQTT message on 'z2m/Küche Decke/get' with data '{"state":""}'
Debug 2024-04-03 19:03:32Publishing get 'get' 'state' to 'Küche Decke'
Debug 2024-04-03 19:03:32Received Zigbee message from 'Küche Decke', type 'readResponse', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Info 2024-04-03 19:03:32MQTT publish: topic 'z2m/Küche Decke', payload '{"brightness":254,"color_mode":"color_temp","color_options":null,"color_temp":250,"last_seen":"2024-04-03T19:03:32+02:00","level_config":{"on_level":"previous"},"linkquality":120,"power_on_behavior":"off","state":"ON","update":{"installed_version":587757105,"latest_version":587757105,"state":"idle"},"update_available":null}'
Info 2024-04-03 19:03:32MQTT publish: topic 'z2m/Küche Decke/color_mode', payload 'color_temp'
Info 2024-04-03 19:03:32MQTT publish: topic 'z2m/Küche Decke/state', payload 'ON'
```

closed 2024-11-01 by github-actions[bot]