Koenkk/zigbee2mqtt #16389: [New device support]: Bosch Door/window contact II (RBSH-SWD-ZB)

g-jaeger, 2023-01-27:
### Link

https://www.bosch-smarthome.com/uk/en/products/devices/doorwindow-contact/

### Database entry

{"id":2,"type":"EndDevice","ieeeAddr":"0x18fc26000000a613","nwkAddr":59315,"manufId":4617,"manufName":"BOSCH","powerSource":"Battery","modelId":"RBSH-SWD-ZB","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,32,1280,2821,64685],"outClusterList":[25],"clusters":{"64685":{"attributes":{"1":0}},"genBasic":{"attributes":{"modelId":"RBSH-SWD-ZB","manufacturerName":"BOSCH","powerSource":3,"zclVersion":3,"appVersion":1,"stackVersion":5,"hwVersion":1,"dateCode":"20220602","swBuildId":"1.18.01"}},"genPowerCfg":{"attributes":{"batteryAlarmState":0}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b002590ef10","zoneState":1}},"genPollCtrl":{"attributes":{"checkinInterval":6480}}},"binds":[{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0x00124b002590ef10","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":5,"hwVersion":1,"dateCode":"20220602","swBuildId":"1.18.01","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1674764688979,"defaultSendRequestWhen":"active"}

### Comments

Please consider to add this device. With the attached converter, I could manage to make the base functionality work.

TIA,
 Gerhard

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;

const fzLocal = {    
    bosch_contact: {
        cluster: 'ssIasZone',
        type: 'commandStatusChangeNotification',
        convert: (model, msg, publish, options, meta) => {
            const zoneStatus = msg.data.zonestatus;
            const lookup = {0: 'none', 1: 'single', 2: 'long'};
            return {
                contact: !((zoneStatus & 1) > 0),
                battery_low: (zoneStatus & 1<<3) > 0,
                action: lookup[(zoneStatus >> 11) & 3],
            };
        },
    },
};

const definition = {
    zigbeeModel: ['RBSH-SWD-ZB'],
    model: 'BSEN-C2',
    vendor: 'Bosch',
    description: 'Bosch Door/window contact II',
    fromZigbee: [fzLocal.bosch_contact],
    toZigbee: [],
    exposes: [e.battery_low(), e.contact(), e.action(['single', 'long'])],
};

module.exports = definition;
```

Koenkk, 2023-01-29:
added, thanks!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-01-29 by Koenkk

bonjour81, 2023-02-18:
Hello, is there something else to do in addition to add the external converter?

bonjour81, 2023-02-19:
Found the issue, it need a pairing procedure similar to https://www.zigbee2mqtt.io/devices/8750001213.html.
I succeed pairing following this.

g-jaeger, 2023-02-21:
Hi,
In fact you are right. The documentation should be improved.

bonjour81, 2023-02-21:
Done ;)

g-jaeger, 2023-02-21:
It's also the only one I have. Thanks for taking care.

bonjour81, 2023-02-21:
And thanks for the external converter! It was a long time I was looking for a z2m door sensor that can fit for outdoor use :).