Koenkk/zigbee2mqtt #16392: AV2010/32 stops sending data

andyinno, 2023-01-27:
### What happened?

There is already a closed issue related to this, without solutions.

I have 4 of them. After some time, two of them stops sending data. If I factory reset them, they works for some time and then they stops again.

I would like to understand if I can do something in order to diagnose why this is happening. Logs at the moment does not show anything.

### What did you expect to happen?

Have them work until battery dies and then replace batteries and have them working again

### How to reproduce it (minimal and precise)

Pair the device, wait a couple of weeks

### Zigbee2MQTT version

1.29.2.1

### Adapter firmware version

20211217

### Adapter

zStack3x0 zig ha za gaa

andyinno, 2023-01-27:
Other closed issue is: #4838

andyinno, 2023-03-01:
@sjorge sorry for addressing you directly. I saw that you pushed some changes in the last z2m release related to this device. Do you have any feedback related to what I am noticing?

Thanks in advance

sjorge, 2023-03-01:
Not really, mine has been working fine. But my sample size is 1.

You could try and hit the reconfigure button if one of yours steps and see if that helps.

andyinno, 2023-03-01:
Thank you for your reply.

I already tried that. It does not work or change anything.

The only thing that works is:

- remove the device from z2m
- reset the device
- join it again
- then it works for another couple of weeks, sometimes more.

At the moment I have 4 of them. Two (the nearest and the farest from the pi) never had an issue. The two in the middle shows the issue

andyinno, 2023-03-01:
At the moment they are 3 on hue and one on ikea.

I have basically all hue except two ikea.

I will check if they are the cause when it will happen.

In case. Can I force another router without removing the lamp?

Thanks again

sjorge, 2023-03-01:
Mine is also hue, ubisys and a few ikea.And unfortunately no, they can switch routers on their own and you cant really force a route.~ sjorge

closed 2023-04-09 by github-actions[bot]