Koenkk/zigbee2mqtt #16395: Philips Hue Motion Sensor (SML001 & SML003) publishes 14 duplicate MQTT messages after 6 seconds of detecting occupancy

rhtenhove, 2023-01-28:
### What happened?

I have a motion sensor in my office detecting if I'm still there, to keep the light on. Whenever `occupancy` is true, 6 seconds later, 14 duplicate MQTT messages are published. These are different than the 6 seconds earlier one (`linkquality` is usually slightly different).

These are indeed published according to MQTT explorer, so it's not a logging bug.

### What did you expect to happen?

Publish just one message.

### How to reproduce it (minimal and precise)

- Have an SML001 or SML003
- Place it where you're sitting for a while
- Watch the logs

### Zigbee2MQTT version

1.29.2-dev commit: 6ce2d2e (also happens on stable)

### Adapter firmware version

20220219

### Adapter

SONOFF ZigBee 3.0 Dongle Plus-P (CC2652P)

### Debug log

[hue_log.txt](https://github.com/Koenkk/zigbee2mqtt/files/10527100/hue_log.txt)

Removed data which would show a bit too much about my setup

bdewacht, 2023-01-31:
Maybe related to #16342 ?

itssimon, 2023-03-26:
I'm having the same problem with my Philips Hue motion sensors. Were you able to resolve this?

itssimon, 2023-03-26:
/reopen

rhtenhove, 2023-03-30:
Haven't resolved it, but also haven't bothered as it hasn't seemed to cause any real problems so far.

bdewacht, 2023-03-30:
Still having these issues. In my case it has impact, as with adaptive
lighting this amounts to an almost infinite amount of messages.

itssimon, 2023-04-07:
/reopen

itssimon, 2023-04-08:
Thanks for reopening @Koenkk. I'll try to add a bit more info to hopefully help with getting to the bottom of this.

Below is are logs from earlier today when a number of motion sensors were spamming the network.

https://pastebin.com/KFcAipkj

Example excerpt:
```
debug 2023-04-08 18:33:44: Received Zigbee message from 'Corridor front motion sensor', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":0}' from endpoint 2 with groupID 0
debug 2023-04-08 18:33:44: Received Zigbee message from 'Corridor front motion sensor', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":0}' from endpoint 2 with groupID 0
debug 2023-04-08 18:33:44: Received Zigbee message from 'Corridor front motion sensor', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":0}' from endpoint 2 with groupID 0
debug 2023-04-08 18:33:44: Received Zigbee message from 'Corridor front motion sensor', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":0}' from endpoint 2 with groupID 0
debug 2023-04-08 18:33:44: Received Zigbee message from 'Corridor front motion sensor', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":0}' from endpoint 2 with groupID 0
debug 2023-04-08 18:33:44: Received Zigbee message from 'Corridor front motion sensor', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":0}' from endpoint 2 with groupID 0
debug 2023-04-08 18:33:44: Received Zigbee message from 'Corridor front motion sensor', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":0}' from endpoint 2 with groupID 0
debug 2023-04-08 18:33:44: Received Zigbee message from 'Corridor front motion sensor', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":0}' from endpoint 2 with groupID 0
debug 2023-04-08 18:33:44: Received Zigbee message from 'Corridor front motion sensor', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":0}' from endpoint 2 with groupID 0
debug 2023-04-08 18:33:44: Received Zigbee message from 'Corridor front motion sensor', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":0}' from endpoint 2 with groupID 0
debug 2023-04-08 18:33:44: Received Zigbee message from 'Corridor front motion sensor', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":0}' from endpoint 2 with groupID 0
debug 2023-04-08 18:33:44: Received Zigbee message from 'Corridor front motion sensor', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":0}' from endpoint 2 with groupID 0
debug 2023-04-08 18:33:44: Received Zigbee message from 'Corridor front motion sensor', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":0}' from endpoint 2 with groupID 0
debug 2023-04-08 18:33:44: Received Zigbee message from 'Corridor front motion sensor', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":0}' from endpoint 2 with groupID 0
```

Strangely it seems to always be exactly 14 duplicated messages.

In an attempt to tame the spamming, at least on the MQTT side, I have configured a 50ms debounce for all of my Philips Hue motion sensor as shown below, which is enough to limit the MQTT messages generated. That also indicates that the Zigbee messages are coming in at extremely short intervals, or almost at the same time. Unfortunately this doesn't solve the problem of delays in the Zigbee network.

```yaml
  '0x0000000000000000':
    friendly_name: Powder room motion sensor
    temperature_calibration: 1.8
    temperature_precision: 1
    debounce: 0.05
    debounce_ignore:
      - occupancy
      - temperature
      - illuminance
      - illuminance_lux
```

I suspect that this spamming occurs in particular when the motion sensors are waking up after a longer sleep period, e.g. when we come home after being out for a few hours.

The impact of this is hard to quantify, but I frequently get long delays (> 5 sec) in lights turning on after motion was detected, and that always coincides with one or multiple motion sensors spamming at that time. At other times there is no noticeable delay.

Details about my setup:

**Adapter**: SONOFF ZigBee 3.0 Dongle Plus-P (CC2652P)
**Motion sensors**: 18x Philips Hue Motion Sensors (SML001)
**Number of devices on Zigbee network**: 110 (47 of which are Philips Hue downlights throughout the house, so very good router coverage)
**Zigbee2MQTT version**: 1.30.3-1
**Device firmware versions**: All up to date

If there's anything else I can provide that would help, please let me know.

Koenkk, 2023-04-08:
@itssimon could you also provide the herdsman debug logging of this?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

itssimon, 2023-04-09:
Sure thing, see https://pastebin.com/bDNg0Fwy

Koenkk, 2023-04-10:
@itssimon it seems to send the message to every endpoint of the coordinator (which is behaviour I haven't seen with any other device yet). I also noticed it only seems to happen for the illuminance (`msIlluminanceMeasurement`)? Can you try to unbind the illuminance in the z2m frontend -> device -> bind tab and see if this does not occur anymore with any other messages?

itssimon, 2023-04-10:
It actually also happens for `msOccupancySensing`, however it appears it may not be happening for `msTemperatureMeasurement`. Interesting. Here's a longer log excerpt demonstrating that: https://pastebin.com/NZZELVie.

I have temporarily unbound Illuminance for a few of my motion sensors and will report back if that makes any difference. I do need the illuminance measurements for my automations though, so can't leave it like this for too long.

itssimon, 2023-04-10:
Unbinding illuminance didn't do the trick. I wasn't getting illuminance updates of course, but occupancy was still sent 14 times.

Koenkk, 2023-04-11:
@itssimon I have the feeling this is a bug in the sensors firmware, is it already on the latest firmware? (from the OTA tab)

itssimon, 2023-04-12:
I think you might be right. They are all on the latest firmware already unfortunately. Is there an easy way to roll back to previous firmware versions using Z2M?

Also, these motion sensors are quite popular I believe, so curious why this issue has not come up before. Do you happen to have a contact at Signify that we could raise this with?

itssimon, 2023-04-17:
I do, unfortunately there is no new firmware version available. It's on 1.1.27575 (reported as 6.1.1.27575 in Z2M), which is the latest version.

itssimon, 2023-04-27:
I've reached out to the Philips Hue support, but without luck. They will only support the Hue Bridge as the controller. Fair enough.

I also realised that the current firmware 1.1.27575 was released in 2019, so if this is a bug in the firmware it would be a very long standing one. I was using deCONZ with a ConBee II stick until I migrated to Z2M and CC2652P a few weeks ago and didn't have these issues with the Hue motion sensors then.

With that in mind I suspect there could be more to it than a bug in the device firmware, or at least in deCONZ it didn't affect the network's reliability as it does with Z2M.

@Koenkk any chance you could investigate further to see if there's something that can be done about this in Z2M or the CC2652P firmware? I'm happy to buy and send you one of these sensors for that purpose. And obviously happy to provide more logs as well if that's helpful?

I have 17 of these sensors in our house and it's really affecting the reliability of our lights, so I'm quite desperate to find a solution.

Koenkk, 2023-04-27:
I've just checked with my Hue outdoor sensor and it has the same behaviour (SML002).  I don't expect we can prevent the device form sending these messages. But what we can do is:

- Since the messages have the same transaction number we can prevent multiple messages with the same transaction number from being published. This should be fixed with this ext converter: https://gist.github.com/Koenkk/3f76a994d2c60087a2f7b97cde589fcb
  - save this as file next to `configuration.yaml` as `ext_converter.js`
  - add it to `configuration.yaml`:
  ```yaml
  external_converters:
    - ext_converter.js
  ```
  - start z2m, check if issue is fixed
- regarding fw stability, did you already check the latest test firmware? https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin

itssimon, 2023-04-28:
Thanks for looking into it, really appreciate it. The external converter is a good idea and I think I've already achieved the same thing by configuring a very short debounce for all my motion sensors:

```yaml
  '0x0000000000000000':
    friendly_name: Corridor back motion sensor 2
    temperature_calibration: 1.8
    temperature_precision: 1
    debounce: 0.05
    debounce_ignore:
      - occupancy
      - temperature
      - illuminance
      - illuminance_lux
```

The logs indicate that this prevents the duplicate messages from being published via MQTT:

```
debug 2023-04-28 19:04:00: Received Zigbee message from 'Corridor back motion sensor 2', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":13346}' from endpoint 2 with groupID 0
debug 2023-04-28 19:04:00: Received Zigbee message from 'Corridor back motion sensor 2', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":13346}' from endpoint 2 with groupID 0
debug 2023-04-28 19:04:00: Received Zigbee message from 'Corridor back motion sensor 2', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":13346}' from endpoint 2 with groupID 0
debug 2023-04-28 19:04:00: Received Zigbee message from 'Corridor back motion sensor 2', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":13346}' from endpoint 2 with groupID 0
debug 2023-04-28 19:04:00: Received Zigbee message from 'Corridor back motion sensor 2', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":13346}' from endpoint 2 with groupID 0
debug 2023-04-28 19:04:00: Received Zigbee message from 'Corridor back motion sensor 2', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":13346}' from endpoint 2 with groupID 0
debug 2023-04-28 19:04:00: Received Zigbee message from 'Corridor back motion sensor 2', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":13346}' from endpoint 2 with groupID 0
debug 2023-04-28 19:04:00: Received Zigbee message from 'Corridor back motion sensor 2', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":13346}' from endpoint 2 with groupID 0
debug 2023-04-28 19:04:00: Received Zigbee message from 'Corridor back motion sensor 2', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":13346}' from endpoint 2 with groupID 0
debug 2023-04-28 19:04:00: Received Zigbee message from 'Corridor back motion sensor 2', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":13346}' from endpoint 2 with groupID 0
debug 2023-04-28 19:04:00: Received Zigbee message from 'Corridor back motion sensor 2', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":13346}' from endpoint 2 with groupID 0
debug 2023-04-28 19:04:00: Received Zigbee message from 'Corridor back motion sensor 2', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":13346}' from endpoint 2 with groupID 0
debug 2023-04-28 19:04:00: Received Zigbee message from 'Corridor back motion sensor 2', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":13346}' from endpoint 2 with groupID 0
debug 2023-04-28 19:04:00: Received Zigbee message from 'Corridor back motion sensor 2', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":13346}' from endpoint 2 with groupID 0
info  2023-04-28 19:04:00: MQTT publish: topic 'zigbee2mqtt/Corridor back motion sensor 2', payload '{"battery":81.5,"illuminance":13346,"illuminance_lux":22,"led_indication":null,"linkquality":116,"motion_sensitivity":"high","occupancy":true,"occupancy_timeout":0,"temperature":25.4,"update":{"installed_version":1107323831,"latest_version":1107323831,"state":"idle"},"update_available":null}'
```

Unfortunately it hasn't fixed or even improved the delays/unreliability in the Zigbee network. I suspect the bottleneck is in the coordinator, which appears to get overwhelmed by too many broadcast messages and fills up a buffer, leading to lots of `0x11: BUFFER_FULL` errors, e.g.:

```
error 2023-04-08 19:40:40: Publish 'set' 'brightness' to 'Corridor back' failed: 'Error: Command 5 genLevelCtrl.moveToLevelWithOnOff({"level":159,"transtime":5}) failed (SREQ '--> AF - dataRequestExt - {"dstaddrmode":1,"dstaddr":"0x0000000000000005","destendpoint":255,"dstpanid":0,"srcendpoint":1,"clusterid":8,"transid":24,"options":0,"radius":30,"len":6,"data":{"type":"Buffer","data":[17,197,4,159,5,0]}}' failed with status '(0x11: BUFFER_FULL)' (expected '(0x00: SUCCESS)'))'
debug 2023-04-08 19:40:40: Error: Command 5 genLevelCtrl.moveToLevelWithOnOff({"level":159,"transtime":5}) failed (SREQ '--> AF - dataRequestExt - {"dstaddrmode":1,"dstaddr":"0x0000000000000005","destendpoint":255,"dstpanid":0,"srcendpoint":1,"clusterid":8,"transid":24,"options":0,"radius":30,"len":6,"data":{"type":"Buffer","data":[17,197,4,159,5,0]}}' failed with status '(0x11: BUFFER_FULL)' (expected '(0x00: SUCCESS)'))
    at Object.func (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:322:27)
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
```

Do you think there's something that could be done in the firmware to better handle these volumes of identical broadcast messages? Maybe a duplicate detection there, similar to what you've done in the external converter, could help in getting these messages out of the queue quicker?

I'm going to try the 20230410 firmware now and will report back on that.

Koenkk, 2023-04-28:
We can increase the buffer sizes, let me know your experiences with 20230410

itssimon, 2023-04-30:
Unfortunately with 20230410 the network has become even less reliable. Delays have increased in frequency and severity. I had to roll back.

Increasing the buffer size sounds worth a shot!

Koenkk, 2023-04-30:
Can you try with the following fw: 
[znp_CC1352P_2_LAUNCHXL_tirtos7_ticlang.hex.zip](https://github.com/Koenkk/zigbee2mqtt/files/11362025/znp_CC1352P_2_LAUNCHXL_tirtos7_ticlang.hex.zip) (sonoff dongle-P)

itssimon, 2023-05-02:
With that firmware Z2M crashes shortly after starting. These are the log messages:

```log
[23:15:15] INFO: Preparing to start...
[23:15:15] INFO: Socat not enabled
[23:15:15] INFO: Zigbee Herdsman debug logging enabled
[23:15:16] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:debug 2023-05-02 23:15:16: Loaded state from file /config/zigbee2mqtt/state.json
Zigbee2MQTT:info  2023-05-02 23:15:16: Logging to console and directory: '/config/zigbee2mqtt/log/2023-05-02.23-15-16' filename: log.txt
Zigbee2MQTT:debug 2023-05-02 23:15:16: Removing old log directory '/config/zigbee2mqtt/log/2023-04-28.23-34-49'
Zigbee2MQTT:info  2023-05-02 23:15:16: Starting Zigbee2MQTT version 1.30.3 (commit #unknown)
Zigbee2MQTT:info  2023-05-02 23:15:16: Starting zigbee-herdsman (0.14.103)
Zigbee2MQTT:debug 2023-05-02 23:15:16: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/config/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","databasePath":"/config/zigbee2mqtt/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyUSB0"}}'
2023-05-02T13:15:16.916Z zigbee-herdsman:adapter Failed to validate path: 'Error: spawn udevadm ENOENT'
2023-05-02T13:15:16.916Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/ttyUSB0"},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}'
2023-05-02T13:15:16.916Z zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with {"path":"/dev/ttyUSB0","baudRate":115200,"rtscts":false,"autoOpen":false}
2023-05-02T13:15:16.920Z zigbee-herdsman:adapter:zStack:znp:log Serialport opened
2023-05-02T13:15:16.920Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
2023-05-02T13:15:16.921Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
2023-05-02T13:15:17.173Z zigbee-herdsman:adapter:zStack:znp:log Writing CC2530/CC2531 skip bootloader payload
2023-05-02T13:15:17.173Z zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239]
2023-05-02T13:15:18.174Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
2023-05-02T13:15:18.175Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
2023-05-02T13:15:18.426Z zigbee-herdsman:adapter:zStack:znp:log Skip bootloader for CC2652/CC1352
2023-05-02T13:15:18.880Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
2023-05-02T13:15:18.881Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
```

I'll provide herdsman debug logs shortly.

itssimon, 2023-05-13:
Apologies for the delay, here's a complete log file from the zigbee2mqtt container with herdsman debug logging enabled: https://www.dropbox.com/s/2izhrxjpw9m53np/z2m_debug.log?dl=0

Koenkk, 2023-05-14:
Did you already check with the [`20230507`](https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin) firmware?

itssimon, 2023-05-15:
Just installed 20230507. Will report back in a few days.

closed 2023-06-23 by github-actions[bot]

itssimon, 2023-06-23:
Sorry, this dropped off my radar. The firmware 20230507 didn't make a difference in regards to this problem, but otherwise has been running fine since I installed it about 5 weeks ago.

However, just FYI for anyone else coming across this issue, I was able to address the issues of network congestion and delays by doing two things:

1) Disable temperature and illuminance reporting for the Hue motion sensors where possible. I ended up getting some additional Aqara temperature & humidity sensors instead.
2) Rearchitected my light groups to end up with less groups overall. That did require some compromises but I believe this has lead to the biggest improvement in terms of responsiveness of the network.

oorakci, 2023-12-17:
> I've just checked with my Hue outdoor sensor and it has the same behaviour (SML002). I don't expect we can prevent the device form sending these messages. But what we can do is:
>
> * Since the [...]

Hello @Koenkk,

When I try the code above, I get the following error. Could there be a mistake in the script?

> ReferenceError: philips is not defined
>      at /app/data/extension/externally-loaded.js:39:38
>      at Script.runInContext (node:vm:135:12)

0rsa, 2024-07-14:
I guess I have the same kind of issue regarding the illuminance with a Philips SML003
When a motion is detected, I receive at least these 2 payloads at same time:

>last_seen: '2024-07-14T06:41:59+02:00'
>illuminance: 20372

>lastupdate: '2024-07-14T06:41:59+02:00'
>illuminance: 18019

A such difference regarding the illuminance with the same 'last_seen' value makes no sense.