Koenkk/zigbee2mqtt #16398: Failed to call 'Bridge' 'start'

OlsiIgitt, 2023-01-28:
### What happened?

Did an update today from v1.29.2 to v1.29.2: 
z2m v1.29.2 on Windows ->
gitt pull 
npm ci
npm start

One file was updated; after 'npm start ' z2m reported the error:
Failed to call 'Bridge' 'start' (TypeError: Cannot read properties of undefined (reading 'ieeeAddr') ... see Debug log below

### What did you expect to happen?

I removed the line 
#659:                 disabled: !!device.options.disabled,
in file 'bridge.ts'.
After this z2m operated normally without errors...

### How to reproduce it (minimal and precise)

see above...

### Zigbee2MQTT version

1.29.2  commit: 1402139

### Adapter firmware version

0x26720700

### Adapter

ConBee2/

### Debug log

error 2023-01-28 12:40:02: Failed to call 'Bridge' 'start' (TypeError: Cannot read properties of undefined (reading 'ieeeAddr')
    at Bridge.publishInfo (C:\Progs\Z2M\lib\extension\bridge.ts:577:81)
    at Bridge.start (C:\Progs\Z2M\lib\extension\bridge.ts:108:20)
    at processTicksAndRejections (node:internal/process/task_queues:96:5)
    at Controller.callExtensions (C:\Progs\Z2M\lib\controller.ts:316:17)
    at Controller.start (C:\Progs\Z2M\lib\controller.ts:153:9)
    at start (C:\Progs\Z2M\index.js:107:5))

ivanfmartinez, 2023-02-09:
same here upgrading from 1.29.1 to 1.30.1

using deconz adapter

error 2023-02-09 11:41:52: Failed to call 'Bridge' 'start' (TypeError: Cannot read properties of undefined (reading 'ieeeAddr')
    at Bridge.publishInfo (/home/zigbee/zigbee2mqtt/lib/extension/bridge.ts:577:81)
    at Bridge.start (/home/zigbee/zigbee2mqtt/lib/extension/bridge.ts:108:20)
    at Controller.callExtensions (/home/zigbee/zigbee2mqtt/lib/controller.ts:319:17)
    at Controller.start (/home/zigbee/zigbee2mqtt/lib/controller.ts:153:9)
    at start (/home/zigbee/zigbee2mqtt/index.js:107:5))

commenting this line 577 on bridge.ts made it start again

Koenkk, 2023-02-09:
@ivanfmartinez could you provide the herdsman debug log of this?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

ivanfmartinez, 2023-02-09:
@Koenkk I have returned the line

ieee_address: this.zigbee.firstCoordinatorEndpoint().getDevice().ieeeAddr,

and tried 4 times with the herdsman logs. And the problem does not appeared again. 

before removing the line I have tried many times with same error, maybe after executing with the current version (without that line that does not get the correct coordinator address)  something was corrected in my installation.

Here is one of the logs, maybe you can think in other options.
[herdsman.log](https://github.com/Koenkk/zigbee2mqtt/files/10699726/herdsman.log)

Koenkk, 2023-02-10:
I found the bug, fixed it!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-02-10 by Koenkk