Koenkk/zigbee2mqtt #16404: <img width="1649" alt="image" src="https://user-images.githubusercontent.com/22066360/212967529-66074707-4903-404b-8b0a-f93ad87c0d33.png">

mayongze, 2023-01-29:
[image]
red: unknown
yellow: onTransitionTime
green: offTransitionTime

https://github.com/Koenkk/zigbee-herdsman/blob/master/src/zcl/definition/cluster.ts
need to add command:
```
tuyaOnOffTransitionTime: {
    ID: 251,
    parameters: [
        { name: 'unknown', type: DataType.uint8 },
        { name: 'onTransitionTime' , type: DataType.uint24 },
        { name: 'offTransitionTime', type: DataType.uint24 },
    ],
},
```
The time uses the big endian byte from the packet sniffed, but DataType.uint24 is the small endian byte
DataType Whether a big-endian byte type is directly available?

[image]

_Originally posted by @mayongze in https://github.com/Koenkk/zigbee2mqtt/discussions/16234_

Koenkk, 2023-01-29:
Could you make a PR to add this?

mayongze, 2023-01-29:
I'll make a PR for this, but I need to figure out if there's a big-endian type
60 ea 00 => DataType.uint24
00 ea 60 => ? (DataType.be_uint24)

Koenkk, 2023-01-29:
Then a new datatype has to be added, example:

- https://github.com/Koenkk/zigbee-herdsman/blob/bb4f9fb5ad0255b98a4fb2b3cc03b3848aae8cf4/src/zcl/definition/cluster.ts#L220
- https://github.com/Koenkk/zigbee-herdsman/blob/bb4f9fb5ad0255b98a4fb2b3cc03b3848aae8cf4/src/zcl/definition/buffaloZclDataType.ts#L14
- https://github.com/Koenkk/zigbee-herdsman/blob/bb4f9fb5ad0255b98a4fb2b3cc03b3848aae8cf4/src/zcl/buffaloZcl.ts#L605
- https://github.com/Koenkk/zigbee-herdsman/blob/bb4f9fb5ad0255b98a4fb2b3cc03b3848aae8cf4/src/zcl/buffaloZcl.ts#L651

closed 2023-02-08 by Koenkk