Koenkk/zigbee2mqtt #16405: MoveToColor (0x07) from Dev Console and MQTT sends the wrong XY values

rechichidaniel, 2023-01-29:
### What happened?

Sending the `Move to Color (0x07)` command sends `colorx = 0` and `colory = 0` regardless of what colorx and colory are set to. `transtime` does send.

I tried using these settings for `Payload` but any values for colox and colory send a 0
```json
{
    "colorx":0.7010,
    "colory":0.2990,
    "transtime":0
}
```

I also put this json message in a Node Red MQTT node which produced the exact same result.
```json
{
    "command": {
        "cluster": 768,
        "command": 7,
        "payload": {
            "colorx": 0.701,
            "colory": 0.299,
            "transtime": 0
        }
    }
}
```

### What did you expect to happen?

The expectation was for `colorx = 0.7010` and `colory = 0.2990` to be sent instead of `0.000`.

### How to reproduce it (minimal and precise)

I used the Z2M Dev Console to send the above command and payload, and Wireshare with nRF52840 to observe the network traffic.

[image]

[image]

### Zigbee2MQTT version

1.29.2

### Adapter firmware version

20220219

### Adapter

TubeZB's CC2652P2 Based Coordinator

### Debug log

See screenshots.

```
Debug 2023-01-29 15:27:53Publishing 'set' 'command' to 'Z2M Test Light'
Info 2023-01-29 15:27:53Invoked '768.7' with payload '{"colorx":0.701,"colory":0.299,"transtime":0}'
```

rechichidaniel, 2023-01-29:
I'm not sure why but the Home Assistant Dev Tools menu does send the correct XY values. It's just the Z2M dev console and direct MQTT messaging that doesn't. Did I set something in my tests up wrong?

[image]

Koenkk, 2023-01-29:
The datatype of x/y is a `uint16`, not a float. Multiply the float by `65535` and round it.

rechichidaniel, 2023-01-30:
Thanks Koen. That worked. Sorry to waste your time on this one.

closed 2023-01-30 by rechichidaniel