Koenkk/zigbee2mqtt #16406: Frankever Smart Water Valve manufacturer suffix keeps changing, so device not recognized

initd0, 2023-01-29:
### What happened?

I bought https://www.zigbee2mqtt.io/devices/FK_V02.html and it was not completely recognized by mqtt, only the model and manufacturer was shown, for me it was also `modelID: 'TS0601', manufacturerName: '_TZE200_` but a different suffix of `_TZE200_XXXXXX`.  I noticed that in https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/frankever.js only 2 specific ones were added:
```
fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_wt9agwf3'},
            {modelID: 'TS0601', manufacturerName: '_TZE200_5uodvhgc'}],
```

It is to my understanding that whoever buys this device, the suffix of `_TZE200_XXXXXX` keeps changing, so for each user its unique hence why many users do not have this device recognized.

### What did you expect to happen?

I expected the device to be recognized since its already supported.

### How to reproduce it (minimal and precise)

Buy a device and watch that it is not recognized because the manufacturer suffix is different everytime.

### Zigbee2MQTT version

1.29.2

### Adapter firmware version

20210708

### Adapter

zzh!

initd0, 2023-01-29:
Is this normal behavior for most devices?  Should each user report their manufacturer id everytime they buy this device to add it  in manually or can you just do something like _TZE200_* ?

Koenkk, 2023-01-29:
This is typical TuYa behaviour, they change the manufacturer name every once in a while. There is no solution to this besides adding the new manufacturer name. If you provide the new manufacturer name I can add it.

initd0, 2023-01-29:
@Koenkk i see, ok here is the new manufacturer name: `_TZE200_1n2zev06`
Thank you!

Koenkk, 2023-01-29:
Added.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-01-29 by Koenkk

rugene76, 2024-12-02:
thanks for updating the mfg!

Do you know if the default power loss behavior of the valve can be exposed in zigbee? From what I can tell, the wifi version lets the valve power loss behavior to be changed. I'd like the valve to stay in its current state when power is lost
thanks