Koenkk/zigbee2mqtt #16412: Ikea 5-Button Remote shows 200% battery level

wiiNinja, 2023-01-29:
### What happened?

I have three Ikea 5-button remote controls. After updating remotes to the latest version 24.4.5, all of them show the battery levels above 100%, two show 200% and one shows 180%. It's obviously due to the firmware update, but I'm not sure if the battery calculation is done in the Z2M addon, or if it is getting the battery from the firmware. This is not a critical issue, but thought I'd mention it.

### What did you expect to happen?

Battery level should not exceed 100%.

### How to reproduce it (minimal and precise)

Update the Ikea 5-button remote control to the latest firmware version, then wait for the battery level to be updated.

### Zigbee2MQTT version

1.29.2

### Adapter firmware version

20220219

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/10529838/log.txt)

wiiNinja, 2023-01-29:
[image]

nonflammable, 2023-01-29:
[image]

Koenkk, 2023-01-29:
This is good as it follows the spec now, basically this line has to be removed to fix: https://github.com/Koenkk/zigbee-herdsman-converters/blob/b8abb17d881e68d10f57acf2c277259ee5f7fa39/devices/ikea.js#L633

However this will impact users on the old firmware (as those batteries will be divided by 2). I want to double check, did you do this firmware update directly via zigbee2mqtt? (the normal way)

wiiNinja, 2023-01-29:
@Koenkk - Thanks for looking into this. I updated the firmware via Zigbee2MQTT after seeing notification that an update was available.

Koenkk, 2023-01-29:
Great, pushed the fix!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-01-29 by Koenkk

MaxRower, 2023-02-02:
This seems to apply to E1525 with the new 24.4.5 firmware as well, but not to E1745, which did not get an update. 
The new firmware for E1525 makes it detect movements much faster in my option.

MaxRower, 2023-02-02:
I did switch to the new 1.30.0 release, which should include the fix, but my E1810 still shows battery values above 100%.

sjorge, 2023-02-08:
@Koenkk would adding a fzLocat.ikea_battery be an option, where we can then check the firmware and device correctly?
A lot of people have held of update due to the missing group binding support.

Koenkk, 2023-02-09:
@sjorge good idea, could you make a pr?

sjorge, 2023-02-09:
> @sjorge good idea, could you make a pr?

Sure, probably somewhere over the weekend.