Koenkk/zigbee2mqtt #16420: [New device support]: ROBB Smarrt 230V warm/koud wit LED Trafo Zigbee ROB_200-061-0

bjkoekkoek, 2023-01-30:
### Link

https://www.robbshop.nl/robb-smarrt-slimme-led-driver-warmkoud-wit-en-voeding-zigbee?sqr=ROBB%20Smarrt%20230V%20warm%2Fkoud%20wit%20LED%20Trafo%20Zigbee%20ROB_200-061-0

### Database entry

{"id":19,"type":"Router","ieeeAddr":"0x04cd15fffeb4b831","nwkAddr":14835,"manufId":4644,"manufName":"ROBB  smarrt","powerSource":"Mains (single phase)","modelId":"ROB_200-061-0","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,3,4,5,6,8,768,2821,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"appVersion":0,"stackVersion":0}},"lightingColorCtrl":{"attributes":{"colorCapabilities":16,"colorTempPhysicalMin":160,"colorTempPhysicalMax":450}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":102,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":0,"hwVersion":1,"dateCode":"NULL","swBuildId":"2.9.2_r60","zclVersion":3,"interviewCompleted":true,"meta":{"configured":88764544},"lastSeen":1675065027938,"defaultSendRequestWhen":"immediate"}

### Comments

I found another [issue](https://github.com/Koenkk/zigbee2mqtt/issues/12158) on GitHub regarding this device after I tried adding the external converter as shown below. I can see in the logging the device is now "supported" and I can see the device in HA and it seems like you can control the device using HA but the lights are not turning off or on. The logging of z2m keeps displaying the following errors:

debug 2023-01-29 19:14:35: Received MQTT message on 'zigbee2mqtt/0x04cd15fffeb4b831/set' with data '{"state":"ON","brightness":129}'
error 2023-01-29 19:14:35: No converter available for 'brightness' (129)
error 2023-01-29 19:14:35: No converter available for 'state' ("ON")
debug 2023-01-29 19:14:45: Received MQTT message on 'zigbee2mqtt/0x04cd15fffeb4b831/set' with data '{"state":"OFF"}'
error 2023-01-29 19:14:45: No converter available for 'state' ("OFF")

Any help would be highly appreciated :) 

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
	zigbeeModel: ['ROB_200-061-0'],
	model: 'ROB_200-061-0',
	description: '50W ZigBee CCT LED Driver(constant voltage)' ,
	vendor: 'ROBB smarrt',
	fromZigbee: [], // We will add this later
	toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
	extend: extend.light_onoff_brightness_colortemp(),
};

module.exports = definition;
```

### Supported color modes

color temperature

### Color temperature range

{"colorTempPhysicalMin":160,"colorTempPhysicalMax":450}

Koenkk, 2023-01-30:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-01-30 by Koenkk

joene-nl, 2023-01-30:
> Added!
> 
> Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Uhm ok? I was asking for some help because it was not working with the external converter file. Have you done some modifications?

Koenkk, 2023-01-30:
It will works after you remove the following from your external converter:

```js
fromZigbee: [], // We will add this later
	toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
```

joene-nl, 2023-01-30:
Awesome! Will try later tonight! 👌

joene-nl, 2023-01-30:
Works like a charm :)