Koenkk/zigbee2mqtt #16421: TS011F_plug_1 has no OTA update at all.

thierrybla, 2023-01-30:
### What happened?

My Tuya plug is seen and works correctly.
I want to update them but it shows the devices with no "Firmware build date" and no "Firmware version".

When I click on "Check for update" it says: "No update available for 'device'".

Is this just a UI bug? Does the update system work but it's not showing the version?

[image]

### What did you expect to happen?

Show the correct "Firmware build date" and "Firmware version".

### How to reproduce it (minimal and precise)

Add a TS011F_plug_1 to Zigbee2Mqtt and look at the OTA tab.

### Zigbee2MQTT version

1.29.2

### Adapter firmware version

20210708

### Adapter

SONOFF ZB Dongle-P USB Dongle Plus

### Debug log

Zigbee2MQTT:debug 2023-01-30 09:55:02: Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/check' with data '{"id":"Dishwasher plug","transaction":"giiv4-45"}'
Zigbee2MQTT:info  2023-01-30 09:55:02: Checking if update available for 'Dishwasher plug'
Zigbee2MQTT:debug 2023-01-30 09:55:02: Check if update available for '0xa4c13873c86da4c5' (TS011F)
Zigbee2MQTT:debug 2023-01-30 09:55:02: Using endpoint '1'
Zigbee2MQTT:debug 2023-01-30 09:55:02: Received Zigbee message from 'Dishwasher plug', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":192,"imageType":54179,"manufacturerCode":4417}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2023-01-30 09:55:02: MQTT publish: topic 'zigbee2mqtt/Dishwasher plug', payload '{"child_lock":"UNLOCK","current":0,"energy":108.57,"indicator_mode":"off/on","last_seen":"2023-01-30T09:55:02+01:00","linkquality":54,"power":0,"power_outage_memory":"on","state":"ON","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":null,"voltage":232}'
Zigbee2MQTT:debug 2023-01-30 09:55:02: Got OTA request '{"fieldControl":0,"manufacturerCode":4417,"imageType":54179,"fileVersion":192}'
Zigbee2MQTT:debug 2023-01-30 09:55:02: ZigbeeOTA: downloaded main index
Zigbee2MQTT:debug 2023-01-30 09:55:02: Is new image available for '0xa4c13873c86da4c5', current '{"fieldControl":0,"manufacturerCode":4417,"imageType":54179,"fileVersion":192}', latest meta '{"fileVersion":192,"fileSize":307682,"url":"https://images.tuyaeu.com/smart/firmware/upgrade/20220907/1662545193-oem_zg_tl8258_plug_OTA_3.0.0.bin","sha512":"01939ca4fc790432d2c233e19b2440c1e0248d2ce85c9299e0b88928cb2341de675350ac7b78187a25f06a2768f93db0a17c4ba950b60c82c072e0c0833cfcfb"}'
Zigbee2MQTT:debug 2023-01-30 09:55:02: Update available for '0xa4c13873c86da4c5': NO
Zigbee2MQTT:info  2023-01-30 09:55:02: No update available for 'Dishwasher plug'
Zigbee2MQTT:info  2023-01-30 09:55:02: MQTT publish: topic 'zigbee2mqtt/Dishwasher plug', payload '{"child_lock":"UNLOCK","current":0,"energy":108.57,"indicator_mode":"off/on","last_seen":"2023-01-30T09:55:02+01:00","linkquality":54,"power":0,"power_outage_memory":"on","state":"ON","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":null,"voltage":232}'
Zigbee2MQTT:info  2023-01-30 09:55:02: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"Dishwasher plug","updateAvailable":false},"status":"ok","transaction":"giiv4-45"}'
Zigbee2MQTT:debug 2023-01-30 09:55:07: Received Zigbee message from 'Livingroom motion sensor', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":20318}' from endpoint 2 with groupID 0

janb14, 2023-02-14:
Can confirm, they can be updated but dont show their versions in gui. Maybe it has to do with the update. When viewing the state in gui, they didnt show any firmware details. After the update they did but only in the state tab, not the OTA section. Maybe some database entry needs to be updated or general version string parsing is missing.

tetsuo55, 2023-03-17:
I can confirm also

Impact123, 2023-04-17:
Unstale

tjorim, 2023-05-09:
Seeing the same here, Zigbee2MQTT version 1.30.4.
However, I was able to update the firmware of the smart plugs I received.
[image]

vincele, 2023-06-06:
+2
I cannot see the version, but a manual check and update "worked" (on the 20A).

* Zigbee Model: TS011F
* SONOFF ZB Dongle-P USB Dongle Plus
* Zigbee Manufacturer: _TZ3000_hdopuwv6 (16A-rated model)
* Zigbee Manufacturer: _TZ3000_gvn91tmx (20A-rated model)

Impact123, 2023-07-07:
Unstale.

tetsuo55, 2023-07-30:
Unstaie

tjorim, 2023-09-08:
Still present in Zigbee2MQTT version 1.33.0.

Impact123, 2023-10-10:
Unstale.

chemuki205, 2023-11-21:
I confirm

**Zigbee2MQTT version**
[1.33.2] commit: [9996c93]

**Adapter firmware version**
20230507

**Adapter**
SONOFF ZB Dongle-P USB Dongle Plus

[image]

**After Check updates...**
```
info  2023-11-21 23:33:01: Checking if update available for 'Enchufe_lavavajillas'
info  2023-11-21 23:33:01: Checking if update available for 'Llave_general_agua'
info  2023-11-21 23:33:01: Checking if update available for 'Enchufe_garaje'
info  2023-11-21 23:33:01: Update available for 'Enchufe_lavavajillas'
info  2023-11-21 23:33:01: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"Enchufe_lavavajillas","updateAvailable":true},"status":"ok","transaction":"3rwx8-16"}'
info  2023-11-21 23:33:02: Update available for 'Enchufe_garaje'
info  2023-11-21 23:33:02: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"Enchufe_garaje","updateAvailable":true},"status":"ok","transaction":"3rwx8-18"}'
info  2023-11-21 23:33:02: Update available for 'Llave_general_agua'
info  2023-11-21 23:33:02: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"Llave_general_agua","updateAvailable":true},"status":"ok","transaction":"3rwx8-17"}'
```

[image]

[image]

**Updating Enchufe_garaje**

```
Info 2023-11-21 23:50:51Updating 'Enchufe_garaje' to latest firmware
Info 2023-11-21 23:50:52Update of 'Enchufe_garaje' at 24.78%, ≈ 0 minutes remaining
Info 2023-11-21 23:50:52MQTT publish: topic 'zigbee2mqtt/Enchufe_garaje', payload '{"child_lock":"UNLOCK","current":0,"energy":0,"indicator_mode":"off/on","linkquality":116,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":77,"latest_version":192,"progress":24.78,"remaining":0,"state":"updating"},"update_available":null,"voltage":225}'
Info 2023-11-21 23:51:22Update of 'Enchufe_garaje' at 26.47%, ≈ 1 minutes remaining
Info 2023-11-21 23:51:22MQTT publish: topic 'zigbee2mqtt/Enchufe_garaje', payload '{"child_lock":"UNLOCK","current":0,"energy":0,"indicator_mode":"off/on","linkquality":116,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":77,"latest_version":192,"progress":26.47,"remaining":84,"state":"updating"},"update_available":null,"voltage":225}'
Info 2023-11-21 23:51:52Update of 'Enchufe_garaje' at 28.24%, ≈ 3 minutes remaining
Info 2023-11-21 23:51:52MQTT publish: topic 'zigbee2mqtt/Enchufe_garaje', payload '{"child_lock":"UNLOCK","current":0,"energy":0,"indicator_mode":"off/on","linkquality":116,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":77,"latest_version":192,"progress":28.24,"remaining":153,"state":"updating"},"update_available":null,"voltage":224}'
Info 2023-11-21 23:52:22Update of 'Enchufe_garaje' at 30.03%, ≈ 4 minutes remaining
Info 2023-11-21 23:52:22MQTT publish: topic 'zigbee2mqtt/Enchufe_garaje', payload '{"child_lock":"UNLOCK","current":0,"energy":0,"indicator_mode":"off/on","linkquality":116,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":77,"latest_version":192,"progress":30.03,"remaining":211,"state":"updating"},"update_available":null,"voltage":224}'
Info 2023-11-21 23:52:53Update of 'Enchufe_garaje' at 31.82%, ≈ 4 minutes remaining
Info 2023-11-21 23:52:53MQTT publish: topic 'zigbee2mqtt/Enchufe_garaje', payload '{"child_lock":"UNLOCK","current":0,"energy":0,"indicator_mode":"off/on","linkquality":116,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":77,"latest_version":192,"progress":31.82,"remaining":259,"state":"updating"},"update_available":null,"voltage":224}'
Info 2023-11-21 23:53:23Update of 'Enchufe_garaje' at 33.61%, ≈ 5 minutes remaining
Info 2023-11-21 23:53:23MQTT publish: topic 'zigbee2mqtt/Enchufe_garaje', payload '{"child_lock":"UNLOCK","current":0,"energy":0,"indicator_mode":"off/on","linkquality":112,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":77,"latest_version":192,"progress":33.61,"remaining":298,"state":"updating"},"update_available":null,"voltage":224}'
Info 2023-11-21 23:53:53Update of 'Enchufe_garaje' at 35.39%, ≈ 6 minutes remaining
Info 2023-11-21 23:53:53MQTT publish: topic 'zigbee2mqtt/Enchufe_garaje', payload '{"child_lock":"UNLOCK","current":0,"energy":0,"indicator_mode":"off/on","linkquality":112,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":77,"latest_version":192,"progress":35.39,"remaining":331,"state":"updating"},"update_available":null,"voltage":224}'
Info 2023-11-21 23:54:23Update of 'Enchufe_garaje' at 37.18%, ≈ 6 minutes remaining
Info 2023-11-21 23:54:23MQTT publish: topic 'zigbee2mqtt/Enchufe_garaje', payload '{"child_lock":"UNLOCK","current":0,"energy":0,"indicator_mode":"off/on","linkquality":116,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":77,"latest_version":192,"progress":37.18,"remaining":357,"state":"updating"},"update_available":null,"voltage":224}'
Info 2023-11-21 23:54:54Update of 'Enchufe_garaje' at 38.97%, ≈ 6 minutes remaining
Info 2023-11-21 23:54:54MQTT publish: topic 'zigbee2mqtt/Enchufe_garaje', payload '{"child_lock":"UNLOCK","current":0,"energy":0,"indicator_mode":"off/on","linkquality":112,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":77,"latest_version":192,"progress":38.97,"remaining":378,"state":"updating"},"update_ava [... cut ...]
info  2023-11-22 00:14:09: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"from":null,"id":"Enchufe_garaje","to":null},"status":"ok","transaction":"3rwx8-25"}'
info  2023-11-22 00:14:10: Successfully configured 'Enchufe_garaje'
info  2023-11-22 00:14:13: MQTT publish: topic 'zigbee2mqtt/Enchufe_garaje', payload '{"child_lock":"UNLOCK","current":0,"energy":0,"indicator_mode":"off/on","linkquality":120,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":null,"voltage":172}'

```

**After update Enchufe_garaje**

[image]

[image]

XisoP01, 2024-06-03:
> The issue is still present on Z2M version 1.37.1

+1. Not able to update on Sonoff dongle and SLZB-06

rvalitov, 2024-10-21:
The issue is still present on Zigbee2MQTT version 1.40.2

rvalitov, 2025-01-05:
Up

thierrybla, 2025-03-08:
Unstale

Ricc68, 2025-05-03:
Fixed by mentioned PR https://github.com/nurikk/zigbee2mqtt-frontend/pull/2470.

closed 2025-07-11 by github-actions[bot]