Koenkk/zigbee2mqtt #16422: Light Switches Frozen After Upgrade TS0012 TS0013

kiwizznz, 2023-01-30:
### What happened?

After upgrading to [1.29.2] (was a few months behind) my light switches started to freeze. Rebooted z2m multiple times, tried turning it off and power cycling the house, each time slightly different results but in all cases at least 2 of my light switches remain frozen (as in, won't perform any action, touch switches don't work, holding down to reset doesn't work). These two are still unusable after power cycling the house 4 times (the others have come back). 

        fingerprint: [{modelID: 'TS0013', manufacturerName: '_TZ3000_khtlvdfc'}],
        model: 'SSW03G',
        vendor: 'Mercator',
        description: 'Ikuü Triple Switch',
        
        fingerprint: [{modelID: 'TS0012', manufacturerName: '_TZ3000_4zf0crgo'}],
        model: 'SSW02G',
        vendor: 'Mercator',
        description: 'Ikuü Double Switch',

### What did you expect to happen?

After upgrading was hoping everything would continue to work as expected. Initially nothing connected but after rebooting the host most devices connected fine.

Note - I don't think the debug logs will help as the device doesn't appear to be reporting anything anymore.

### How to reproduce it (minimal and precise)

Not sure as haven't been able to pinpoint the actual cause. 

### Zigbee2MQTT version

1.29.2

### Adapter firmware version

20220219

### Adapter

SONOFF ZB Dongle-P Plus Zigbee 3.0 USB

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/10534920/log.txt)

makonde-on-git, 2023-01-30:
Same problem for me - Z2M 1.29.2, both TS0012 and TS0011 stopped working after ~24h. Adapter SONOFF ZB Dongle-E with fw 6.10.3.0 build 297), Z2M running as a HA addon.

kiwizznz, 2023-02-01:
@makonde-on-git - did you try going back a version? are they still frozen/stuck?

makonde-on-git, 2023-02-01:
@kiwizznz Integrations broke at evening so due to a "wife factor" I needed to wait until the morning. After allowing devices to join in Z2M I re-paired all broken switches. Z2M reported those left the network and joined right after. So far everything is working fine.
No downgrade, not even a Z2M restart was needed.

kiwizznz, 2023-02-01:
@makonde-on-git hahaha - have the same wife factor issue. Still stuck with a couple switches which I can't put in pairing mode, even after turning the house off and on a few times via the main breaker.

kiwizznz, 2023-02-01:
I just did another power of/on and different switches are now frozen.... so weird. It's like roulette!

kiwizznz, 2023-02-01:
@makonde-on-git out of curiosity, how many devices have you got. I'm wondering if my ~65 devices might be causing the switches to freeze somehow. Might test this weekend by removing a few and restarting the house again...

makonde-on-git, 2023-02-01:
@kiwizznz I currently have 24 devices connected, with four being either TS001 or TS0012.

kiwizznz, 2023-02-02:
Have tried rebuilding the entire network and changed channel. Still have a couple devices that seem frozen completely, others started working again but returned to freeze state after adding everything back.

closed 2023-03-13 by github-actions[bot]

MnM001, 2023-10-01:
Hi @makonde-on-git  and @kiwizznz  - did you guys managed to sort out this issue at all?

I have the same thing happening for me. After a house power off/on different switches are frozen and dont work while the rest are OK and work as it should. 
Every time I do house power off/on deferent switches (1 or 2 maximum) are frozen while the others work as they should. I know the switched are fine as a switch that works without any issues now might become frozen after my next house power off/on.

Its so annoying and will really like to fix this issue.