Koenkk/zigbee2mqtt #16428: Bulbs randomly turning on at different times of the day

KidA001, 2023-01-30:
### What happened?

I have a three philips hue bulbs that seemingly randomly turn on at different times of the day and night. These three bulbs are in a group together but they turn on individually. I also have a remove that is bound to the group. I am running Z2M as it's own instance but it is connected to HomeAssistant. I have verified through the logs in HA that they are not being turned on by another users/automation...etc.

I am perplexed and trying to understand why these go on, especially at night which will wake me. Any help appreciated.

### What did you expect to happen?

I don't expect them to turn on unless I initiate it.

### How to reproduce it (minimal and precise)

There is nothing specific I do that makes it happen. It seems to be random

### Zigbee2MQTT version

1.29.0

### Adapter firmware version

20220219

### Adapter

zStack3x0

### Debug log

I pinpointed a time when they turned on, the culprit bulbs are `living_room_lamp_bulb_1` (2 and 3). They are in a group `living_room_lamp`. In the logs below you can see the bulb turns on at `2023-01-30 14:38:41` which I did not initiate myself. Does this give any insight into what turned the state to on?

```
info  2023-01-30 14:37:17: MQTT publish: topic 'zigbee2mqtt/Pump House Heater', payload '{"linkquality":40,"state":"OFF"}'
info  2023-01-30 14:37:19: MQTT publish: topic 'zigbee2mqtt/Sonoff Switch NonOp', payload '{"linkquality":29,"state":"ON"}'
info  2023-01-30 14:37:19: MQTT publish: topic 'zigbee2mqtt/Speakers', payload '{"linkquality":32,"state":"OFF"}'
info  2023-01-30 14:38:39: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Living Room Lamp Bulb 1","ieee_address":"0x0017880108194709"},"type":"device_announce"}'
info  2023-01-30 14:38:41: MQTT publish: topic 'zigbee2mqtt/living_room_lamp', payload '{"brightness":254,"state":"ON"}'
info  2023-01-30 14:38:41: MQTT publish: topic 'zigbee2mqtt/Living Room Lamp Bulb 1', payload '{"brightness":254,"linkquality":25,"power_on_behavior":null,"state":"ON","update":{"installed_version":16786432,"latest_version":16786432,"state":"idle"},"update_available":null}'
info  2023-01-30 14:38:45: MQTT publish: topic 'zigbee2mqtt/Living Room Lamp Bulb 1', payload '{"brightness":254,"linkquality":29,"power_on_behavior":null,"state":"ON","update":{"installed_version":16786432,"latest_version":16786432,"state":"idle"},"update_available":null}'
info  2023-01-30 14:38:45: MQTT publish: topic 'zigbee2mqtt/living_room_lamp', payload '{"brightness":254,"state":"ON"}'
info  2023-01-30 14:38:45: MQTT publish: topic 'zigbee2mqtt/Living Room Lamp Bulb 1', payload '{"brightness":254,"linkquality":29,"power_on_behavior":null,"state":"ON","update":{"installed_version":16786432,"latest_version":16786432,"state":"idle"},"update_available":null}'
info  2023-01-30 14:39:11: MQTT publish: topic 'zigbee2mqtt/Garage Door', payload '{"battery":90,"contact":true,"device_temperature":7,"linkquality":160,"power_outage_count":5,"voltage":2985}'

KidA001, 2023-01-30:
These are Hue Edison bulbs and this issue seems similar, but there was no resolve posted: https://github.com/Koenkk/zigbee2mqtt/issues/6638

danieledwardgeorgehitchcock, 2023-01-31:
It looks like your devices may be dropping off of the network - the `device announce` messages seems a bit suspect to me...

link1873, 2023-02-10:
Having the exact same issue with my bedroom lights. Haven't had any changes to my network only updates to z2m. Last few versions. Haven't been able to pinpoint any other issue causing this

KidA001, 2023-02-24:
> It looks like your devices may be dropping off of the network - the `device announce` messages seems a bit suspect to me...

They are surrounded by other router devices that have no issues. It seemed to happen after upgrading the firmware of these bulbs

KidA001, 2023-02-25:
So I think I was able to fix this by setting the `power_on_behavior` to `previous` over MQTT. Based on what @danieledwardgeorgehitchcock mentioned I thought these bulbs may think they are reconnecting or powering on. To be clear they are most definitely _not_ losing power.

So `power_on_behavior` was and has always been `null` which will default to `on`. What's interesting is I initially set `power_on_behavior` to `off` last night and the opposite problem would happen in that they would randomly turn OFF. So then I set it to `previous` and it seems to have resolved the issue. I have not had them randomly turning on or off.

Why these bulbs are reconnecting in this way and randomly acting as if they are just being powered on is beyond me. This seems to have started happening after I upgraded these bulbs to `16786432` which as of current is the latest firmware. In any case setting the behavior to `previous` so far seems to have resolved the issue of them randomly turning ON. This will save me a lot of sleep.

I hope this issue gets some attention, or at the very lease helps someone else. I will close this in a few days after I confirm the issue is no longer happening. It very well may be a firmware issue and not a zigbee2mqtt issue, but that is unclear to me.

This is exactly how I resolved the issue:

- I used [MQTT Explorer](http://mqtt-explorer.com/) to connect to my MQTT Server
- I published the following to the bulbs that were having the issue:
  - topic: `zigbee2mqtt/name-of-bulb-here/set`
  - JSON payload: `{ "power_on_behavior": "previous" }`

link1873, 2023-02-25:
> So I think I was able to fix this by setting the `power_on_behavior` to `previous` over MQTT. Based on what @danieledwardgeorgehitchcock mentioned I thought these bulbs may think they are [...]

Did the same and same think they've been good for a few days for me too.  If easier for people can also just go inside Z2M>click the bulb>exposes and select power on behavior there.

KidA001, 2023-02-25:
@link1873 are these also Hue Edison bulbs?

link1873, 2023-02-25:
> @link1873 are these also Hue Edison bulbs?
> 
> Did not know you could set that in 'exposes' I will update my last comment. Way easier

regular hue colour bulbs

closed 2023-03-01 by KidA001