Koenkk/zigbee2mqtt #16433: state from rtczcgq11lm

arteck, 2023-01-31:
### What happened?

https://github.com/Koenkk/zigbee-herdsman-converters/blob/cee48c8a56a19fc58a1519417e14989b68624173/devices/xiaomi.js#L1684

why is this a enum ???

### How to reproduce it (minimal and precise)

sdf

### Zigbee2MQTT version

sfd

### Adapter firmware version

sdf

### Adapter

sf

### Debug log

fdssdgf

danieledwardgeorgehitchcock, 2023-01-31:
enum what is used by Z2M frontend to render buttons - it should still work properly via MQTT messages

arteck, 2023-02-01:
lol
I would have expected a button

danieledwardgeorgehitchcock, 2023-02-01:
[Here](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/lib/exposes.js#L175-L211) is all of the available item types that can be used in the exposes section - `enum` allows for button groups to be created as well as single buttons in the UI

closed 2023-03-07 by arteck