Koenkk/zigbee2mqtt #16449: 14592 Vimar cant get state on/off

datsrt02, 2023-02-01:
### What happened?

I meet 2 issues with this Vimar device:
**First**: I try connect Vimar switch 14592 to zigbee2mqtt, expose have: sate (on/off), linkquality. But i can only click ON to turn switch on, click OFF nothing happen, if i want to turn switch Off, must click to ON again. 
**Second**: When i press switch, sate of switch in zigbee2mqtt dashboard no responding. No state ON or OFF change.
[image]
[image]

### Zigbee2MQTT version

1.29.2-1 

### Adapter firmware version

20210708

### Adapter

Silicon_Labs_Sonoff_Zigbee_3.0_USB_Dongle_Plus

### Debug log

Zigbee2MQTT:debug 2023-02-01 11:19:15: Saving state to file /config/zigbee2mqtt/state.json
Zigbee2MQTT:debug 2023-02-01 11:21:32: Received MQTT message on 'zigbee2mqtt/vimar switch/set' with data '{"state":"ON"}'
Zigbee2MQTT:debug 2023-02-01 11:21:32: Publishing 'set' 'state' to 'vimar switch'
Zigbee2MQTT:info  2023-02-01 11:21:32: MQTT publish: topic 'zigbee2mqtt/vimar switch', payload '{"device":{"applicationVersion":1,"dateCode":"20200501","friendlyName":"vimar switch","hardwareVersion":11,"ieeeAddr":"0xf4ce36928b984730","manufacturerID":4660,"manufacturerName":"Vimar","model":"14592.0","networkAddress":35106,"powerSource":"Mains (single phase)","softwareBuildID":"","stackVersion":10,"type":"Router","zclVersion":2},"linkquality":174,"state":"ON"}'
Zigbee2MQTT:debug 2023-02-01 11:21:33: Received MQTT message on 'zigbee2mqtt/vimar switch/set' with data '{"state":"ON"}'
Zigbee2MQTT:debug 2023-02-01 11:21:33: Publishing 'set' 'state' to 'vimar switch'
Zigbee2MQTT:info  2023-02-01 11:21:33: MQTT publish: topic 'zigbee2mqtt/vimar switch', payload '{"device":{"applicationVersion":1,"dateCode":"20200501","friendlyName":"vimar switch","hardwareVersion":11,"ieeeAddr":"0xf4ce36928b984730","manufacturerID":4660,"manufacturerName":"Vimar","model":"14592.0","networkAddress":35106,"powerSource":"Mains (single phase)","softwareBuildID":"","stackVersion":10,"type":"Router","zclVersion":2},"linkquality":174,"state":"ON"}'
Zigbee2MQTT:debug 2023-02-01 11:21:34: Received MQTT message on 'zigbee2mqtt/vimar switch/set' with data '{"state":"ON"}'
Zigbee2MQTT:debug 2023-02-01 11:21:34: Publishing 'set' 'state' to 'vimar switch'
Zigbee2MQTT:info  2023-02-01 11:21:34: MQTT publish: topic 'zigbee2mqtt/vimar switch', payload '{"device":{"applicationVersion":1,"dateCode":"20200501","friendlyName":"vimar switch","hardwareVersion":11,"ieeeAddr":"0xf4ce36928b984730","manufacturerID":4660,"manufacturerName":"Vimar","model":"14592.0","networkAddress":35106,"powerSource":"Mains (single phase)","softwareBuildID":"","stackVersion":10,"type":"Router","zclVersion":2},"linkquality":174,"state":"ON"}'
Zigbee2MQTT:debug 2023-02-01 11:21:35: Received MQTT message on 'zigbee2mqtt/vimar switch/set' with data '{"state":"OFF"}'
Zigbee2MQTT:debug 2023-02-01 11:21:35: Publishing 'set' 'state' to 'vimar switch'
Zigbee2MQTT:info  2023-02-01 11:21:35: MQTT publish: topic 'zigbee2mqtt/vimar switch', payload '{"device":{"applicationVersion":1,"dateCode":"20200501","friendlyName":"vimar switch","hardwareVersion":11,"ieeeAddr":"0xf4ce36928b984730","manufacturerID":4660,"manufacturerName":"Vimar","model":"14592.0","networkAddress":35106,"powerSource":"Mains (single phase)","softwareBuildID":"","stackVersion":10,"type":"Router","zclVersion":2},"linkquality":174,"state":"OFF"}'
Zigbee2MQTT:debug 2023-02-01 11:21:36: Received MQTT message on 'zigbee2mqtt/vimar switch/set' with data '{"state":"ON"}'
Zigbee2MQTT:debug 2023-02-01 11:21:36: Publishing 'set' 'state' to 'vimar switch'
Zigbee2MQTT:info  2023-02-01 11:21:36: MQTT publish: topic 'zigbee2mqtt/vimar switch', payload '{"device":{"applicationVersion":1,"dateCode":"20200501","friendlyName":"vimar switch","hardwareVersion":11,"ieeeAddr":"0xf4ce36928b984730","manufacturerID":4660,"manufacturerName":"Vimar","model":"14592.0","networkAddress":35106,"powerSource":"Mains (single phase)","softwareBuildID":"","stackVersion":10,"type":"Router","zclVersion":2},"linkquality":177,"state":"ON"}'
Zigbee2MQTT:debug 2023-02-01 11:21:37: Received MQTT message on 'zigbee2mqtt/vimar switch/set' with data '{"state":"OFF"}'
Zigbee2MQTT:debug 2023-02-01 11:21:37: Publishing 'set' 'state' to 'vimar switch'
Zigbee2MQTT:info  2023-02-01 11:21:37: MQTT publish: topic 'zigbee2mqtt/vimar switch', payload '{"device":{"applicationVersion":1,"dateCode":"20200501","friendlyName":"vimar switch","hardwareVersion":11,"ieeeAddr":"0xf4ce36928b984730","manufacturerID":4660,"manufacturerName":"Vimar","model":"14592.0","networkAddress":35106,"powerSource":"Mains (single phase)","softwareBuildID":"","stackVersion":10,"type":"Router","zclVersion":2},"linkquality":180,"state":"OFF"}'

closed 2023-02-01 by datsrt02

dregini, 2023-06-04:
Sorry to resurrect this is issue, I have the same problem. I'm totally new to Zigbee in genera could you explain how did you solved?

datsrt02, 2023-06-05:
> Sorry to resurrect this issue, I have the same problem. I'm totally new to Zigbee, could you explain how did you solved?

U should connect device with load ( as bulb light,...), it will work if have a load been connected